awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #631: [QUESTION] Kinesis video streams for live video grid display

bml1g12, 2020-12-10:
I saw that Kinesis supports both HLS and WebRTC for browser playback; was wondering if anyone has any thoughts on whether it would be possible to syncronise more than one video being played back in the browser using Kinesis video streams? i.e. to view a grid of live camera feeds (or historical feeds) for CCTV application. 

I imagined this is a fairly common use case, but I can find little resources online about how would go about achieving this.

Thanks for any ideas,

MushMal, 2020-12-10:
From the technical perspective - there is no reason why this wouldn't work. We just don't have an example of one. However, one could take the sample and put on different canvases on the same page? 

BTW, good suggestion for a sample/demo app in the demo repository here: https://github.com/aws-samples/amazon-kinesis-video-streams-demos

I would strongly encourage you to create one in there with an "enhancement" tag

bml1g12, 2020-12-11:
> o reason why this wouldn't work. We just don't have an example of one. However, one could take the sample and put on different canvases on the sam

I see, its useful to know that maybe this is not something already implemented in a demo/sample. I think from a technical standpoint the questions would be things like:

* Can one obtain the precise timestamp of each frame from HLS or WebRTC streams?
* Is there any native support for synchronising these between more than 1 stream via either skipping frames or via pausing the stream, or would it need to be 100% custom built.

MushMal, 2020-12-11:
For frame timestamps you will need to crack the packaging. There is no existing tool that I am aware of that could handle cross-stream sync, I am afraid it needs to be written. None of these are hard to implement but need to be prioritized

bml1g12, 2020-12-11:
Thanks, but could I ask what you mean by "crack the packaging" does that mean you would have to manually parse the encoded packets or somesuch?

MushMal, 2020-12-11:
You need to parse the mpeg package. No need to decode the frames

bml1g12, 2020-12-11:
I see thank you, Ill close this, put if anyone wants to add any more comments please do

closed 2020-12-11 by bml1g12