awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #635: [QUESTION]SDK piling up frames in buffer (like a network outage) and when application restarted working seamlessly

Subburaj47, 2020-12-22:
We have a scenario where the SDK piles up the frames in the pipeline. We have an application logic where we have the uninit/re-init done when the frame ack is not obtained for close to ~1 minute. Due to this logic there is an uninit and then application is restarted. 
Immediately after a restart, the SDK is working seamlessly. In the event of a network issue, we don't see a staleconnectionHandler callback being hit (the time configured is 200ms) and if there is a network outage, I wonder how it got recovered after an SDK restart. 
Could you pls suggest if there is something to be done in our configuration or any other suggestions on this?

** Any design considerations/constraints **
Explain in detail how you would like to integrate our SDK into your solution 

** If you would not like to open an issue to discuss your solution in open-platform, please email your question to kinesis-video-support@amazon.com **

MushMal, 2020-12-22:
I don't fully understand the question. What does "piling" mean? Does it mean the SDK is buffering? If so, you could see what's happening - it's either unable to send the frames at the sustained rate or it's able to send the frames but the ACKs are not received. 

The latter case is what's known as stale connection. The stale connection callback is what gets fired when this happens and the default continuous callback provider resets the connection. 
The staleness could happen due to a number of reasons - for example, an intermediary (a switch or a router) could be getting the packets, sending network acks but dropping the packets without forwarding to the AWS. We had customers who's routers were misconfigured causing this issue.

Setting the staleness in StreamInfo.StreamCaps.connectionStelenessDuration (https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/master/src/client/include/com/amazonaws/kinesis/video/client/Include.h#L977) to fire is the right way to get notified when the staleness event happens. However, setting it to 200ms is not going to work. ACKs are sent on fragment granularity. The default configuration is I believe 40 seconds.

Subburaj47, 2020-12-23:
Thanks for the response @MushMal . Apologies for the ambiguity. By "piling" I meant buffering. There was a typo in my comment. The staleness value we have configured so far is 20 seconds.

Subburaj47, 2020-12-23:
Also I'm curious that whenever we have the application restarted, this behavior (of latency pressure handler and then frame drops and eventually no fragment ack for close to 1 minute) is recovered. Do you have any comments to solve/suggestions to debug such problems?

MushMal, 2020-12-23:
It's hard to say. We have seen cases when a particular network route is bad and resetting the connection (which is the default behavior in continuous retry policy on staleness and pressure) resolves it just fine as a new routing is able to stream faster than realtime and deplete the accumulated buffer.

However, there have been cases on custom platforms when the network stack is buggy (some related to WiFi drivers/NDIS in user mode) when a restart of the process "clears" it.

It's truly hard to debug these cases but trying different scenarios and bisecting the problem usually helps. In this case, make sure you differentiate between the staleness (no acks) and network throughput (no data is sent and the buffer gets accumulated). Next, see if the default behavior fires - for example, in case of staleness, after the max staleness duration is reached, you should see a reset connection happening in the logs. In case of latency, the same. Check whether the new session is able to recover. Run with verbose log mode to get more info.

If the recovery on connection reset doesn't happen but you restarting the app works OK then it points more towards a driver type of issues.

disa6302, 2020-12-28:
Closing based on tag.

closed 2020-12-28 by disa6302