awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #636: No such element or plugin 'kvssink'

AI-07, 2020-12-22:
I have just installed this SDK on my laptop with Ubuntu OS on it. But I'm facing a weird problem that it only works in super user mode, and throws error in other mode. I'm running `gst-inspect-1.0 kvssink`  command, to check if its properly installed, as mentioned in the documentation. what could possibly wrong here, help would be much appreciated.

disa6302, 2020-12-22:
Hello @AI-07 ,

Have you run `cmake` with `-DBUILD_GSTREAMER_PLUGIN=ON` option?

Can you verify `libgstkvssink` exists in the build directory?

Have you run `export GST_PLUGIN_PATH=<path-to-build-directory` ?

Can you verify your cmake and make steps ran without any errors?

AI-07, 2020-12-22:
> Hello @AI-07 ,
> 
> Have you run `cmake` with `-DBUILD_GSTREAMER_PLUGIN=ON` option?
> 
> Can you verify `libgstkvssink` exists in the build directory?
> 
> Have you run `export GST_PLUGIN_PATH=<path-to-build-directory` ?
> 
> Can you verify your cmake and make steps ran without any errors?

performed each of the above mentioned steps, I guess that's why it works in super user (sudo su) mode. And yes **libgstkvssink** exists in the build directory.

disa6302, 2020-12-22:
@AI-07 ,

It does not seem to be a KVS specific issue. Kvssink does not require super user. I would encourage you to check if you ever ran `sudo -i` or `sudo  -S` in the past that moved from user to super user. I am closing this issue since it is unrelated to kvs.

closed 2020-12-22 by disa6302