awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #638: [QUESTION] IAM Policy for IoT specific resource for kinesis video stream

wlinInspire, 2020-12-25:
Hi KVS team,

Following this doc [Controlling Access to Kinesis Video Streams Resources Using AWS IoT](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/how-iot.html), I successfully stream video to the thing name channel using gstreamer plugin based on the following IAM policy
```yml
Resource: "arn:aws:kinesisvideo:*:*:stream/${credentails-iot:ThingName}/*"
```
However, it fails to send video to the corresponding channels if I change the policy to
```yml
Resource: 
    - "arn:aws:kinesisvideo:*:*:stream/${credentails-iot:ThingName}_1/*"
    - "arn:aws:kinesisvideo:*:*:stream/${credentails-iot:ThingName}_2/*"
```
Is there a way to stream video to multiple different but specific channels?

Thanks.

wlinInspire, 2021-01-06:
@MushMal Could you please take a look at my question? Thanks.

MushMal, 2021-01-06:
I assume you are referring to a stream rather than a channel. Each IoT certificate can be used for a single stream resource as the stream corresponds to the thing name. 

I would also recommend contacting your AWS technical account manager for these types of questions as we might have limited knowledge of the IAM/IoT

disa6302, 2021-01-12:
Closing since the question is answered. Feel free to reopen if you have any more questions.

closed 2021-01-12 by disa6302