awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #639: [Feature] support macOS big sur

dfang, 2020-12-26:
**Is your feature request related to a problem? Please describe.**
```
cd /tmp
git clone https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git
mkdir -p amazon-kinesis-video-streams-producer-sdk-cpp/build
cd amazon-kinesis-video-streams-producer-sdk-cpp/build
cmake ..
make
```

```
λ  brew install gstreamer gst-plugins-base gst-plugins-good gst-plugins-bad gst-plugins-ugly gst-libav
λ  cmake ..
CMake Deprecation Warning at CMakeLists.txt:5 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Kinesis Video Cpp Producer path is /tmp/amazon-kinesis-video-streams-producer-sdk-cpp
-- dependencies install path is /tmp/amazon-kinesis-video-streams-producer-sdk-cpp/open-source
-- Begin building dependencies.
-- autoconf already built
-- automake already built
-- log4cplus already built
-- Finished building dependencies.
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer
[ 11%] Performing update step for 'libkvscproducer-download'
[ 22%] No patch step for 'libkvscproducer-download'
[ 33%] No configure step for 'libkvscproducer-download'
[ 44%] No build step for 'libkvscproducer-download'
[ 55%] No install step for 'libkvscproducer-download'
[ 66%] No test step for 'libkvscproducer-download'
[ 77%] Completed 'libkvscproducer-download'
[100%] Built target libkvscproducer-download
CMake Deprecation Warning at dependency/libkvscproducer/kvscproducer-src/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Kinesis Video Producer path is /tmp/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src
-- Begin building dependencies.
-- openssl already built
-- curl already built
-- Finished building dependencies.
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic
[ 11%] Performing update step for 'libkvspic-download'
[ 22%] No patch step for 'libkvspic-download'
[ 33%] No configure step for 'libkvspic-download'
[ 44%] No build step for 'libkvspic-download'
[ 55%] No install step for 'libkvspic-download'
[ 66%] No test step for 'libkvspic-download'
[ 77%] Completed 'libkvspic-download'
[100%] Built target libkvspic-download
CMake Deprecation Warning at dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Found OpenSSL: /tmp/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/libcrypto.dylib (found version "1.1.1g")  
-- Bitness 64 bits
-- Found OpenSSL: /tmp/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/libcrypto.dylib (found suitable version "1.1.1g", minimum required is "1")  
found log4cplus.
-- Configuring done
-- Generating done

 λ  make
[ 43%] Built target kvspic
[ 66%] Built target kvspicUtils
[ 75%] Built target kvsCommonCurl
[ 88%] Built target cproducer
[100%] Built target KinesisVideoProducer

 λ  cd ..
 λ  export GST_PLUGIN_PATH=`pwd`/build
 λ  gst-inspect-1.0 kvssink
No such element or plugin 'kvssink'
 λ  ls build
CMakeCache.txt  CMakeFiles  Makefile  cmake_install.cmake  dependency  libKinesisVideoProducer.dylib
```

**Describe the solution you'd like**
`gst-inspect-1.0 kvssink` works as expected

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
no errors when `cmake .. and make`

disa6302, 2020-12-28:
@dfang , 

There is no explicit support for MacOS Bigsur. You need to run `cmake .. -DBUILD_GSTREAMER_PLUGIN=ON`. Follow the instructions on the github `README.md`

closed 2020-12-28 by disa6302

dfang, 2020-12-29:
```
/tmp/amazon-kinesis-video-streams-producer-sdk-cpp/build λ  cmake .. -DBUILD_GSTREAMER_PLUGIN=ON
CMake Deprecation Warning at CMakeLists.txt:5 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Kinesis Video Cpp Producer path is /tmp/amazon-kinesis-video-streams-producer-sdk-cpp
-- dependencies install path is /tmp/amazon-kinesis-video-streams-producer-sdk-cpp/open-source
-- Begin building dependencies.
-- autoconf already built
-- automake already built
-- log4cplus already built
-- Finished building dependencies.
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer
[ 11%] Performing update step for 'libkvscproducer-download'
[ 22%] No patch step for 'libkvscproducer-download'
[ 33%] No configure step for 'libkvscproducer-download'
[ 44%] No build step for 'libkvscproducer-download'
[ 55%] No install step for 'libkvscproducer-download'
[ 66%] No test step for 'libkvscproducer-download'
[ 77%] Completed 'libkvscproducer-download'
[100%] Built target libkvscproducer-download
CMake Deprecation Warning at dependency/libkvscproducer/kvscproducer-src/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Kinesis Video Producer path is /tmp/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src
-- Begin building dependencies.
-- openssl already built
-- curl already built
-- Finished building dependencies.
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic
[ 11%] Performing update step for 'libkvspic-download'
[ 22%] No patch step for 'libkvspic-download'
[ 33%] No configure step for 'libkvspic-download'
[ 44%] No build step for 'libkvspic-download'
[ 55%] No install step for 'libkvspic-download'
[ 66%] No test step for 'libkvspic-download'
[ 77%] Completed 'libkvspic-download'
[100%] Built target libkvspic-download
CMake Deprecation Warning at dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
[... 170 lines cut ...]
/tmp/amazon-kinesis-video-streams-producer-sdk-cpp/src/Logger.h:40:37: note: expanded from macro 'LOG_CONFIGURE_STDOUT'
#define LOG_CONFIGURE_STDOUT(level) _LOG_CONFIGURE_CONSOLE(level, false)
                                    ^
/tmp/amazon-kinesis-video-streams-producer-sdk-cpp/src/Logger.h:26:10: note: expanded from macro '_LOG_CONFIGURE_CONSOLE'
    std::auto_ptr<log4cplus::Layout> _layout(new log4cplus::PatternLayout("%D [%t] ")); \
         ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/memory:2135:28: note: 'auto_ptr<log4cplus::Layout>'
      has been explicitly marked deprecated here
class _LIBCPP_TEMPLATE_VIS _LIBCPP_DEPRECATED_IN_CXX11 auto_ptr
                           ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/__config:1030:39: note: expanded from macro
      '_LIBCPP_DEPRECATED_IN_CXX11'
#  define _LIBCPP_DEPRECATED_IN_CXX11 _LIBCPP_DEPRECATED
                                      ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/__config:1019:48: note: expanded from macro
      '_LIBCPP_DEPRECATED'
#    define _LIBCPP_DEPRECATED __attribute__ ((deprecated))
                                               ^
1 warning generated.
[100%] Linking CXX shared module libgstkvssink.so
[100%] Built target gstkvssink

/tmp/amazon-kinesis-video-streams-producer-sdk-cpp/build λ  gst-inspect-1.0 kvssink
No such element or plugin 'kvssink'
```