awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #641: KVS is not receiving my RTSP video

hdnh2006, 2021-01-08:
Hi, I am trying o use my camera who uses RTSP protocol to send a video stream to KVS.

I am using [this](https://www.ezvizlife.com/product/c6c-ez360/722) camera who clearly support h264 according to the specifications and I haved tried on VLC. 

So I tried both ways:

**1) Directly into my local:** 
After following all the instruccions [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp), I was able to send images using a USB camera with: 
`gst-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! x264enc  bframes=0 key-int-max=45 bitrate=500 ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink stream-name="my_kvs_test" storage-size=512 access-key="AKIAY***" secret-key="H31****" aws-region="eu-west-1"` 

And I succesfully see the image in my console:
[image]

So I decided to do the same (with the correct command) but using a RTSP camera and I got the following logs:
```
user:~$ gst-launch-1.0 rtspsrc location="rtsp://*****:****@192.***.*.**/stream1" short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! kvssink stream-name="my_kvs_test" storage-size=512 access-key="AKIAY***" secret-key="H31****" aws-region="eu-west-1"

Setting pipeline to PAUSED ...
log4cplus:ERROR could not open file ./kvs_log_configuration
INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2021-01-08 17:45:52 [140376197571520] INFO - heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
2021-01-08 17:45:52 [140376197571520] INFO - heapInitialize(): Creating AIV heap.
2021-01-08 17:45:52 [140376197571520] INFO - heapInitialize(): Heap is initialized OK
2021-01-08 17:45:52 [140376197571520] DEBUG![image](https://user-images.githubusercontent.com/17271049/104042678-14961580-51db-11eb-88aa-4ac0f62b0552.png)
![image](https://user-images.githubusercontent.com/17271049/104043111-a0a83d00-51db-11eb-93e5-78f2cb212681.png)
![image](https://user-images.githubusercontent.com/17271049/104043120-a3a32d80-51db-11eb-94ce-535fc5c30969.png - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2021-01-08 17:45:52 [140376197571520] DEBUG - getSecurityTokenHandler invoked
2021-01-08 17:45:52 [140376197571520] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1610124352102600714 Expiration: 0
2021-01-08 17:45:52 [140376197571520] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
2021-01-08 17:45:52 [140376197571520] INFO - createDeviceResultEvent(): Create device result event.
2021-01-08 17:45:52 [140376197571520] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
2021-01-08 17:45:52 [140376197571520] DEBUG - clientReadyHandler invoked
2021-01-08 17:45:52 [140376197571520] INFO - try creating stream
2021-01-08 17:45:52 [140376197571520] INFO - Creating Kinesis Video Stream my_kvs_test
2021-01-08 17:45:52 [140376197571520] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2021-01-08 17:45:52 [140376197571520] DEBUG - logStreamInfo(): Kinesis Video Stream Info
2021-01-08 17:45:52 [140376197571520] DEBUG - logStreamInfo(): 	Stream name: my_kvs_test 
2021-01-08 17:45:52 [140376197571520] DEBUG - logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME 
2021-01-08 17:45:52 [140376197571520] DEBUG - logStreamInfo(): 	Content type: video/h264 
2021-01-08 17:45:52 [140376197571520] DEBUG - logStreamInfo(): 	Max latency (100ns): 600000000
2021-01-08 17:45:52 [140376197571520] DEBUG - logStreamInfo(): 	Fragment duration (100ns): 20000000
2021-01-08 17:45:52 [140376197571520] DEBUG - logStreamInfo(): 	Key frame fragmentation: Yes
2021-01-08 17:45:52 [140376197571520] DEBUG - logStreamInfo(): 	Use frame timecode: Yes
2021-01-08 17:45:52 [140376197571520] DEBUG - logStreamInfo(): 	Absolute frame timecode: Yes
2021-01-08 17:45:52 [140376197571520] DEBUG - logStreamInfo(): 	Nal adaptation flags: 0
2021-01-08 17:45:52 [140376197571520] DEBUG - logStreamInfo(): 	Average bandwith (bps): 4194304
2021-01-08 17:45:52 [140376197571520] DEBUG - logStreamInfo(): 	Framerate: 25
2021-01-08 17:45:52 [140376197571520] DEBUG - logStreamInfo(): 	Buffer duration (100ns): 1200000000
2021-01-08 17:45:52 [140376197571520] DEBUG - logStreamInfo(): 	Replay duration (100ns): 400000000
2021-01-08 17:45:52 [140376197571520] DEBUG - logStreamInfo(): 	Connection Staleness duration (100ns): 600000000
2021-01-08 17:45:52 [140376197571520] DEBUG - logStreamInfo(): 	Store Pressure Policy: 1
2021-01-08 17:45:52 [140376197571520] DEBUG - logStreamInfo(): 	View Overflow Policy: 1
2021-01-08 17:45:52 [140376197571520] DEBUG - logStreamInfo(): 	Segment UUID: NULL
2021-01-08 17:45:52 [140376197571520] DEBUG - logStreamInfo(): 	Frame ordering mode: 0
2021-01-08 17:45:52 [140376197571520] DEBUG - logStreamInfo(): Track list
2021-01-08 17:45:52 [140376197571520] DEBUG - logStreamInfo(): 	Track id: 1
2021-01-08 17:45:52 [140376197571520] DEBUG - logStreamInfo(): 	Track name: kinesis_video
2021-01-08 17:45:52 [140376197571520] DEBUG - logStreamInfo(): 	Codec id: V_MPEG4/ISO/AVC
2021-01-08 17:45:52 [140376197571520] DEBUG - logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
2021-01-08 17:45:52 [140376197571520] DEBUG - logStreamInfo(): 	Track cpd: NULL
2021-01-08 17:45:52 [140376197571520] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2021-01-08 17:45:52 [140376197571520] INFO - writeHeaderCallback(): RequestId: 3ad16d20-ec58-4eba-bd9b-dff4c8892a45
2021-01-08 17:45:52 [140375635449600] DEBUG - describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.610098997649E9,"DataRetentionInHours":24,"DeviceName":null,"IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:eu-west-1:605273844167:alias/aws/kinesisvideo","MediaType":null,"Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:eu-west-1:605273844167:stream/my_kvs_test/1610098997649","StreamName":"my_kvs_test","Version":"l5vWDDynRA7rWroUeJsr"}}
2021-01-08 17:45:52 [140375635449600] INFO - describeStreamResultEvent(): Describe stream result event.
2021-01-08 17:45:52 [140375635449600] WARN - describeStreamResult(): Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
2021-01-08 17:45:52 [140375635449600] WARN - describeStreamResult(): Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
2021-01-08 17:45:52 [140375635449600] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
2021-01-08 17:45:52 [140375635449600] INFO - writeHeaderCallback(): RequestId: 174bd1e9-916f-4d13-a991-17e5a2aec769
2021-01-08 17:45:52 [140375627056896] DEBUG - getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-5ac677c0.kinesisvideo.eu-west-1.amazonaws.com"}
2021-01-08 17:45:52 [140375627056896] INFO - getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[... 13 lines cut ...]
Progress: (open) Opened Stream
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Progress: (request) Sending PLAY request
Progress: (request) Sending PLAY request
Progress: (request) Sent PLAY request
```

As you can see, there is an error at the beginning, but I get the same error with the USB camera and it works and it seems is sending the images to KVS, but I cannot see nothing into the Media Playback:
[image]

2) Using a docker image that AWS recommends:

I used the [docker image from aws](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-rtsp.html) and I wasn't abel to run the docker image:
`sudo docker run -it df7f4a384647 'AKIAY***' 'H31****' 'rtsp://*****:****@192.***.*.**/stream1' 'my_kvs_test'
`
And I got the following error: 
```
docker: Error response from daemon: OCI runtime create failed: container_linux.go:349: starting container process caused "exec: \"AKIAY***\": executable file not found in $PATH": unknown.
ERRO[0000] error waiting for container: context canceled 
```

So I decided to "log" into the docker image with `sudo docker run -it df7f4a384647 ` and try to run inside the docker image, and I got the same log as in step 1), but the results were the same.

Any suggestion for this case? I cannot see what I am doing wrong :-(. Thanks in advance.

disa6302, 2021-01-12:
@hdnh2006 , 

One way of debugging this is to use `fakesink` instead of KVS Sink. You first need to confirm if your pipeline is correct. If the pipeline does not work with fakesink, then there is an issue with your pipeline that needs to fixed. Enable `GST_DEBUG=4` to get more info. As such, we do not provide specific GStreamer support. Gstreamer forums would be your best resource.

hdnh2006, 2021-01-13:
Maybe the command was not right I used:
```
export AWS_ACCESS_KEY_ID=fakeid
export AWS_SECRET_ACCESS_KEY=evenmorefake
gst-launch-1.0 -v rtspsrc location=rtsp://user:pass@192.168.0.20/stream1 short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! h264parse ! kvssink stream-name=my_kvs_test storage-size=128 aws-region="eu-west-1"
```

And it worked!

closed 2021-01-13 by hdnh2006