awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #642: SDK version[3.0.0] Dropped Frame/Fragment[QUESTION]

derek1875, 2021-01-11:
Hi, I am testing 3.0.0 SDK and a few video/audio frames at the beginning of a stream may be dropped. could you please check this log?

- Here is what I did:
1. create a stream.
1. start sending video/audio frames via `putKinesisVideoFrame`. 
2. stop sending frames for a while.
3. goto step 2.

- Here are some logs:
>[KVS] stepStateMachine(): State Machine - Current state: 0x0000000000000200, Next state: 0x0000000000000040
[KVS] stepStateMachine(): State Machine - Current state: 0x0000000000000040, Next state: 0x0000000000000080
[KVS] putStreamResultEvent(): Put stream result event. New upload handle 36
[KVS] stepStateMachine(): State Machine - Current state: 0x0000000000000080, Next state: 0x0000000000000100
[KVS] logStreamMetric(): Kinesis Video client and stream metrics:
[KVS] logStreamMetric():    Overall storage byte size: 9437184 
[KVS] logStreamMetric():    Available storage byte size: 9270558 
[KVS] logStreamMetric():    Allocated storage byte size: 166626 
[KVS] logStreamMetric():    Total view allocation byte size: 374480 
[KVS] logStreamMetric():    Total streams frame rate (fps): 1136 
[KVS] logStreamMetric():    Total streams transfer rate (bps): 2566464 (2506 Kbps)
[KVS] logStreamMetric():    Current view duration (ms): 0 
[KVS] logStreamMetric():    Overall view duration (ms): 606 
[KVS] logStreamMetric():    Current view byte size: 0 
[KVS] logStreamMetric():    Overall view byte size: 96939 
[KVS] logStreamMetric():    Current frame rate (fps): 1136.901982 
[KVS] logStreamMetric():    Current transfer rate (bps): 2566464 (2506 Kbps)
[KVS] logStreamMetric():    Stream uptime in (ms): 4238733 
[KVS] logStreamMetric():    Total transferred bytes: 29948712 
[KVS] logStreamMetric():    Total number of streaming sessions: 37 
[KVS] logStreamMetric():    Total number of active streaming sessions: 31 
[KVS] logStreamMetric():    Average streaming sessions duration: 171981528 
[KVS] logStreamMetric():    Number of Buffered ACKs: 154 
[KVS] logStreamMetric():    Number of Received ACKs: 137 
[KVS] logStreamMetric():    Number of Persisted ACKs: 137 
[KVS] logStreamMetric():    Number of Error ACKs: 15 
[KVS] logStreamMetric():    Number of dropped frames: 23 
[KVS] logStreamMetric():    Number of skipped frames: 1130 
[KVS] logStreamMetric():    Number of storage pressure events: 0 
[KVS] logStreamMetric():    Number of latency pressure events: 0 
[KVS] logStreamMetric():    Number of buffer pressure events: 0 
[KVS] logStreamMetric():    Number of stream staleness events: 144 
[KVS] logStreamMetric():    Number of latency pressure events: 0 
[KVS] logStreamMetric():    Number of Put Frame errors: 0 
[KVS] logStreamMetric():    Average Control Plane API latency (ms): 226 
[KVS] logStreamMetric():    Average Data Plane API latency (ms): 0 
[KVS] viewItemRemoved(): Reporting a dropped frame/fragment.
[kvs_module.c:165][count:1]Reported droppedFrame callback for stream handle 8802264. Dropped frame timecode in 100ns: 16101088013380000[KVS] viewItemRemoved(): Reporting a dropped frame/fragment.
[KVS] contentViewTrimTailItems(): ContentView is not big enough to contain a single fragment.
[KVS] logStreamMetric(): Kinesis Video client and stream metrics:
[KVS] logStreamMetric():    Overall storage byte size: 9437184 
[KVS] logStreamMetric():    Available storage byte size: 9256885 
[KVS] logStreamMetric():    Allocated storage byte size: 180299 
[KVS] logStreamMetric():    Total view allocation byte size: 374480 
[KVS] logStreamMetric():    Total streams frame rate (fps): 3032 
[KVS] logStreamMetric():    Total streams transfer rate (bps): 2566464 (2506 Kbps)
[KVS] logStreamMetric():    Current view duration (ms): 1920 
[KVS] logStreamMetric():    Overall view duration (ms): 1960 
[KVS] logStreamMetric():    Current view byte size: 120858 
[KVS] logStreamMetric():    Overall view byte size: 123427 
[KVS] logStreamMetric():    Current frame rate (fps): 3032.171797 
[KVS] logStreamMetric():    Current transfer rate (bps): 2566464 (2506 Kbps)
[KVS] logStreamMetric():    Stream uptime in (ms): 4239270 
[KVS] logStreamMetric():    Total transferred bytes: 29948712 
[KVS] logStreamMetric():    Total number of streaming sessions: 37 
[KVS] logStreamMetric():    Total number of active streaming sessions: 31 
[KVS] logStreamMetric():    Average streaming sessions duration: 171981528 
[KVS] logStreamMetric():    Number of Buffered ACKs: 154 
[KVS] logStreamMetric():    Number of Received ACKs: 137 
[KVS] logStreamMetric():    Number of Persisted ACKs: 137 
[KVS] logStreamMetric():    Number of Error ACKs: 15 
[... 357 lines cut ...]
[KVS] postWriteCallback(): Curl post body write function for stream with handle: 8100da58d7bbd192e4d7f485a62ff34bd70224d285516b2e391f9efc48a8a6b8 and upload handle: 38 returned: {"EventType":"PERSISTED","FragmentTimecode":1
[KVS] postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 38
[KVS] postReadCallback(): Wrote 394 bytes to Kinesis Video. Upload stream handle: 38
[KVS] postReadCallback(): Pausing CURL read for upload handle: 38
[KVS] curlCompleteSync(): curl perform failed for url https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seco
[KVS] curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com/putMedia
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=ASIAQ7AHN6CQ4LDS3QWG/20210108/us-west-2/kinesisvideo/aws4_request, SignedHeaders=connection;host;transfer-encoding;user-agent;x-amz-date;x-amzn-fragment-acknowledgment-required
    connection: keep-alive
    content-type: application/json
    host: s-4010bf70.kinesisvideo.us-west-2.amazonaws.com
    transfer-encoding: chunked
    user-agent: AWS-SDK-KVS/3.0.0 GCC/4.7.2 Linux/3.10.14 mips
    X-Amz-Date: 20210108T122648Z
    x-amz-security-token: IQoJb3JpZ2luX2VjEJT//////////wEaCXVzLXdlc3QtMiJHMEUCIDa1ZKa/o3NjPDicD/sNopYTGIlqGxgbzdiSJQzSOOdKAiEAv6oT6M1Znv4o+c9jXbalMABM+nq0p/DTYcgkZyIN7Y4qvQQIXRACGgwwNjY1ODc1ODY3MjEiDEPpt1nugu3Mj7WIJyqaBCIhB
[KVS] putStreamCurlHandler(): Network thread for Kinesis Video stream: 8100da58d7bbd192e4d7f485a62ff34bd70224d285516b2e391f9efc48a8a6b8 with upload handle: 38 exited. http status: 0
[KVS] putStreamCurlHandler(): Stream with streamHandle 8802264 has exited without triggering end-of-stream. Service call result: 599
[KVS] kinesisVideoStreamTerminated(): Stream 0x864fd8 terminated upload handle 38 with service call result 599.
[KVS] stepStateMachine(): State Machine - Current state: 0x0000000000000100, Next state: 0x0000000000000200

- From the log, we could see `[KVS] contentViewTrimTailItems(): ContentView is not big enough to contain a single fragment`, what does this mean? is there anything wrong with video or audio operations? 

More information would be provided if you needed. 
Thanks.

disa6302, 2021-01-12:
@derek1875 , 

What I notice is that your network has a problem: 

`curlCompleteSync(): curl perform failed for url https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds`

Also, when you say `stop sending frames for a while`, do you call the stopStreamSync` or the application just does not send frames and sleeps?

Are there any setting deviations from the defaults?

derek1875, 2021-01-13:
Thanks disa6302.

I do not call the stopStreamSync, and my application just stop sending frames and sleep.

Here is my major configuration and pseudo streaming logic:

```
PStreamInfo pstreamInfo;
CLIENT_HANDLE clientHandle;
PDeviceInfo pDeviceInfo;
STREAM_HANDLE streamHandle;
PClientCallbacks pClientCallbacks;
PCHAR defaultRegion = (PCHAR) "us-west-2";
UINT64 storageSize = 9 * 1024 * 1024;

void kinesis_video_init()
{
    CHK_STATUS(createRealtimeAudioVideoStreamInfoProvider(streamName,
                                                              2 * HUNDREDS_OF_NANOS_IN_AN_HOUR,
                                                              60 * HUNDREDS_OF_NANOS_IN_A_SECOND,
                                                              &pstreamInfo));
    for (i = 0; i < pstreamInfo->streamCaps.trackInfoCount && pstreamInfo->streamCaps.trackInfoList[i].trackType != MKV_TRACK_INFO_TYPE_AUDIO; ++i);
    CHK(i < pstreamInfo->streamCaps.trackInfoCount, STATUS_TRACK_INFO_MISSING);
    pTrackInfoAudio = pstreamInfo->streamCaps.trackInfoList + i;
    STRNCPY(pTrackInfoAudio->codecId, (PCHAR) "A_MS/ACM", MKV_MAX_CODEC_ID_LEN);
    pTrackInfoAudio->codecPrivateData = audioCpd;
    pTrackInfoAudio->codecPrivateDataSize = SIZEOF(audioCpd);
     
    STRNCPY(pstreamInfo->streamCaps.contentType, (PCHAR) "video/h264,audio/alaw", MAX_CONTENT_TYPE_LEN);
    pstreamInfo->streamCaps.frameRate = 130; 
    pstreamInfo->streamCaps.nalAdaptationFlags = NAL_ADAPTATION_ANNEXB_NALS | NAL_ADAPTATION_ANNEXB_CPD_NALS;
    pstreamInfo->streamCaps.absoluteFragmentTimes = TRUE;

    CHK_STATUS(createDefaultDeviceInfo(&pDeviceInfo));
    CHK_STATUS(setDeviceInfoStorageSize(pDeviceInfo, storageSize));
    pDeviceInfo->clientInfo.loggerLogLevel = LOG_LEVEL_ERROR;//LOG_LEVEL_VERBOSE;
    CHK_STATUS(createDefaultCallbacksProviderWithIotCertificate(endPoint,
                                                                        IOT_CERT_PATH,
                                                                        IOT_PRIVATE_KEY_PATH,
                                                                        IOT_CA_CERT_PATH,
                                                                        roleAlias,
                                                                        thingName,
                                                                        defaultRegion,
                                                                        NULL, NULL, &pClientCallbacks));

    CHK_STATUS(createStreamCallbacks(&pStreamCallbacks));
    pStreamCallbacks->customData = (UINT64)NULL;
    pStreamCallbacks->bufferDurationOverflowPressureFn = NULL;
    pStreamCallbacks->streamLatencyPressureFn = NULL;
    pStreamCallbacks->streamConnectionStaleFn = NULL;
    pStreamCallbacks->droppedFrameReportFn = defaultDroppedFrameReportCallback;
    pStreamCallbacks->droppedFragmentReportFn = defaultDroppedFragmentReportCallback;
    pStreamCallbacks->streamErrorReportFn = defaultStreamErrorReportCallback;
    pStreamCallbacks->fragmentAckReceivedFn = defaultFragmentAckReceivedCallback;
    pStreamCallbacks->streamReadyFn = NULL;
    pStreamCallbacks->streamClosedFn = NULL;
    pStreamCallbacks->streamShutdownFn = NULL;
    pStreamCallbacks->freeStreamCallbacksFn = NULL;

    CHK_STATUS(addStreamCallbacks(pClientCallbacks, pStreamCallbacks));

    PlatformCallbacks platformCallbacks = {0};
    platformCallbacks.logPrintFn = myLogging;
    CHK_STATUS(setPlatformCallbacks(pClientCallbacks, &platformCallbacks));
    CHK_STATUS(createKinesisVideoClient(pDeviceInfo, pClientCallbacks, &clientHandle));
    CHK_STATUS(freeDeviceInfo(&pDeviceInfo));

    createKinesisVideoStream(clientHandle, pstreamInfo, &streamHandle);
}

int main()
{
    kinesis_video_init();
    
    while(1) {
      //video fps is 15, so actually we are pushing a 12s video. 
      for(int i=0; i < 180; i++) {
        get_frame(frame);
        putKinesisVideoFrame(streamHandle, frame);
      }
      sleep(30);
    }
}
```

On my device it works well at most of the time but the question is a few video/audio frames **at the beginning of a video may be dropped**. and what I confirmed is that the network is not in a trouble. 

So how should I do to avoid `[KVS] contentViewTrimTailItems(): ContentView is not big enough to contain a single fragment`. Thanks a lot.

MushMal, 2021-01-13:
RE: [KVS] contentViewTrimTailItems(): ContentView is not big enough to contain a single fragment

This is a warning with the default drop frame policy. What it means is that if the content view needs to drop a frame for any reason (due to temporal pressures like in your case where you have an intermittent producer case) it will not only drop a single frame but will continue dropping all of the frames of the same fragment. This makes sense for video as dropping for example an I-frame only will lead to semantically invalid fragment so the attempt is made to remove all of the P-frames after the frame that's being dropped. This is a warning.

What happens in your scenario is that you have an "unenlightened" intermittent producer. As you simply pause and don't put frames, the session times out. Next time you produce a frame, the state machine will attempt to re-stream the last fragment as it hasn't been ACK-ed and will end up in a situation of max latency breach which is a latency pressure. The frame will be dropped and this warning will be caused.

This commit should fix the issue with the automatic intermittent producer scenario: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/pull/640

Please update to this and give it a try.

Capture verbose logs and attach as a file to the issue if you happen to have any problems. Please provide detailed description of the issue with the detailed description of your scenario and settings and whether you've modified any code.

derek1875, 2021-01-18:
@MushMal  Thanks for your answer.  I'm doing test on the commit you recommended and will attached logs once the problem happened.

disa6302, 2021-01-18:
@derek1875 , 

I am closing this ticket. If you face any issues with intermittent streaming, feel free to open a new issue with the question.

closed 2021-01-18 by disa6302