awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #644: [QUESTION]"Invalid heap" logging in the middle of frame upload

Subburaj47, 2021-01-18:
During the upload of frames, there was error log seen mentioning heaps and subsequently the frame uploads failed and stopped working. Based on our application logic, we restart the application when there is no fragment ack received for quite some time. 
Can you pls help in explaining the following logs and if that causes any adverse issues?

** Logging **
`[ERROR][2021-01-17 23:56:35] aivHeapMap(): Invalid handle or previously freed.
[ERROR][2021-01-17 23:56:35] getKinesisVideoStreamData(): operation returned status code: 0x00000008
[ERROR][2021-01-17 23:56:35] postReadCallback(): Failed to get data from the stream with an error: 0x00000008
[ERROR][2021-01-17 23:56:35] kinesisVideoStreamTerminated(): operation returned status code: 0x52000037`

** If you would not like to open an issue to discuss your solution in open-platform, please email your question to kinesis-video-support@amazon.com **

MushMal, 2021-01-19:
0x00000008 is the STATUS_INVALID_HANDLE_ERROR. It's not clear what's causing this issue - the only indication is that either get size or map API is called with an invalid handle.

Can you please provide more details? This is the first time I am getting this. Are you able to reproduce this on a stock samples? Are you using non-default StorageInfo flags?

Subburaj47, 2021-01-20:
@MushMal We observed this issue once and we don't see any particular repro steps for this. We are specifying 3MB for storage.

Just to understand better, can you please provide more info on this  -- **get size or map API is called with an invalid handle**

MushMal, 2021-01-20:
@Subburaj47 the SDK is designed in a layered fashion. Heap component has a public interfaces which the PIC Client object is utilizing when using heap as a Content Store. The output you've logged comes only from two public APIs. Both are used during streaming so I can't know what could cause it.

The most likely answer is that either there is a memory corruption or an allocation in the application failed (not the SDK as SDK has result checking in every single API call).

I am going to resolve this issue. Please re-open or cut a new issue once you have detailed information with verbose logs.

P.S. We have never seen this type of issues before in the field.

closed 2021-01-20 by MushMal