awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #645: [QUESTION] keep credentials in credential-path fresh

wlinInspire, 2021-01-20:
The `credential-path` example in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/477 is super helpful! Questions:
* Can I have a corn job that periodically refreshes credentials in `credential-path` file? 
* Will the gstreamer plugin automatically fetch the new credentials in the `credential-path` file when the old one is about to expire?

MushMal, 2021-01-20:
Not sure what you mean. Your application will be the one to periodically supply fresh credentials in the file. This is how we can dynamically supply fresh credentials to the SDK. We don't have an example of code that refreshes the file but it's not very hard to create.

Alternatively, as described in the linked issue, you could simply use IoT credential provider which handles the fetching of the credentinals.

wlinInspire, 2021-01-20:
@MushMal Yes. I have a separate program (controlled by a cron job) dedicated to periodically supply fresh credentials to the file. Then, If I start streaming using gstreamer kvssink plugin with `credential-path` specified to be this file, the streaming will keep running as long as the credentials in the file get updated periodically, right?

MushMal, 2021-01-20:
Yep, just need to ensure the expiration is set correctly

hassanctech, 2021-01-27:
@wlinInspire Were you able to get this to work?  If your problem was solved please go ahead and resolve this issue.

wlinInspire, 2021-01-27:
@hassanctech @MushMal It works. Thanks for the help

closed 2021-01-27 by wlinInspire