awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #646: [QUESTION] How do I build on Mac OS without building dependencies?

MixMasterMitch, 2021-01-21:
I am trying to build the SDK and example on MacOS 10.15.7 without building the dependencies, but have run into issues.

I have run the following:
```
brew install pkg-config openssl cmake gstreamer gst-plugins-base gst-plugins-good gst-plugins-bad gst-plugins-ugly log4cplus gst-libav
```

```
cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_DEPENDENCIES=OFF
```
^ this `cmake` fails with the following error:
```
...
-- Found PkgConfig: /usr/local/bin/pkg-config (found version "0.29.2")
-- Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the system variable OPENSSL_ROOT_DIR (missing: OPENSSL_INCLUDE_DIR)
CMake Error at dependency/libkvscproducer/kvscproducer-src/CMakeLists.txt:118 (message):
  OpenSSL is not found.  Make sure to export PKG_CONFIG_PATH to where
  OpenSSL's pc file is

-- Configuring incomplete, errors occurred!
See also "/Users/mrl/workspaces/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeOutput.log".
```

So I ran the following command which I found in the console output when brew installed openssl (note that I am using the fish shell):

```
set -gx PKG_CONFIG_PATH "/usr/local/opt/openssl@1.1/lib/pkgconfig"
```

I was then able to successfully re-run `cmake`. **I think that the documentation or `cmake` configuration could use some enhancement to help avoid this issue.**

When I then ran `make`, I got the following error and I am not sure how to get around it:

```
...
[ 79%] Linking C shared library libcproducer.dylib
ld: cannot link directly with dylib/framework, your binary is not an allowed client of /usr/lib/libcrypto.dylib for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [dependency/libkvscproducer/kvscproducer-src/libcproducer.dylib] Error 1
make[1]: *** [dependency/libkvscproducer/kvscproducer-src/CMakeFiles/cproducer.dir/all] Error 2
make: *** [all] Error 2
```

**How can I resolve this issue? And what can be done to prevent others from running into this in the future?**

hassanctech, 2021-01-26:
This might help:  https://developer.apple.com/forums/thread/124782

It seems Apple added his recently and they intend for libraries like libcrypto.dylib that they've built to be used by them only.  With 10.15 and greater that library is no longer a symlink like it used to be rather some sort of stub/wrapper library.  To solve your problem it looks like you will not be able to just use the system libcrypto rather you'll need to `brew install openssl` and then make sure to add that to link to the newly installed openssl.  Make sure to wipe out your old cmake cache and start with a fresh cmake run after you've installed openssl.

MushMal, 2021-01-27:
Resolving.

closed 2021-01-27 by MushMal