awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #650: [QUESTION]

terraeca, 2021-02-04:
I was wondering if there is a way for ./kvs_gstreamer_sample to exit when the stream gets deleted on AWS, any help would be appreciated.

hassanctech, 2021-02-09:
This is a sample so it doesn't have all of those checks in place, we have an init step which creates the stream and then we stream to it assuming it exists.  

What behavior do you see when the stream is deleted, my suspicion is that we will just re-create the stream and keep going, but want to see what you're seeing?  The GStreamer plugin was really designed for continuous / never stop streaming and auto retry failures.

Basically you want to call `g_main_loop_quit` and currently this happens in the gstreamer bus error callback, when end of stream is received, or when a client error callback is invoked, i.e.

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/samples/kvs_gstreamer_sample.cpp#L465
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/samples/kvs_gstreamer_sample.cpp#L311
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/samples/kvs_gstreamer_sample.cpp#L263

If the stream is deleted then put media will fail, you can #define a set of error codes that you want to immediately terminate on (currently IS_RETRIABLE_ERROR and IS_RECOVERABLE_ERROR will currently retry).

So define your own case for failures and add this single one:  STATUS_SERVICE_CALL_RESOURCE_NOT_FOUND_ERROR

Then when that is returned then set `terminate_pipeline` to true.  And that's it now it should terminate when the stream is deleted.

hassanctech, 2021-02-16:
Closing because the question has been answered.  If you need more help please re-open or if you have a new question please open a new issue.

closed 2021-02-16 by hassanctech