awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #652: [QUESTION] How long can the KVS connection stays idle?

crcsylvia, 2021-02-05:
Want kinesis producer and stream stay ready and keep idle until we start to put frame in stream.

** Any design considerations/constraints **
We want to start to upload recording to KVS when some specific events happened.
We plan to have KVS producer & stream ready first and then waiting for frames.
Does KVS producer or stream have connection time limits? How long can we keep the connection?

SDK version: v3.0.0

Thank you!

disa6302, 2021-02-05:
@crcsylvia , 

If I understand correctly, you are looking for an intermittent producer scenario use case ---- stream for some time, stop and then stream again. With CPP SDK 3.1.0, this feature is available. 

KVS producer does not have connection time limits per say. With KVS producer, a streaming session can run for a maximum of 40 minutes after which the streaming continues on another session after token rotation.

hassanctech, 2021-02-10:
@crcsylvia Please try the latest commit from master:  d3ba97b985fe65bd10cec6960df83862c142eaa4.  There was a bug in the default configuration in the v3.1.0 release which impacts multi track streams.  If you have only a single track then you should be OK with v3.1.0.

To expand on what @disa6302 stated, the 40 minute time limit for token rotation so behind the scenes in the SDK after getting the next token we start the new streaming session in parallel with the old one which is about to expire and we seamlessly switch between the two, so **from the user perspective there is no time limit you can keep streaming uninterrupted forever**.

hassanctech, 2021-02-16:
Closing because the question has been answered.  If you have a new question, please open a new issue.

closed 2021-02-16 by hassanctech