awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #653: [QUESTION]Try to get the time of the video clip uploaded to the server

wangziheng437, 2021-02-05:
Try to get the time of the video clip uploaded to the server

I used the v1.0 version before. By registering a callback to `fragmentAckReceivedFn`, I can receive the callback when every 2s of video clips successfully uploaded.
But when I update to version 3.0, this callback looks out of time. Here are some logs(Please note that "send ok" is the upper code triggered by the callback):

** Logging **
[2021-02-05 19:35:13] ***********************************************************************
[2021-02-05 19:35:13] thing_name :     2caa8eb88d8d
[2021-02-05 19:35:13] streamName :     9ca012f850d35bcd983c58a949a505bf3cd1cfc3fb0829ebac9aeffc8127747b
[2021-02-05 19:35:13] roleAlias  :     alias_iam_wyze_kvs_role
[2021-02-05 19:35:13] endPoint   :     c1ybkrkbr1j10x.credentials.iot.us-west-2.amazonaws.com
[2021-02-05 19:35:13] >>>>>>>>>>>>>>>0<<<<<<<<<<<<<<***********************************************************************
[2021-02-05 19:35:15] 2021-02-05 11:35:13 INFO    heapInitialize(): Initializing native heap with limit size 9437184, spill ratio 0% and flags 0x00000001
[2021-02-05 19:35:15] 2021-02-05 11:35:13 INFO    heapInitialize(): Creating AIV heap.
[2021-02-05 19:35:15] 2021-02-05 11:35:13 INFO    heapInitialize(): Heap is initialized OK
[2021-02-05 19:35:15] 2021-02-05 11:35:13 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[2021-02-05 19:35:15] 2021-02-05 11:35:13 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[2021-02-05 19:35:15] 2021-02-05 11:35:13 INFO    createDeviceResultEvent(): Create device result event.
[2021-02-05 19:35:15] 2021-02-05 11:35:13 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[2021-02-05 19:35:15] 2021-02-05 11:35:14 INFO    createKinesisVideoStream(): Creating Kinesis Video Stream.
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo(): Kinesis Video Stream Info
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Stream name: 9ca012f850d35bcd983c58a949a505bf3cd1cfc3fb0829ebac9aeffc8127747b 
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Streaming type: STREAMING_TYPE_REALTIME 
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Content type: video/h264,audio/alaw 
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Max latency (100ns): 510000000
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Fragment duration (100ns): 20000000
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Key frame fragmentation: Yes
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Use frame timecode: Yes
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Absolute frame timecode: Yes
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Nal adaptation flags: 40
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Average bandwith (bps): 2097152
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Framerate: 130
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Buffer duration (100ns): 600000000
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Replay duration (100ns): 300000000
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Connection Staleness duration (100ns): 50000000
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Store Pressure Policy: 1
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      View Overflow Policy: 1
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Segment UUID: NULL
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Frame ordering mode: 3
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo(): Track list
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Track id: 1
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Track name: kvs_video_track
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Codec id: V_MPEG4/ISO/AVC
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Track type: TRACK_INFO_TYPE_VIDEO
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Track cpd: NULL
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Track id: 2
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Track name: kvs_audio_track
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Codec id: A_MS/ACM
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Track type: TRACK_INFO_TYPE_AUDIO
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   logStreamInfo():      Track cpd: 06000100401F0000803E0000020010000000
[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[2021-02-05 19:35:16] 2021-02-05 11:35:14 DEBUG   logStreamMetric(): Kinesis Video client and stream metrics:
[2021-02-05 19:35:16] 2021-02-05 11:35:14 DEBUG   logStreamMetric():    Overall storage byte size: 9437184 
[2021-02-05 19:35:16] 2021-02-05 11:35:14 DEBUG   logStreamMetric():    Available storage byte size: 9383310 
[2021-02-05 19:35:16] 2021-02-05 11:35:14 DEBUG   logStreamMetric():    Allocated storage byte size: 53874 
[2021-02-05 19:35:16] 2021-02-05 11:35:14 DEBUG   logStreamMetric():    Total view allocation byte size: 374480 
[2021-02-05 19:35:16] 2021-02-05 11:35:14 DEBUG   logStreamMetric():    Total streams frame rate (fps): 130 
[2021-02-05 19:35:16] 2021-02-05 11:35:14 DEBUG   logStreamMetric():    Total streams transfer rate (bps): 16777216 (16384 Kbps)
[2021-02-05 19:35:16] 2021-02-05 11:35:14 DEBUG   logStreamMetric():    Current view duration (ms): 0 
[2021-02-05 19:35:16] 2021-02-05 11:35:14 DEBUG   logStreamMetric():    Overall view duration (ms): 0 
[2021-02-05 19:35:16] 2021-02-05 11:35:14 DEBUG   logStreamMetric():    Current view byte size: 0 
[2021-02-05 19:35:16] 2021-02-05 11:35:14 DEBUG   logStreamMetric():    Overall view byte size: 0 
[2021-02-05 19:35:16] 2021-02-05 11:35:14 DEBUG   logStreamMetric():    Current frame rate (fps): 130.000000 
[2021-02-05 19:35:16] 2021-02-05 11:35:14 DEBUG   logStreamMetric():    Current transfer rate (bps): 16777216 (16384 Kbps)
[2021-02-05 19:35:16] 2021-02-05 11:35:14 DEBUG   logStreamMetric():    Stream uptime in (ms): 82 
[2021-02-05 19:35:16] 2021-02-05 11:35:14 DEBUG   logStreamMetric():    Total transferred bytes: 0 
[2021-02-05 19:35:16] 2021-02-05 11:35:14 DEBUG   logStreamMetric():    Total number of streaming sessions: 0 
[2021-02-05 19:35:16] 2021-02-05 11:35:14 DEBUG   logStreamMetric():    Total number of active streaming sessions: 0 
[2021-02-05 19:35:16] 2021-02-05 11:35:14 DEBUG   logStreamMetric():    Average streaming sessions duration: 0 
[2021-02-05 19:35:16] 2021-02-05 11:35:14 DEBUG   logStreamMetric():    Number of Buffered ACKs: 0 
[... 321 lines cut ...]
[2021-02-05 19:35:25] 2021-02-05 11:35:24 DEBUG   logStreamMetric():    Number of Error ACKs: 0 
[2021-02-05 19:35:25] 2021-02-05 11:35:24 DEBUG   logStreamMetric():    Number of dropped frames: 0 
[2021-02-05 19:35:25] 2021-02-05 11:35:24 DEBUG   logStreamMetric():    Number of skipped frames: 0 
[2021-02-05 19:35:25] 2021-02-05 11:35:24 DEBUG   logStreamMetric():    Number of storage pressure events: 0 
[2021-02-05 19:35:25] 2021-02-05 11:35:24 DEBUG   logStreamMetric():    Number of latency pressure events: 0 
[2021-02-05 19:35:25] 2021-02-05 11:35:24 DEBUG   logStreamMetric():    Number of buffer pressure events: 0 
[2021-02-05 19:35:25] 2021-02-05 11:35:24 DEBUG   logStreamMetric():    Number of stream staleness events: 1 
[2021-02-05 19:35:25] 2021-02-05 11:35:24 DEBUG   logStreamMetric():    Number of latency pressure events: 0 
[2021-02-05 19:35:25] 2021-02-05 11:35:24 DEBUG   logStreamMetric():    Number of Put Frame errors: 0 
[2021-02-05 19:35:25] 2021-02-05 11:35:24 DEBUG   logStreamMetric():    Average Control Plane API latency (ms): 122 
[2021-02-05 19:35:25] 2021-02-05 11:35:24 DEBUG   logStreamMetric():    Average Data Plane API latency (ms): 0 
[... cut ...]
[2021-02-05 19:35:26] 2021-02-05 11:35:24 DEBUG   postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 1
[2021-02-05 19:35:26] 2021-02-05 11:35:24 DEBUG   postWriteCallback(): Curl post body write function for stream with handle: 9ca012f850d35bcd983c58a949a505bf3cd1cfc3fb0829ebac9aeffc8127747b and upload handle: 1 returned: {"EventType":"PERSISTED","FragmentTimecode":1612524911136,"FragmentNumber":"91343852333181580945}
[2021-02-05 19:35:26] 
[2021-02-05 19:35:26] kvs_module.c,247:[-kvs-]---------------send ok------------------2

MushMal, 2021-02-09:
Information on the ACKs is available here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/structures.md#fragment-acks

There are multiple types of ACKs and they are sent at certain events so you can check the type of ACK you are getting. Basically, the buffering ACK is sent as soon as the new fragment is started getting parsed. The received ACK is sent as soon as the fragment is "closed" and the persistence and indexing has just started. The persisted ACK is sent after the fragment has been persisted. It takes certain period of time for the persistence and due to highly asynchronous nature of the operations, there is no immediate guarantee that it will be performed in real-time. 

Looking at your log I can see the following entry:

`2021-02-05 19:35:22] 2021-02-05 11:35:20 WARN streamFragmentAckEvent(): An ACK is received for an already expired upload handle 0
`
This happens after you get an ACK on a connection that was deemed to be stale so the continuous retry callback has re-set the connection. From your log I can see

`[2021-02-05 19:35:15] 2021-02-05 11:35:14 DEBUG logStreamInfo(): Connection Staleness duration (100ns): 50000000
`

This means that you are going to get your connection reset after 5 seconds of not getting any feedback from the service. This might be too aggressive in your network case as you will get your connection reset every time there is a perceived staleness event. Try changing it to a larger value.

hassanctech, 2021-02-16:
Closing because this question has been answered.  If you still need help please re-open, or if you have a new question please open a new issue.

closed 2021-02-16 by hassanctech