awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #655: SDK version[3.1.0] Dropped Frame/Fragment[QUESTION]

derek1875, 2021-02-09:
Hi, I happen to see a log that the Frames/Fragments were being dropped for a while (actually I losed more than 10s video on my APP). This is related to #651. (not sure it's the same issue) 

Attached is the verbose log.
[kvs_3.1.0_debug.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/5948428/kvs_3.1.0_debug.txt)

The log shows ```viewItemRemoved(): Reporting a dropped frame/fragment```.  I'm debugging the error code(0x30000005), it should be a new issue and may not be related to the frame/fragment dropping issue. So please help confirm the log because I'm going to release this version if there is no obvious issue.

hassanctech, 2021-02-09:
In your case of streaming and stopping, do you ever start streaming from a non-key frame?  Since a fragment cannot start from a non key frame we automatically drop all frames before the first key frame encountered, you should not be losing 10s+ video all at once though since fragments should be smaller than that.  I assume you're using key-frame fragmentation, can you tell me at what interval you're issuing key frames (and marking them as key frames)?

Another thing that can cause this is if network is bad and the buffer filled up then we drop frames from the tail.

hassanctech, 2021-02-09:
Also I see a lot of these in your logs:

Feb  9 06:22:05 kvs_stream: [KVS] curlCompleteSync(): curl perform failed for url https://s-b1731d75.kinesisvideo.us-west-2.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seco

It looks like network is bad.  Also I see:

Feb  9 06:32:27 kvs_stream: [KVS] streamLatencyStateMachineHandleStreamLatency(): Stream Latency State Machine starting from NORMAL_STATE
Feb  9 06:32:27 kvs_stream: [KVS] streamLatencyStateMachineToResetConnectionState(): Stream Latency State Machine move to RESET_CONNECTION_STATE

There is latency pressure (based on the max buffer duration chosen) it's filling up so frames might be dropped. 

Can you please try your scenario with a good network, you should not see any of these issues.

hassanctech, 2021-02-09:
Also I saw this:

Feb  9 06:26:45 kvs_stream: [KVS] postWriteCallback(): Curl post body write function for stream with handle: 61af4553e45a2ea664f28d460b717ef303527381c0d5b6b9bf1adcbba11cb642 and upload handle: 16 returned: {"EventType":"ERROR","FragmentTimecode":16128
Feb  9 06:26:46 kvs_stream: K: {"EventType":"ERROR","FragmentTimecode":1612805142238,"FragmentNumber":"91343852333204527461476092574950081151709692942","ErrorCode":"FRAMES_MISSING_FOR_TRACK","ErrorId":4011}
Feb  9 06:26:46 kvs_stream:  with status code: 0x52000047

If you are streaming multi track (which you are, I see a video and audio track) then when a video (or audio) frame is marked as a key frame we will close out the fragment.  If at that time *all* tracks do not have at least 1 frame, it will throw this error.  What must be happening here is either your audio or video track has no frames, this is not allowed.

derek1875, 2021-02-09:
> do you ever start streaming from a non-key frame?

Every streaming started with a key frame.

>can you tell me at what interval you're issuing key frames (and marking them as key frames)

The key frame interval is 2s.

Of course I will do more tests on various scenario to avoid the influence of the poor network.
Also I will update curl and ssl version next time, the version I'm using are too old.

derek1875, 2021-02-09:
> If you are streaming multi track (which you are, I see a video and audio track) then when a video (or audio) frame is marked as a key frame we will close out the fragment. If at that time all tracks [...]

Thanks, I will check on this issue, maybe there is a mistakes. But I'm only marked the video as a key frame(as the audio flags is none ).

MushMal, 2021-02-09:
To point out that not every video frame should be a key frame. In your case you should set the key frame flag only for the I-frames

derek1875, 2021-02-09:
>To point out that not every video frame should be a key frame. In your case you should set the key frame flag only for the I-frames

Yes, this is what I did. Most of our device(>>100K) are using KVS-2.x.x version, it also works good. Now it's time to update a 3.x.x version. -:)

MushMal, 2021-02-09:
This should be in-place upgrade as we try to preserve the compatibility as much as possible when it comes to behavior and we preserve the backward compatibility always. We even version the public structures. In your case, with the update, use the “current” versions defined in the public headers. Other than that, you will benefit from the automatic intermittent producer handling and many of the improvements that went in. There should be no integ problems and if you do have any issues please let us know

hassanctech, 2021-02-15:
@derek1875 Is there still an issue here or can we close this?

derek1875, 2021-02-16:
Sorry for the late response. I'm closing this until I get more details.

closed 2021-02-16 by derek1875

Tangyun715, 2023-12-27:
> "If you are streaming multi track (which you are, I see a video and audio track) then when a video (or audio) frame is marked as a key frame we will close out the fragment. If at that time _all_ [...]

What if : sometimes two I-frame is very close,and the interval of them is 100ms. At that case there is no time for audio frame to insert between them. (in our producte,the AAC audio frame's interval is 120ms) .  I found  error 4011 when two I-frame is very close.