awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #659: [QUESTION] SDK kvs_gstreamer_sample.cpp C++ Producer Timeout results in Freezing

bml1g12, 2021-02-12:
When running kvs_gstreamer_sample.cpp from latest master branch from an RTSP stream, it works for several hours but after some number of hours it gets a timeout, followed by a state change ( Current state: 0x0000000000000100, Next state: 0x0000000000000200) but then seems to freeze with no errors I can see, and no CPU usage indefinitely.

> [WARN ] [11-02-2021 00:41:31:726.078 JST] curlCompleteSync(): curl perform failed for url https://s-a7395519.kinesisvideo.us-east-1.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds                     

Based on the warning here, I presume for at least 30 seconds there was too little upload bandwidth. 

My goal is to have the stream robustly attempting to reconnect, even if there are network issues or issues with the source RTSP stream. As I am running it in a container that will restart dead processes; even if the software were to exit in this situation it would be more reliable than freezing indefinitely without an error code, as I could restart it. 

 Do you have any recommendations on how to debug or fix this issue? Thank you!

`top` shows 0% CPU usage.:
```
    1 root      20   0   18384      4      4 S   0.0  0.0   0:00.01 kinesis.sh                                                                                                                                                              
   11 root      20   0 1235280  43632  12468 S   0.0  0.3   5:19.93 kvs_gstreamer_s                                                                                                                                                         
  105 root      20   0   18516   3380   3012 S   0.0  0.0   0:00.00 bash                                                                                                                                                                    
  122 root      20   0   38748   3252   2748 R   0.0  0.0   0:00.00 top 
```
The kinesis.sh is just 
```
AWS_DEFAULT_REGION=$AWS_REGION ./kvs_gstreamer_sample $AWS_VIDEO_STREAM_NAME YOUR_RTSP_STREAM
```
I am running inside a docker container (as part of microk8s kubernetes deployment on Ubuntu 18). Restarting the container restores the stream instantly suggesting there is no persistent problem with the RTSP stream or network preventing the producer from resuming. 

** Logging **

Below is an excerpt of last few hundred lines of the log:
```
[DEBUG] [11-02-2021 00:40:29:628.131 JST] Reporting fragment ack received. Ack timecode 1612971627372                                                                                                                             [112/1870]
[DEBUG] [11-02-2021 00:40:29:628.158 JST] postReadCallback(): Wrote 478 bytes to Kinesis Video. Upload stream handle: 5                                                                                                                     
[DEBUG] [11-02-2021 00:40:29:628.228 JST] postReadCallback(): Pausing CURL read for upload handle: 5
[DEBUG] [11-02-2021 00:40:29:715.522 JST] postWriteCallback(): Curl post body write function for stream with handle: Facility_EntranceInner and upload handle: 5 returned: {"EventType":"PERSISTED","FragmentTimecode":1612971626305,"Frag
mentNumber":"91343852333182359466170442827609771867793590848"}              
                                                                                                     
[DEBUG] [11-02-2021 00:40:29:715.579 JST] fragmentAckReceivedHandler invoked                                            
[DEBUG] [11-02-2021 00:40:29:715.590 JST] Reporting fragment ack received. Ack timecode 1612971626305
[DEBUG] [11-02-2021 00:40:29:715.604 JST] postReadCallback(): Wrote 963 bytes to Kinesis Video. Upload stream handle: 5                                                                                                                     
[DEBUG] [11-02-2021 00:40:29:715.654 JST] postReadCallback(): Pausing CURL read for upload handle: 5
[DEBUG] [11-02-2021 00:40:30:054.407 JST] Kinesis Video client and stream metrics
        >> Overall storage byte size: 134217728                                                                          
        >> Available storage byte size: 134154919                                                    
        >> Allocated storage byte size: 62809                                                                                                                                                                                                       >> Total view allocation byte size: 144080            
        >> Total streams frame rate (fps): 29
        >> Total streams transfer rate (bps): 296384 (289 Kbps)             
        >> Current view duration (ms): 334                                                           
        >> Overall view duration (ms): 1068                                                                                                                                                                                                 
        >> Current view byte size: 27363                                                            
        >> Overall view byte size: 61225                                         
        >> Current frame rate (fps): 29.874                                 
        >> Current transfer rate (bps): 296384 (289 Kbps)                                            
[DEBUG] [11-02-2021 00:40:30:716.726 JST] postReadCallback(): Wrote 37593 bytes to Kinesis Video. Upload stream handle: 5                                                                                                                   
[DEBUG] [11-02-2021 00:40:30:716.871 JST] postReadCallback(): Pausing CURL read for upload handle: 5
[DEBUG] [11-02-2021 00:40:31:059.497 JST] postWriteCallback(): Curl post body write function for stream with handle: Facility_EntranceInner and upload handle: 5 returned: {"EventType":"RECEIVED","FragmentTimecode":1612971627372,"Fragm
entNumber":"91343852333182359471122202984751293371972860615"}               
                                                                                                     
[DEBUG] [11-02-2021 00:40:31:059.547 JST] fragmentAckReceivedHandler invoked                        
[DEBUG] [11-02-2021 00:40:31:059.557 JST] Reporting fragment ack received. Ack timecode 1612971627372                 
[DEBUG] [11-02-2021 00:40:31:059.574 JST] postReadCallback(): Wrote 5146 bytes to Kinesis Video. Upload stream handle: 5
[DEBUG] [11-02-2021 00:40:31:059.629 JST] postReadCallback(): Pausing CURL read for upload handle: 5                                                                                                                                        
[DEBUG] [11-02-2021 00:40:31:065.088 JST] postWriteCallback(): Curl post body write function for stream with handle: Facility_EntranceInner and upload handle: 5 returned: {"EventType":"BUFFERING","FragmentTimecode":1612971628440,"Frag
mentNumber":"91343852333182359476073963141892814858656722113"}                                                           
                                                                                                    
[DEBUG] [11-02-2021 00:40:31:065.130 JST] fragmentAckReceivedHandler invoked                                                                                                                                                                
[DEBUG] [11-02-2021 00:40:31:065.139 JST] Reporting fragment ack received. Ack timecode 1612971628440                                                                                                                                       
[DEBUG] [11-02-2021 00:40:31:122.153 JST] Kinesis Video client and stream metrics
        >> Overall storage byte size: 134217728                             
        >> Available storage byte size: 134114874                                                    
        >> Allocated storage byte size: 102854                                                                          
        >> Total view allocation byte size: 144080                                                  
        >> Total streams frame rate (fps): 29                                                                                                                                                                                                       >> Total streams transfer rate (bps): 286976 (280 Kbps)
[... cut ...]
[WARN ] [11-02-2021 00:41:31:726.335 JST] putStreamCurlHandler(): Stream with streamHandle 94151997124024 uploadHandle 5 has exited without triggering end-of-stream. Service call result: 599                                              
[INFO ] [11-02-2021 00:41:31:726.347 JST] kinesisVideoStreamTerminated(): Stream 0x55a17857edb8 terminated upload handle 5 with service call result 599.
[DEBUG] [11-02-2021 00:41:31:726.358 JST] stepStateMachine(): State Machine - Current state: 0x0000000000000100, Next state: 0x0000000000000200

```

MushMal, 2021-02-12:
Can you ensure your rtsp source is still producing frames? Enable verbose logs to see whether KVS is actually getting any frames

bml1g12, 2021-02-12:
> Can you ensure your rtsp source is still producing frames? Enable verbose logs to see whether KVS is actually getting any frames

Thank you for your quick response. I will enable verbose logging and re-post when I get the timeout later today or on Monday.

If it was the case that it's stopped receiving frames - I wonder what we could do to ensure it recovers? I know that restaring the Producer SDK can recover it; so I guess would need it to get re-request the RTSP feed in such a case.

bml1g12, 2021-02-12:
```
log4cplus:ERROR Unrecognized log level: VERBOSE
```
```
root@thri-kinesis-deployment-entranceinner-779c79dc96-c9w2m:/src/amazon-kinesis-video-streams-producer-sdk-cpp/build# cat ../samples/kvs_log_configuration 
log4cplus.rootLogger=VERBOSE, KvsConsoleAppender, KvsFileAppender

#log4cplus.logger.MemoryCheck=TRACE, KvsConsoleAppender

#KvsConsoleAppender:
log4cplus.appender.KvsConsoleAppender=log4cplus::ConsoleAppender
log4cplus.appender.KvsConsoleAppender.layout=log4cplus::PatternLayout
log4cplus.appender.KvsConsoleAppender.layout.ConversionPattern=[%-5p] [%D{%d-%m-%Y %H:%M:%S:%Q %Z}] %m%n

#KvsFileAppender
log4cplus.appender.KvsFileAppender=log4cplus::DailyRollingFileAppender
log4cplus.appender.KvsFileAppender.File=./log/${AWS_VIDEO_STREAM_NAME}_${AWS_VIDEO_STREAM_NAME}_${AWS_VIDEO_STREAM_NAME}_${AWS_VIDEO_STREAM_NAME}_${AWS_VIDEO_STREAM_NAME}_${AWS_VIDEO_STREAM_NAME}_kvs.log
log4cplus.appender.KvsFileAppender.Schedule=HOURLY
log4cplus.appender.KvsFileAppender.CreateDirs=true
log4cplus.appender.KvsFileAppender.layout=log4cplus::PatternLayout
log4cplus.appender.KvsFileAppender.layout.ConversionPattern=[%-5p] [%D{%d-%m-%Y %H:%M:%S:%Q %Z}] %m%n
log4cplus.appender.KvsFileAppender.Append=true
```
How does one enable VERBOSE logging? is it via setting this to TRACE?

MushMal, 2021-02-12:
From what I remember it is trace. Regarding the rtsp. It will depend on your source and the media pipeline. You need to ask in their forums as we might not be familiar as much but detecting the staleness and restarting the pipeline sounds like a plan

bml1g12, 2021-02-12:
I see thank you. Been running on trace for most of today but not yet reproduced that error; will leave it running over the weekend so on Monday can likely provide the TRACE but the idea that the RTSP server has given up sending frames for some reason sounds possible. 

The device is an AXIS encoder, and it is being read directly from the RTSP address over wifi; so in terms of the media pipeline I think its "rtsp-kinesis-pipeline" https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/106228f30aa506f7648d5337e99023387711f721/samples/kvs_gstreamer_sample.cpp#L998 on the AWS SDK side.   

Would you be able to provide any guidance on how one could efficiently implement such a staleness detector? To be honest, even just crashing out in this situation would solve the problem as my container manager will auto-restart it. Would be grateful if you could point me to e.g. a line of code where the error 

> Timeout was reached: Operation too slow.

is thrown, as I expect I could hackily solve this by just placing an `exit()`  at that line.  If you have any advice on a more elegant solution, even better :D

MushMal, 2021-02-12:
I would not recommend checking for the timeout - that could happen from time-to-time and the SDK is capable of recovering. Perhaps a better approach is to have your own filter or your own logic in the application that will keep the track of the frames arriving from rtsp source and if they stop being produced after some period of time then do your application specific logic there.

bml1g12, 2021-02-15:
@MushMal  please see the below trace log, would you be able to confirm whether this is consistent with your theory that the RTSP stream has stopped sending frames?

```
[TRACE] [12-02-2021 22:32:07:807.250 JST] putKinesisVideoFrame(): debug frame info pts: 16131367260882810, dts: 253178423544, duration: 0, size: 986, trackId: 1, isKey 0                                                                   
[TRACE] [12-02-2021 22:32:07:807.301 JST] notifyDataAvailable(): Note data received: duration(100ns): 2340000 bytes 26613 for stream handle 10                                                                                              
[TRACE] [12-02-2021 22:32:07:807.313 JST] streamDataAvailableHandler invoked for stream: TokyoEbisu_EntranceInner and stream upload handle: 10                                                                                              
[TRACE] [12-02-2021 22:32:07:840.703 JST] putKinesisVideoFrame(): debug frame info pts: 16131367261216702, dts: 253178757940, duration: 0, size: 581, trackId: 1, isKey 0                                                                   
[TRACE] [12-02-2021 22:32:07:840.763 JST] notifyDataAvailable(): Note data received: duration(100ns): 2670000 bytes 27207 for stream handle 10                                                                                              
[TRACE] [12-02-2021 22:32:07:840.778 JST] streamDataAvailableHandler invoked for stream: TokyoEbisu_EntranceInner and stream upload handle: 10                                                                                              
[TRACE] [12-02-2021 22:32:07:873.931 JST] putKinesisVideoFrame(): debug frame info pts: 16131367261550261, dts: 253179090357, duration: 0, size: 618, trackId: 1, isKey 0                                                                   
[TRACE] [12-02-2021 22:32:07:873.984 JST] notifyDataAvailable(): Note data received: duration(100ns): 3010000 bytes 27838 for stream handle 10                                                                                              
[TRACE] [12-02-2021 22:32:07:873.997 JST] streamDataAvailableHandler invoked for stream: TokyoEbisu_EntranceInner and stream upload handle: 10                                                                                              
[TRACE] [12-02-2021 22:32:07:907.212 JST] putKinesisVideoFrame(): debug frame info pts: 16131367261884041, dts: 253179423140, duration: 0, size: 592, trackId: 1, isKey 0                                                                   
[TRACE] [12-02-2021 22:32:07:907.265 JST] notifyDataAvailable(): Note data received: duration(100ns): 3340000 bytes 28443 for stream handle 10                                                                                              
[TRACE] [12-02-2021 22:32:07:907.277 JST] streamDataAvailableHandler invoked for stream: TokyoEbisu_EntranceInner and stream upload handle: 10                                                                                              
[TRACE] [12-02-2021 22:32:07:908.308 JST] postWriteCallback(): postBodyStreamingWriteFunc (curl callback) invoked                                                                                                                           
[DEBUG] [12-02-2021 22:32:07:908.328 JST] postWriteCallback(): Curl post body write function for stream with handle: TokyoEbisu_EntranceInner and upload handle: 10 returned: {"EventType":"RECEIVED","FragmentTimecode":1613136723919,"Frag
mentNumber":"91343852333182522492970096397953252136006060056"}                                                                                                                                                                              
                                                                                                                                                                                                                                            
[DEBUG] [12-02-2021 22:32:07:908.359 JST] fragmentAckReceivedHandler invoked                                                                                                                                                                
[DEBUG] [12-02-2021 22:32:07:908.374 JST] Reporting fragment ack received. Ack timecode 1613136723919                                                                                                                                       
[TRACE] [12-02-2021 22:32:07:908.385 JST] postWriteCallback(): Processed ACK OK.                                                                                                                                                            
[TRACE] [12-02-2021 22:32:07:908.397 JST] postReadCallback(): postBodyStreamingReadFunc (curl callback) invoked                                                                                                                             
[TRACE] [12-02-2021 22:32:07:908.419 JST] postReadCallback(): Get Stream data returned: buffer size: 65524 written bytes: 28443 for upload handle: 10 current stream handle: 94017391665592                                                 
[DEBUG] [12-02-2021 22:32:07:908.428 JST] postReadCallback(): Wrote 28443 bytes to Kinesis Video. Upload stream handle: 10                                                                                                                  
[... cut ...]
[TRACE] [12-02-2021 22:32:08:082.947 JST] postWriteCallback(): postBodyStreamingWriteFunc (curl callback) invoked                                                                                                                           
[DEBUG] [12-02-2021 22:32:08:082.983 JST] postWriteCallback(): Curl post body write function for stream with handle: TokyoEbisu_EntranceInner and upload handle: 10 returned: {"EventType":"RECEIVED","FragmentTimecode":1613136724987,"Frag
mentNumber":"91343852333182522497921856555094773623291269982"}     
```

bml1g12, 2021-02-15:
The device delivering the RTSP stream logs the following at the time of the issue

>  "<CRITICAL> Feb 12 22:32:08 axis-accc8efb1a95 monolith[1113]: udpsrc59: Could not read from resource."

hassanctech, 2021-02-15:
It seems like then the camera did indeed stop producing frames?  You mentioned you're using WiFi, are you able to try a wired network to see if the issue is related to your WiFi?

Your question though is about if something like this happens you'd prefer to crash and be restarted (which is a typical scenario) than to just freeze.  Yes it looks like you would need to detect this happened somehow and terminate yourself.

To confirm so this callback doesn't get invoked:  https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/e664fb1616c5d5c35966bafd3dcc6506b7b1d82a/samples/kvs_gstreamer_sample.cpp#L454 

?

If so it looks like the application vending the RTP packets coming out of the axis camera is not propagating the error. 

Here are some additional callbacks you can register with, perhaps it's not being reported as an error but a warning or something else, I would advise you to setup the additional callbacks (I'm including the error one just to provide context):

```
data->bus = gst_element_get_bus(data->pipeline);
gst_bus_add_signal_watch(data->bus);
g_signal_connect (G_OBJECT(data->bus), "message::error", (GCallback) error_bus_cb, data);
g_signal_connect (G_OBJECT(data->bus), "message::warning", (GCallback) warning_bus_cb, data);
g_signal_connect (G_OBJECT(data->bus), "message::info", (GCallback) info_bus_cb, data);
g_signal_connect (G_OBJECT(data->bus), "message::eos", (GCallback) eos_bus_cb, data);
g_signal_connect (G_OBJECT(data->bus), "message::unknown", (GCallback) unknown_bus_cb, data);
 ```

You need to add them here:  https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/e664fb1616c5d5c35966bafd3dcc6506b7b1d82a/samples/kvs_gstreamer_sample.cpp#L1015

Then define those additional callback functions, here is a barebones example to help you:

```
static void error_bus_cb(GstBus *bus, GstMessage *msg, CustomData *data) {
    GError *err = NULL;
    gchar *debug_info = NULL;

    gst_message_parse_error(msg, &err, &debug_info);
    print_and_free_errordebug("ERROR", msg, err, debug_info);
    g_main_loop_quit(data->main_loop);
}

static void warning_bus_cb(GstBus *bus, GstMessage *msg, CustomData *data) {
    GError *err = NULL;
    gchar *debug_info = NULL;

    gst_message_parse_warning(msg, &err, &debug_info);
    print_and_free_errordebug("WARNING", msg, err, debug_info);
}

static void info_bus_cb(GstBus *bus, GstMessage *msg, CustomData *data) {
    GError *err = NULL;
    gchar *debug_info = NULL;

    gst_message_parse_info(msg, &err, &debug_info);
    print_and_free_errordebug("INFO", msg, err, debug_info);
}

static void eos_bus_cb(GstBus *bus, GstMessage *msg, CustomData *data) {
    GError *err = NULL;
    gchar *debug_info = NULL;

    LOG4CPLUS_DEBUG(logger, "Received EOS message on bus...");
    print_and_free_errordebug("EOS", msg, err, debug_info);
}

static void unknown_bus_cb(GstBus *bus, GstMessage *msg, CustomData *data) {
    GError *err = NULL;
    gchar *debug_info = NULL;

    LOG4CPLUS_DEBUG(logger, "Received UNKNOWN message on bus...");
    print_and_free_errordebug("UNKNOWN", msg, err, debug_info);
}
```

and

```
static void print_and_free_errordebug(const std::string& messageTypeString, GstMessage *msg, GError* err, gchar* debug_info) {

    switch(msg->type) {
        case GST_MESSAGE_ERROR:
            LOG4CPLUS_ERROR(logger, messageTypeString + " message received on bus,  from element "
                    << (GST_OBJECT_NAME(msg->src)) << ": " << (err ? err->message : "none") << "\n");
            LOG4CPLUS_ERROR(logger, "Debugging information: " << (debug_info ? debug_info : "none") << "\n");
            break;
        default:
            LOG4CPLUS_DEBUG(logger, messageTypeString + " message received on bus,  from element "
                    << (GST_OBJECT_NAME(msg->src)) << ": " << (err ? err->message : "none") << "\n");
            LOG4CPLUS_DEBUG(logger, "Debugging information: " << (debug_info ? debug_info : "none") << "\n");
            break;
    }

    g_error_free(err);
    g_free(debug_info);
}
```

This is just to help you get started in case you're unfamiliar with how to set these up in GStreamer.

Notice in the examples above `g_main_loop_quit ` is only present in the ERROR callback.  Perhaps if you find what you need in a WARNING callback you can tailor the behavior for the specific type of error you wan't to exit on to also call `g_main_loop_quit `.

mbdevpl, 2021-02-22:
I'm working together with @bml1g12 on this problem.

This is a long post so I've divided it in sections

### Reproducing the issue

I'm able to consistently reproduce the behaviour locally using the following approach:

I stream a video using VLC at `rtsp://localhost:5554/`, and with AWS configured (both credentials and Kinesis video stream created on the cloud side) then I launch the producer pointing to that stream.

Exact steps to reproduce:
1. Open VLC
2. select Media → Stream
3. “Add” a suitable file
4. click “Stream, click “Next”
5. in “New destination” select RTSP and click “Add
6. select suitable port (e.g. 5554), leave URL as is (i.e. /), click “Next”
7. select suitable video profile - could be just H.264, click 'Next”
8. finally, click “Stream”
9. launch the sample app like so (assuming you have the env set and authenticated with AWS):
   ```./kvs_gstreamer_sample $STREAM_NAME rtsp://localhost:5554/```
10. wait a bit for the stream to start, then click “Pause” in VLC
11. wait for about 30 seconds
12. error should occur

### Using gstreamer callbacks to catch the event

@hassanctech Thank you for the solution idea - I went ahead and forked this repo and implemented according to the idea: See this commit: https://github.com/scoville/amazon-kinesis-video-streams-producer-sdk-cpp/commit/116fe35de39d56b58dcd80c3f289f4234fc871e3

Unfortunately, even with very naive catch-all approach with all callbacks behaving as if there is an error (i.e. even for the info/debug level gstreamer messages it should exit) and no filtering of the messages - i.e. really any callback at all should cause exit - but it seems those routines are not invoked at all, i.e. the log messages we're seeing are not caused by gstreamer.

### Modifying curl-related code in C producer

I was able to get promising results by modifying the C producer code, which is a dependency of the C++ producer. I also forked that other repo and put my code here: https://github.com/scoville/amazon-kinesis-video-streams-producer-c/commit/d74e89e468dc2eb51ef27532e2ede32c2d2ad1fa - this approach works, i.e. the app crashes as expected after the "Operation too slow" event.

This is a very hacky solution, however, and I'd very much appreciate @hassanctech if you might have ideas on how to approach it more properly. Gstreamer seems oblivious to those timeout events, but maybe you're aware of some flags that could make it aware of them, if it's the right approach here.

### Core issue - proper handling of idle/paused streams

Finally, even with the timeouts causing crash and subsequent quick restart of the software, there's timeout of 30 seconds which introduces quite a stillness. To catch idle/paused streams earlier, I've reduced the timeout to 5 seconds in this commit: https://github.com/scoville/amazon-kinesis-video-streams-producer-c/commit/caa0351310920b8221024a47815c0992822516b0

This makes the idle time shorter, but of course our current approach is quite hacky overall, and we'd appreciate any hints about better solution. Do you have any better ideas how to detect idle/paused streams faster? Especially, just before pausing, maybe the stream sends some RTSP messages signalling that it enters such a state? Wondering if such a thing exists and if we could have an instantaneous reaction.

hassanctech, 2021-02-23:
I'm just trying to make sure I understand here so it looks like the issue is with the source itself and you need to be able to detect that somehow the Axis camera has stopped sending frames when you request an RTSP stream.  I have some ideas here but just want to make sure I understood that part correctly because if that's the case I would also advise you to reach out to Axis and ask them why this is happening it sounds like a product defect (assuming it's not due to bad network?).

Ok here's an idea but it will require burning an extra thread.  Create a Timer Queue object:  https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/master/src/utils/src/TimerQueue.c#L6

Then create at timer which fires say every 1s.  You give your callback function (the one that fires every 1s) some shared global structure which is shared between it and the `appsink` callback.  Basically you're just going to checkpoint every time `appsink` is invoked, update a new field you create to track the last update time.  In your Timer Queue callback, compare the last update time to time now.  If it's > 1s you can what you'd like i.e. kill the other thread.  It has to be in the other thread because the whole problem is if there's no data flowing the `appsink` is actually not getting called so you can't detect it from that thread (well maybe you can I can't think of an easy way off the top of my head).  But this should work.

Here is an example of creating a timer using that API I mentioned (By the way if you're using Producer SDK you have access to all of PIC primitives, the C Producer SDK depends on PIC and the CPP Producer SDK depends on the C Producer SDK), here:  https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/master/src/client/src/Client.c#L216

It's a super simple API to use and should solve your problem.  Use a MUTEX for locking around read and write the last update time variable.  You don't really care about what order it happens that's not important for correctness, you just want to make sure it's atomic. 

I hope that helps, let me know if something is unclear.

bml1g12, 2021-02-24:
Thanks again for your assistance.

>  I would also advise you to reach out to Axis and ask them why this is happening it sounds like a product defect (assuming it's not due to bad network?).

We have reached out to AXIS but they have not encountered this issue before and given its an old firmware are unlikely to be able to provide much support until we have upgraded device later this year. Their best suggestion is to give us a custom firmware with more verbose logging on their device to try and understand what is going on. It is unclear to me if its a network issue related; but if it was network issue related I hope we can build a Kinesis SDK app which is able to handle such a case; as I imagine many KVS users will encounter issues with their network.

MushMal, 2021-02-24:
Interesting thread. Just read through the provided log and it is a clear case when the Producer SDK is not receiving the frames and the underlying connection times out. This is a hard-limit enforced by not only the SDK but the service side API. The SDK is fully supporting the "intermittent" producer case but it doesn't know how to "fix" the upstream media source which has a problem not producing frames.

There are two issues here which you might need to work on.

1) Detecting when your source goes "stale"
2) How do you fix the "staleness" once you detect it

I am unfamiliar with the RTSP source/Axis camera so I can give a few simple suggestions you could try

* Check whether the source element has a way to handle detection/recovery on networking or other issues.
* If it's impossible then you can use some sort of a timeout timer (samples are C++ so you can use C++ libraries for a timer or opt for our own TimerQueue). Depending on a timer implementation, it could either fire regularly at the timeout cadence and check the last produced frame timestamp or get re-set whenever the frame is produced. Net result is that it can detect if your media source haven't produced frame for a while - Detection.
* You can force EoS into the media pipeline once detected
* You can tear down and re-created the entire media pipeline in the application
* If your media source supports an out-of-band reset - use it after detecting.

NOTE: You don't need to do anything with the KVS Producer - it will just continue working as soon as there are frames that are produced

bml1g12, 2021-02-26:
AXIS have confirmed the issue occurs when the encoder is "not possible anymore for the camera/encoder to transfer packets or receive a response from the destination address" and we are communicating with AXIS to see if they can provide a solution on their end.

We will confirm shortly if upgrading the encoder helps with this issue; but if not, we will start diving into the code suggestions you have kindly provided above - thank you!

bml1g12, 2021-03-07:
We found upgrading our AXIS encoder to a later version resolved this issue so I will close this for now; but if we make further progress on the handling and need further discussion will re-open. 

Thanks for all your assistance here.

closed 2021-03-07 by bml1g12

hassanctech, 2021-03-09:
Ok great!

qraleq, 2021-04-06:
Hi, @bml1g12, we're observing very similar issues with our AXIS RTSP streams and KVS. Can you please share more details on your solution?

hassanctech, 2021-04-06:
@qraleq It might help to also share your Axis product (model) and if you have upgraded to the latest firmware?

bml1g12, 2021-04-06:
Sure I can share a bit more details; we encountered this issue specifically with the Axis P7216 encoder; which is on firmware 5.51. This erratic timeout issue was resolved by using Axis Encoder P7304 which is on firmware 10.4.0

By the way, we are currently working on ACAP integration directly with the encoder to avoid the need for RTSP streaming at all and to send the feed from the device directly; if you or your company would be interested in this functionality keep me posted and we can get in touch when it is ready.

qraleq, 2021-04-07:
Hi, now I understand that you weren't talking about a software encoder, but a hardware encoder. We're using AXIS Q6135-LE PTZ (10.2.0) and AXIS P3807-PVE (9.80.2.2) cameras, and most of the time everything is working fine, but sometimes our pipeline gets stuck at the curl error and doesn't recover from that.

@bml1g12 ACAP and KVS integration sounds very interesting to me. I would really like to know more about it, I'll send you a private message about it.