awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #662: Question/Bug: Can't locate Text/Tabs.pm in @INC (you may need to install the Text::Tabs module) while compiling.

Winneee, 2021-02-20:
When running cmake from master branch with build dependencies ON for aarch64 poky linux i am getting error as Can't locate Text/Tabs.pm in @INC (you may need to install the Text::Tabs module) .
Do you have any recommendations on how to fix this issue? Thank you!

disa6302, 2021-02-20:
@Winneee ,

Not sure what the error is and have not encountered it before. Can you add the build logs? Have you made any modifications to the scripts?

Winneee, 2021-02-22:
[kinesis-cpp-log.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/6019947/kinesis-cpp-log.txt)

MushMal, 2021-02-22:
The logs indicate that you are unable to build libautomake. 

Can you please provide more info on your build/target platform? 

Are you able to build it on a commodity OS like osx or Linux?

Winneee, 2021-02-23:
Build perl from source fixed the issue.