awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #663: [QUESTION] Dropped frame/fragment

pwijeratne, 2021-02-23:
Seeing dropped fragments at the beginning of a video clip

The log shows putKinesisVideoFrame being called on the correct frame at time stamp (1614045903) 06:05:03 PM. However the first frame I playback is 4s later (06:05:07 PM). Two fragments have dropped.

- I am using intermittent producer case
- C Producer SDK version 1.1.0
- I have the following setting: 
pstreamInfo->streamCaps.frameOrderingMode = FRAME_ORDERING_MODE_MULTI_TRACK_AV_COMPARE_DTS_ONE_MS_COMPENSATE;

** Logging **
Feb 22 18:05:06 kvs_stream: [KVS] putKinesisVideoFrame(): debug frame info pts: 16140459038412770, dts: 16140459038412770, duration: 0, size: 77396, trackId: 1, isKey 1
Feb 22 18:05:06 kvs_stream: [KVS] stepStateMachine(): State Machine - Current state: 0x0000000000000200, Next state: 0x0000000000000040
Feb 22 18:05:06 kvs_stream: [KVS] stepStateMachine(): State Machine - Current state: 0x0000000000000040, Next state: 0x0000000000000080
Feb 22 18:05:06 kvs_stream: [KVS] putStreamResultEvent(): Put stream result event. New upload handle 6
Feb 22 18:05:06 kvs_stream: [KVS] notifyDataAvailable(): Note data received: duration(100ns): 0 bytes 0 for stream handle 6
Feb 22 18:05:06 kvs_stream: [KVS] notifyDataAvailable(): Failed to un-pause curl with error: 43. Curl object 0x7035c8
Feb 22 18:05:06 kvs_stream: [KVS] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 7352272. Upload handle 6. Duration available in 100ns: 0. Bytes available: 0
Feb 22 18:05:06 kvs_stream: [KVS] stepStateMachine(): State Machine - Current state: 0x0000000000000080, Next state: 0x0000000000000100
Feb 22 18:05:06 kvs_stream: [KVS] putKinesisVideoFrame(): debug frame info pts: 16140459038794310, dts: 16140459038794310, duration: 0, size: 320, trackId: 2, isKey 0
Feb 22 18:05:06 kvs_stream: [KVS] getKinesisVideoMetrics(): Get the memory metrics size.
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric(): Kinesis Video client and stream metrics:
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Overall storage byte size: 9437184 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Available storage byte size: 9192386 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Allocated storage byte size: 244798 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Total view allocation byte size: 374480 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Total streams frame rate (fps): 732 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Total streams transfer rate (bps): 1367872 (1335 Kbps)
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Current view duration (ms): 0 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Overall view duration (ms): 1269 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Current view byte size: 0 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Overall view byte size: 164886 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Current frame rate (fps): 732.674828 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Current transfer rate (bps): 1367872 (1335 Kbps)
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Stream uptime in (ms): 1099777 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Total transferred bytes: 16565783 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Total number of streaming sessions: 7 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Total number of active streaming sessions: 6 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Average streaming sessions duration: 123710428 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Number of Buffered ACKs: 72 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Number of Received ACKs: 68 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Number of Persisted ACKs: 68 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Number of Error ACKs: 2 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Number of dropped frames: 4 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Number of skipped frames: 321 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Number of storage pressure events: 0 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Number of latency pressure events: 0 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Number of buffer pressure events: 0 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Number of stream staleness events: 6 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Number of latency pressure events: 0 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Number of Put Frame errors: 173 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Average Control Plane API latency (ms): 24 
Feb 22 18:05:06 kvs_stream: [KVS] logStreamMetric():    Average Data Plane API latency (ms): 0 
Feb 22 18:05:06 kvs_stream: [KVS] notifyDataAvailable(): Note data received: duration(100ns): 0 bytes 77792 for stream handle 6
Feb 22 18:05:06 kvs_stream: [KVS] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 7352272. Upload handle 6. Duration available in 100ns: 0. Bytes available: 77792
Feb 22 18:05:06 kvs_stream: [KVS] putKinesisVideoFrame(): debug frame info pts: 16140459039075430, dts: 16140459039075430, duration: 0, size: 4098, trackId: 1, isKey 0
Feb 22 18:05:06 kvs_stream: [KVS] notifyDataAvailable(): Note data received: duration(100ns): 380000 bytes 78125 for stream handle 6
Feb 22 18:05:06 kvs_stream: [KVS] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 7352272. Upload handle 6. Duration available in 100ns: 380000. Bytes available: 78125
Feb 22 18:05:06 kvs_stream: [KVS] putKinesisVideoFrame(): debug frame info pts: 16140459039193280, dts: 16140459039193280, duration: 0, size: 320, trackId: 2, isKey 0
Feb 22 18:05:06 kvs_stream: [KVS] viewItemRemoved(): Reporting a dropped frame/fragment.
Feb 22 18:05:06 kvs_stream: [kvs_module.c:164][count:1]Reported droppedFrame callback for stream handle 7352272. Dropped frame timecode in 100ns: 16140459038410000[KVS] viewItemRemoved(): Reporting a dropped frame/fragment.
Feb 22 18:05:06 kvs_stream: [KVS] contentViewTrimTailItems(): ContentView is not big enough to contain a single fragment.
Feb 22 18:05:06 kvs_stream: [KVS] notifyDataAvailable(): Note data received: duration(100ns): 0 bytes 4111 for stream handle 6
Feb 22 18:05:06 kvs_stream: [KVS] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 7352272. Upload handle 6. Duration available in 100ns: 0. Bytes available: 4111
Feb 22 18:05:06 kvs_stream: [KVS] putKinesisVideoFrame(): debug frame info pts: 16140459039593620, dts: 16140459039593620, duration: 0, size: 320, trackId: 2, isKey 0
Feb 22 18:05:06 kvs_stream: [KVS] putKinesisVideoFrame(): debug frame info pts: 16140459039723810, dts: 16140459039723810, duration: 0, size: 2913, trackId: 1, isKey 0
Feb 22 18:05:06 kvs_stream: [KVS] notifyDataAvailable(): Note data received: duration(100ns): 120000 bytes 4444 for stream handle 6
Feb 22 18:05:06 kvs_stream: [KVS] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 7352272. Upload handle 6. Duration available in 100ns: 120000. Bytes available: 4444
Feb 22 18:05:06 kvs_stream: [KVS] notifyDataAvailable(): Note data received: duration(100ns): 520000 bytes 4777 for stream handle 6
Feb 22 18:05:06 kvs_stream: [KVS] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 7352272. Upload handle 6. Duration available in 100ns: 520000. Bytes available: 4777
Feb 22 18:05:06 kvs_stream: [KVS] putKinesisVideoFrame(): debug frame info pts: 16140459039993370, dts: 16140459039993370, duration: 0, size: 320, trackId: 2, isKey 0
Feb 22 18:05:06 kvs_stream: [KVS] notifyDataAvailable(): Note data received: duration(100ns): 650000 bytes 7703 for stream handle 6
Feb 22 18:05:06 kvs_stream: [KVS] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 7352272. Upload handle 6. Duration available in 100ns: 650000. Bytes available: 7703
Feb 22 18:05:06 kvs_stream: [KVS] putKinesisVideoFrame(): debug frame info pts: 16140459040406390, dts: 16140459040406390, duration: 0, size: 7199, trackId: 1, isKey 0
Feb 22 18:05:06 kvs_stream: [KVS] notifyDataAvailable(): Note data received: duration(100ns): 920000 bytes 8036 for stream handle 6
[... cut ...]
Feb 22 18:05:06 kvs_stream: [KVS] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 7352272. Upload handle 6. Duration available in 100ns: 1320000. Bytes available: 8369
Feb 22 18:05:06 kvs_stream: [KVS] putKinesisVideoFrame(): debug frame info pts: 16140459040800390, dts: 16140459040800390, duration: 0, size: 320, trackId: 2, isKey 0
Feb 22 18:05:06 kvs_stream: [KVS] notifyDataAvailable(): Note data received: duration(100ns): 1330000 bytes 15581 for stream handle 6

disa6302, 2021-02-23:
@pwijeratne , 

From the logs, it looks like you are trying to run a multi track scenario. There was a bug where the frame order mode was not being set to the appropriate mode in multi track scenario. The top of master has the fix for setting the appropriate mode by default and will go into a revision release soon. The logs do not give any information about the error code, but looks like there are 173 put frame errors and 2 error acks. Can you try using this mode: `FRAME_ORDERING_MODE_MULTI_TRACK_AV_COMPARE_PTS_ONE_MS_COMPENSATE_EOFR ` and see if it helps?

pwijeratne, 2021-02-23:
Thank you @disa6302. I set that option and it seems to be working well so far. I will keep it running/testing for a day and update you.

disa6302, 2021-02-23:
Sure thing. You can also switch over to the new version 3.1.1 that we published today. It automatically sets the frame order mode.

pwijeratne, 2021-02-25:
Continued testing passed. Closing issue. Thank you!

closed 2021-02-25 by pwijeratne