awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #664: [QUESTION]createKinesisVideoStreamSync(): Failed to create Kinesis Video Stream - timed out.

Winneee, 2021-02-23:
When running sample code from kvsvideostream.cpp getting timeout .
curlCompleteSync(): curl perform failed for url https://kinesisvideo.ap-south-1.amazonaws.com/describeStream with result Timeout was reached: Resolving timed out after 5000 milliseconds
[WARN ] [23-02-2021 13:20:04:900.382 GMT] curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://kinesisvideo.ap-south-1.amazonaws.com/describeStream

Attached log file.

[log.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/6029134/log.txt)

disa6302, 2021-02-23:
@Winneee , 

Please check your network. Curl calls seem to be failing.

Winneee, 2021-02-24:
@disa6302 Network seems okay below are the logs
root:~/amazon-kinesis-video-streams-producer-sdk-cpp/build# ping amazon.com
PING amazon.com (205.251.242.103) 56(84) bytes of data.
64 bytes from s3-console-us-standard.console.aws.amazon.com (205.251.242.103): icmp_seq=1 ttl=233 time=219 ms
64 bytes from s3-console-us-standard.console.aws.amazon.com (205.251.242.103): icmp_seq=2 ttl=233 time=219 ms
64 bytes from s3-console-us-standard.console.aws.amazon.com (205.251.242.103): icmp_seq=3 ttl=233 time=219 ms
64 bytes from s3-console-us-standard.console.aws.amazon.com (205.251.242.103): icmp_seq=4 ttl=233 time=219 ms
64 bytes from s3-console-us-standard.console.aws.amazon.com (205.251.242.103): icmp_seq=5 ttl=233 time=219 ms
64 bytes from s3-console-us-standard.console.aws.amazon.com (205.251.242.103): icmp_seq=6 ttl=233 time=219 ms
64 bytes from s3-console-us-standard.console.aws.amazon.com (205.251.242.103): icmp_seq=7 ttl=233 time=220 ms
64 bytes from s3-console-us-standard.console.aws.amazon.com (205.251.242.103): icmp_seq=8 ttl=233 time=219 ms
64 bytes from s3-console-us-standard.console.aws.amazon.com (205.251.242.103): icmp_seq=9 ttl=233 time=219 ms

MushMal, 2021-02-24:
@Winneee pinging the backend and issuing a full HTTPS request are very different things. There is nothing we can help you. Try to debug your case. Run with different network configuration and on a commodity OS to eliminate the most obvious issues first. The snap of logs indicate a timeout issuing a call (5 seconds) which is indicative of either a network routing issue/packet loss or a certain topology issue. Try different region as well.

Please resolve this issue and cut a new one with detailed and targeted questions which are related to KVS SDK.

Winneee, 2021-02-24:
@MushMal I tried the same sample code on ubuntu which is on same network as my other device, where i can see upload successful.

MushMal, 2021-02-24:
Debug your device further. As I said, there is absolutely nothing useful we can do unless you debug your own application. 

* Use Verbose logging to get more information that could help you to identify the root cause
* Use curl executable on the device to ensure your libCurl is properly built and can access network
* Use different region (us-east-1 for example)
* Use different AP/Network
* Build debug version of curl to dump more info (check CURL documentation)

NOTE: The issue you are having doesn't seem to be related to KVS SDK nor to KVS service but rather the network/network drivers/kernel on your device as you are able to stream from a commodity OS.

Please resolve this issue and cut a new one if you have any issues related to KVS

disa6302, 2021-02-24:
Resolving the issue. Feel free to reopen if you have any questions.

closed 2021-02-24 by disa6302

burak-meshine, 2022-09-30:
I have same issue on Jetson Nano, it connects after 10 seconds response time so that i can't overcome  5000 millisecond timeout issue. 
- I changed closest region
- I used different network

I worked on that all day long. I am open to suggestions.