awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #668: Sample RTSP code cannot read valid RTSP stream

gherlein, 2021-02-27:
gst-launch-1.0 rtspsrc location="rtsp://192.168.2.120:554/s1" short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! kvssink stream-name="MyKVStream" storage-size=512 access-key="XXX-MYKEY" secret-key="XXX-MYSECRET” aws-region="us-west-2" >out 2>&1
---

ERROR: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc3: Internal data stre
am error.
Additional debug info:
../libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc
0/GstUDPSrc:udpsrc3:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:00.251204964
Setting pipeline to NULL ...
Freeing pipeline ...
INFO - Freeing Kinesis Video Stream MyKVStream
2021-02-26 15:13:16 [140000644548416] INFO - freeKinesisVideoStream(): Freeing Kinesis Video stream.
2021-02-26 15:13:16 [140000644548416] DEBUG - curlApiCallbacksShutdownActiveRequests(): pActiveReque
sts hashtable is empty
2021-02-26 15:13:16 [140000644548416]

**Describe the bug**
Should read video into the kvs

**To Reproduce**
using a Unifi G3 IP camera in standalone mode.

**SDK version number**
Build from HEAD today

[image]

**Platform (please complete the following information):**
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 78
model name	: Intel(R) Core(TM) i3-6100U CPU @ 2.30GHz

Ubuntu 20.10
[out.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/6053456/out.txt)

**Additional context**
Add any other context about the problem here.

disa6302, 2021-02-27:
@gherlein ,

This does not look like a KVS error. I would suggest you to get the pipeline right. Test with `fakesink` Instead of `kvssink` To ensure your pipeline works without errors and then use `kvssink`

gherlein, 2021-02-27:
Fakesink seems to work just fine.

But I found this: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/193

That added the h264parse in the pipeline and that did the trick.  The correct CLI was:

gst-launch-1.0 rtspsrc location="rtsp://192.168.2.120:554/s1" short-header=TRUE ! rtph264de\
pay ! h264parse ! video/x-h264, format=avc,alignment=au ! kvssink stream-name=${STREAM} storage-siz\
e=512 access-key=${ACCESS} secret-key=${SECRET} aws-region=${REGION}

Camera is a Ubiquity G3 Dome - https://dl.ubnt.com/guides/unifivideo/UVC-G3-DOME_QSG.pdf

closed 2021-02-27 by gherlein