awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #669: many error while trying to compile for mip

waikup83, 2021-02-28:
I try to compile the kinesis producer cpp for mips like https://github.com/aws/aws-sdk-cpp/issues/1288 but i have many problem with log4cplus during compilation.

i add --host=mips32-linux --enable-threads=no to liblog4cplus-CMakeList.txt but it failed again

disa6302, 2021-02-28:
It looks like you are creating an issue in the wrong repository? If you are looking for log4cplus cross compilation for Mips, i would suggest you to look at the log4cplus forums for more details. Closing this issue.

closed 2021-02-28 by disa6302

waikup83, 2021-02-28:
But it’s the kinesis cmake file that won’t compile it iwth the REL_1_2_2

Provenance : Courrier<https://go.microsoft.com/fwlink/?LinkId=550986> pour Windows 10

Envoyé le :28 février 2021 15:42
À : awslabs/amazon-kinesis-video-streams-producer-sdk-cpp<mailto:amazon-kinesis-video-streams-producer-sdk-cpp@noreply.github.com>
Cc : waikup83<mailto:waikup@hotmail.com>; Author<mailto:author@noreply.github.com>

disa6302, 2021-03-01:
@waikup83 , 

I would suggest you to cross compile log4cplus for MIPS separately to confirm it works before trying out KVS.