awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #671: need help for compiling to MIPS

waikup83, 2021-03-02:
I would like to integrate it in a camera with a mips toolchain but nothing works

disa6302, 2021-03-02:
@waikup83 , 

Nothing works is not descriptive enough. Please add details - what does not work and what kind of debugging have you tried.

waikup83, 2021-03-02:
im using a toolchain 

SET(CMAKE_CXX_STANDARD 14)

# file name: mips.toolchain.cmake
SET(CMAKE_SYSTEM_NAME Linux)

# specify the cross compiler
SET(CMAKE_C_COMPILER /usr/bin/mips-linux-gnu-gcc)
SET(CMAKE_CXX_COMPILER /usr/bin/mips-linux-gnu-g++)

# where is the target environment
SET(CMAKE_FIND_ROOT_PATH /home/Projet/aws/cross_output)

# search for programs in the build host directories
SET(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)

# for libraries and headers in the target directories
SET(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
SET(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)

i cmake it with
cmake ../ -DCMAKE_TOOLCHAIN_FILE="/home/Projet/aws/mips.toolchain.cmake" -DBUILD_SHARED_LIBS=ON -DCMAKE_INSTALL_PREFIX=$CROSS_INSTALL_PREFIX -DCMAKE_PREFIX_PATH=$CROSS_INSTALL_PREFIX -DAUTORUN_UNIT_TESTS=OFF -DBUILD_DEPENDENCIES=OFF

when i make it by now im at this point
[image]

hassanctech, 2021-03-05:
As part of the dependencies pulled the Producer C SDK is pulled down and built (which depends on kvspic), in our Producer C repo we actually have Travis CI which builds an ARM build (cross-compile), Can you please try to build the Producer C SDK using the example here:

https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/blob/master/.travis.yml#L116

Substitute for your MIPS compiler but follow the steps there and see if you can get that to work.  Once that's working then use that same setup to build the Producer CPP SDK. Let me know if you are still stuck.

waikup83, 2021-03-08:
thanks for the respond
I'm retring the kinesis c producer right now
the c producer have this error
[image]

with this cmake command => cmake ../ -DBUILD_ONLY="s3;iam;sts" -DCMAKE_TOOLCHAIN_FILE="/home/Projet/aws/mips.toolchain.cmake" -DBUILD_SHARED_LIBS=ON -DCMAKE_INSTALL_PREFIX=$CROSS_INSTALL_PREFIX -DCMAKE_PREFIX_PATH=$CROSS_INSTALL_PREFIX -DAUTORUN_UNIT_TESTS=OFF -DBUILD_OPENSSL_PLATFORM=linux-generic32

hassanctech, 2021-03-08:
It looks like you were previously trying to build the aws sdk.  For Producer C SDK is completely indepdent from that, so you do not need this: -DBUILD_ONLY="s3;iam;sts"

Question, are you using `UCLIBC` ?

If so see here:

https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/blob/master/CMakeLists.txt#L7

You will need to pass `-DADD_MUCLIBC=ON` to the cmake

waikup83, 2021-03-08:
i remove the -DBUILD_ONLY... and i added -DADD_MUCLIBC=ON because yes im thinking the camera used the uclib but i get the same error ...

hassanctech, 2021-03-08:
Looks like you need to add the `no-async` option to the Configure command, please go here:

https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/blob/7e0d30f83e41cc9c565bd45d277be6a2c1bb86f7/CMake/Dependencies/libopenssl-CMakeLists.txt#L15

And add `no-async` immediatelya after ${OPENSSL_EXTRA}

waikup83, 2021-03-09:
[image]

hassanctech, 2021-03-09:
It looks like `find_package` could not find it for some reason?  Can you try to pass those variables on the `cmake` command line?  I see output for CURL and libkvspic but did openssl actually build correctly?  Can you also please share the output in a text file instead of an image it is much easier to search through it that way.

waikup83, 2021-03-12:
sorry for the respond time but im trying hard to find some way to compile it.
I'm having error while include cproducer.h

In file included from /home/Projet/Kinesis/cproducer/Include.h:17:0,
                 from main.cpp:5:
/home/Projet/Kinesis/common/Include.h:438:5: error: 'UINT32' does not name a type
/home/Projet/Kinesis/common/Include.h:439:5: error: 'UINT32' does not name a type
/home/Projet/Kinesis/common/Include.h:440:5: error: 'PCHAR' does not name a type
/home/Projet/Kinesis/common/Include.h:441:5: error: 'UINT32' does not name a type
/home/Projet/Kinesis/common/Include.h:442:5: error: 'PCHAR' does not name a type
/home/Projet/Kinesis/common/Include.h:443:5: error: 'UINT32' does not name a type
/home/Projet/Kinesis/common/Include.h:444:5: error: 'PCHAR' does not name a type
/home/Projet/Kinesis/common/Include.h:445:5: error: 'UINT32' does not name a type
/home/Projet/Kinesis/common/Include.h:446:5: error: 'UINT64' does not name a type
/home/Projet/Kinesis/common/Include.h:456:5: error: 'PCHAR' does not name a type
/home/Projet/Kinesis/common/Include.h:457:5: error: 'UINT32' does not name a type
/home/Projet/Kinesis/common/Include.h:458:5: error: 'PCHAR' does not name a type
/home/Projet/Kinesis/common/Include.h:459:5: error: 'UINT32' does not name a type
/home/Projet/Kinesis/common/Include.h:468:14: error: 'ATOMIC_BOOL' does not name a type
/home/Projet/Kinesis/common/Include.h:470:5: error: 'PCHAR' does not name a type
/home/Projet/Kinesis/common/Include.h:473:5: error: 'UINT32' does not name a type
/home/Projet/Kinesis/common/Include.h:474:5: error: 'CHAR' does not name a type
/home/Projet/Kinesis/common/Include.h:475:5: error: 'CHAR' does not name a type
/home/Projet/Kinesis/common/Include.h:476:5: error: 'CHAR' does not name a type
/home/Projet/Kinesis/common/Include.h:477:5: error: 'CHAR' does not name a type
/home/Projet/Kinesis/common/Include.h:479:5: error: 'CHAR' does not name a type
/home/Projet/Kinesis/common/Include.h:480:5: error: 'UINT64' does not name a type
/home/Projet/Kinesis/common/Include.h:481:5: error: 'UINT64' does not name a type
/home/Projet/Kinesis/common/Include.h:482:5: error: 'UINT64' does not name a type
/home/Projet/Kinesis/common/Include.h:483:5: error: 'UINT64' does not name a type
/home/Projet/Kinesis/common/Include.h:484:5: error: 'UINT64' does not name a type
/home/Projet/Kinesis/common/Include.h:485:5: error: 'UINT64' does not name a type
/home/Projet/Kinesis/common/Include.h:487:5: error: 'PSingleList' does not name a type
/home/Projet/Kinesis/common/Include.h:497:5: error: 'UINT32' does not name a type
/home/Projet/Kinesis/common/Include.h:498:5: error: 'SERVICE_CALL_RESULT' does not name a type
/home/Projet/Kinesis/common/Include.h:499:5: error: 'CHAR' does not name a type
/home/Projet/Kinesis/common/Include.h:500:5: error: 'PStackQueue' does not name a type
/home/Projet/Kinesis/common/Include.h:502:5: error: 'PCHAR' does not name a type
/home/Projet/Kinesis/common/Include.h:503:5: error: 'UINT32' does not name a type
/home/Projet/Kinesis/common/Include.h:520:16: warning: ISO C++ forbids declaration of 'STATUS' with no type [-fpermissive]
/home/Projet/Kinesis/common/Include.h:520:16: error: typedef 'STATUS' is initialized (use decltype instead)
/home/Projet/Kinesis/common/Include.h:520:18: error: 'GetCredentialsFunc' was not declared in this scope
/home/Projet/Kinesis/common/Include.h:525:5: error: 'GetCredentialsFunc' does not name a type
/home/Projet/Kinesis/common/Include.h:551:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:560:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:570:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:586:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:595:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:610:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:625:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:642:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:660:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:670:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:680:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:692:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:701:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:723:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:733:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:742:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:751:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:762:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:774:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:785:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:797:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:807:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:820:1: error: 'PUBLIC_API' does not name a type
[... cut ...]
/home/Projet/Kinesis/cproducer/Include.h:690:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/cproducer/Include.h:699:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/cproducer/Include.h:712:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/cproducer/Include.h:721:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/cproducer/Include.h:738:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/cproducer/Include.h:752:1: error: 'PUBLIC_API' does not name a type
make: *** [Makefile:32: build/main.o] Error 1

hassanctech, 2021-03-12:
Can you please share the exact steps and commands you ran from the beginning?

waikup83, 2021-03-12:
apt install -y m4 wget git cmake automake pkg-config
apt install -y libcurl4-openssl-dev

export CROSS_COMPILE_HOME=/home/foscam/mips

export CROSS_INSTALL_PREFIX=/home/Projet/aws/cross_output
export CC=$CROSS_COMPILE_HOME/bin/mips-linux-gnu-gcc
export CXX=$CROSS_COMPILE_HOME/bin/mips-linux-gnu-g++
export RANLIB=$CROSS_COMPILE_HOME/bin/mips-linux-gnu-ranlib
export CPPFLAGS="-I$CROSS_COMPILE_HOME/include/"
export LDFLAGS="-L$CROSS_COMPILE_HOME/lib/ -L$CROSS_COMPILE_HOME/lib64/ -L$CROSS_INSTALL_PREFIX/lib/"
export PATH=$CROSS_COMPILE_HOME/bin:$PATH
export CFLAGS=-FPIC

cd /home/Projet/aws
wget https://www.openssl.org/source/openssl-1.1.1c.tar.gz
tar xvzf openssl-1.1.1c.tar.gz
cd openssl-1.1.1c
./Configure --prefix=$CROSS_INSTALL_PREFIX linux-mips32
make -j
make install

cd /home/Projet/aws
git clone --recursive https://github.com/awslabs/amazon-kinesis-video-streams-producer-c.git
cd /home/Projet/aws/amazon-kinesis-video-streams-producer-c
mkdir build && cd build
cmake ../ -DCMAKE_TOOLCHAIN_FILE="/home/Projet/aws/mips.toolchain.cmake" -DBUILD_SHARED_LIBS=ON -DCMAKE_INSTALL_PREFIX=$CROSS_INSTALL_PREFIX -DCMAKE_PREFIX_PATH=$CROSS_INSTALL_PREFIX -DBUILD_OPENSSL_PLATFORM=linux-generic32 -DADD_MUCLIBC=ON
make -j

ToolChains file :
SET(CMAKE_SYSTEM_NAME Linux)

SET(CMAKE_C_COMPILER /home/foscam/mips/bin/mips-linux-uclibc-gnu-gcc)
SET(CMAKE_CXX_COMPILER /home/foscam/mips/bin/mips-linux-uclibc-gnu-g++)

SET(CMAKE_FIND_ROOT_PATH /home/Projet/aws/cross_output)

SET(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
SET(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
SET(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)

waikup83, 2021-03-12:
doing that failed for linking in sample

[100%] Linking C executable kvsVideoOnlyStreamingSample
[100%] Linking C executable kvsAacAudioVideoStreamingSample
/home/foscam/mips/bin/../lib/gcc/mips-linux-gnu/4.7.2/../../../../mips-linux-gnu/bin/ld: warning: libdl.so.2, needed by /home/Projet/aws/cross_output/lib/libssl.so, not found (try using -rpath or -rpath-link)
/home/foscam/mips/bin/../lib/gcc/mips-linux-gnu/4.7.2/../../../../mips-linux/-gnuhome//foscam/binmips//ldbin:/ .warning.:/ liblibc.so.6/,gcc needed/ mipsby- linux/-homegnu//Projet4.7.2//aws./.cross_output//.lib.//libssl.so.,. /not. .found/ mips(-trylinux -usinggnu /bin/-ldrpath:  orwarning :- rpathlibdl.so.2-,link needed)
by/ home//home/ProjetProjet//awsaws//cross_outputcross_output//lib/liblibssl.so/,libcrypto.so :not  undefinedfound  reference( tryto  using` feof-@rpathGLIBC_2.0 'or
 /-homerpath/-Projetlink/)aws
//cross_outputhome//libfoscam//libcrypto.somips:/bin /undefined. .reference to `realloc/@libGLIBC_2.0/'gcc
//mipshome-/linuxProjet-/gnuaws//4.7.2cross_output//.lib.//libcrypto.so.:. /undefined. .reference/ .to. /`mipsclearerr-@linuxGLIBC_2.0-'gnu
//binhome//ldProjet:/ awswarning/cross_output:/ liblibc.so.6/,libcrypto.so :needed  undefinedby  reference/home/Projet/ aws/tocross_output /`lib_longjmp/@libssl.soGLIBC_2.0,'
not/home/Projet/aws/amazon-kinesis-video-streams -foundproducer -(ctry/ openusing --sourcerpath/ libor/ libcurl.so-:rpath -undefinedlink )reference
 /tohome /`Projetstdout/@awsGLIBC_2.0/'cross_output
//libhome//libcrypto.soProjet:/ awsundefined/ cross_output/libreference/ libcrypto.soto:  `undefinedfeof @referenceGLIBC_2.0 'to
 /`homegmtime_r/Projet@/GLIBC_2.0aws'/
cross_output//homelib//Projetlibcrypto.so/:aws /undefinedcross_output /referencelib /tolibcrypto.so :` reallocundefined@ referenceGLIBC_2.0 'to
 /`hometcsetattr@GLIBC_2.0'
//Projethome//awsProjet//cross_outputaws//libamazon/libcrypto.so-: kinesisundefined- video-referencestreams -toproducer -`cclearerr/@openGLIBC_2.0-'source
//libhome/libcurl.so/:Projet /awsundefined/ cross_output/referencelib /tolibcrypto.so :` strpbrkundefined@ GLIBC_2.0reference'
to/ home`/_longjmpProjet@/GLIBC_2.0aws'/
cross_output//homelib//Projetlibcrypto.so/:aws /undefinedamazon -referencekinesis -tovideo -`streams__register_atfork-@producerGLIBC_2.3.2-'c
//openhome-/sourceProjet//aws/cross_outputlib//lib/libcrypto.so: undefined reference to `libcurl.sofgets@GLIBC_2.0':
 /undefinedhome/Projet/aws/amazon-kinesis-video-streams -producerreference- cto/ open`-stdoutsource@/GLIBC_2.0lib'/
libcurl.so/:home /undefinedProjet /referenceaws /tocross_output /`libinet_pton@/GLIBC_2.0'libcrypto.so
:/ homeundefined/ Projetreference/ awsto/ cross_output`/libgmtime_r/@libssl.so:GLIBC_2.0 'undefined
 /referencehome /toProjet/aws /`cross_outputstrncasecmp/@libGLIBC_2.0/'libcrypto.so
:/ homeundefined/ Projetreference/ awsto/ cross_output`/tcsetattrlib@/GLIBC_2.0libcrypto.so:'
undefined/ homereference/ Projetto/ aws`/strcspnamazon@-GLIBC_2.0kinesis'-
video/-homestreams/Projet-/produceraws-/ccross_output//openlib-/sourcelibcrypto.so/:lib /undefinedlibcurl.so :reference  undefinedto  reference` mlockto@ GLIBC_2.0'`
strpbrk/home@/GLIBC_2.0Projet'/
aws//homecross_output//Projetlib//awslibcrypto.so/:cross_output /undefinedlib /reference libcrypto.soto :` readdirundefined@ GLIBC_2.0reference'
to/ home`/__register_atforkProjet@/GLIBC_2.3.2aws'/
amazon/home-kinesis/-Projetvideo/-awsstreams/-cross_outputproducer/-libc//libcrypto.soopen:- sourceundefined/ libreference to/ `libcurl.sofgets:@ GLIBC_2.0undefined'
reference /tohome /`pthread_mutex_init@ProjetGLIBC_2.0/'aws
//amazonhome-/kinesisProjet-/videoaws-/streams-cross_outputproducer/-libc//libcrypto.soopen:- undefinedsource /referencelib /tolibcurl.so : undefined` openreference @to GLIBC_2.0'`
inet_pton/@homeGLIBC_2.0/'Projet
//awshome//cross_outputProjet//awslib/cross_output//liblibssl.so/libssl.so::  undefinedundefined reference  referenceto  `memsetto@ GLIBC_2.0`'strncasecmp
@/GLIBC_2.0'
home//homeProjet//Projetaws//awscross_output//cross_outputlib//liblibssl.so/:libcrypto.so : undefinedundefined  referencereference  toto  ``strcasecmp@strcspnGLIBC_2.0@'
GLIBC_2.0/'home
//Projethome//aws/cross_output/lib/libcrypto.so: undefined reference to `Projetshutdown/aws@/GLIBC_2.0cross_output'/
lib//homelibcrypto.so/: Projetundefined /aws/cross_output/lib/libcrypto.so: undefined reference to `htonl@GLIBC_2.0'
/home/referenceProjet/ awsto/ amazon`-mlockkinesis@-GLIBC_2.0video'-
streams-/home/Projet/produceraws/cross_output/-lib/clibcrypto.so/open:-source /undefinedlib /libcurl.so: undefined reference to `setvbuf@GLIBC_2.0'
reference/home /Projetto/ aws`/readdircross_output@/GLIBC_2.0lib'/
libcrypto.so/:home /undefinedProjet /referenceaws /toamazon -`kinesisgetpid-@videoGLIBC_2.0-streams-producer-c/open-source/lib/libcurl.so: undefined reference to `pthread_mutex_init@GLIBC_2.0'
/home/Projet/aws/cross_output/lib/libcrypto.so: undefined reference to `open@GLIBC_2.0'
/home/Projet/aws/cross_output/lib/libssl.so: undefined reference to `memset@GLIBC_2.0'
/home/Projet/aws/cross_output'/
/libhome/libssl.so:/ Projetundefined/ awsreference/ cross_outputto/ lib`strcasecmp@GLIBC_2.0'
/home/Projet/aws/cross_output/lib/libcrypto.so: undefined reference to `shutdown@GLIBC_2.0'
/home//Projet/aws/cross_output/lib/libcrypto.so: undefined reference to libcrypto.so`htonl:@ undefinedGLIBC_2.0 '
[... cut ...]
make[2]: *** [CMakeFiles/kvsAacAudioVideoStreamingSample.dir/build.make:90: kvsAacAudioVideoStreamingSample] Error 1
make[2]: *** [CMakeFiles/kvsVideoOnlyStreamingSample.dir/build.make:90: kvsVideoOnlyStreamingSample] Error 1
make[1]: *** [CMakeFiles/Makefile2:111: CMakeFiles/kvsAacAudioVideoStreamingSample.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:141: CMakeFiles/kvsVideoOnlyStreamingSample.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
root@e7bbc22da04e:/home/Projet/aws/amazon-kinesis-video-streams-producer-c/build#

waikup83, 2021-03-12:
but this error are for the sample ..

if i dont care and just want the libproducer.so ... i can put it in my project with the header in amazon producer/src/include
but when build my project the hader seem to have many error 

mips-linux-uclibc-gnu-g++ -c -o build/main.o main.cpp -std=c++11 -I/home/Projet/Kinesis/ -Os -Wall -fmessage-length=0 -fPIC -ffunction-sections -fdata-sections -march=mips32r2 -muclibc -fpermissive -g
In file included from /home/Projet/Kinesis/cproducer/Include.h:18:0,
                 from main.cpp:5:
/home/Projet/Kinesis/common/Include.h:440:5: error: 'UINT32' does not name a type
/home/Projet/Kinesis/common/Include.h:441:5: error: 'UINT32' does not name a type
/home/Projet/Kinesis/common/Include.h:442:5: error: 'PCHAR' does not name a type
/home/Projet/Kinesis/common/Include.h:443:5: error: 'UINT32' does not name a type
/home/Projet/Kinesis/common/Include.h:444:5: error: 'PCHAR' does not name a type
/home/Projet/Kinesis/common/Include.h:445:5: error: 'UINT32' does not name a type
/home/Projet/Kinesis/common/Include.h:446:5: error: 'PCHAR' does not name a type
/home/Projet/Kinesis/common/Include.h:447:5: error: 'UINT32' does not name a type
/home/Projet/Kinesis/common/Include.h:448:5: error: 'UINT64' does not name a type
/home/Projet/Kinesis/common/Include.h:458:5: error: 'PCHAR' does not name a type
/home/Projet/Kinesis/common/Include.h:459:5: error: 'UINT32' does not name a type
/home/Projet/Kinesis/common/Include.h:460:5: error: 'PCHAR' does not name a type
/home/Projet/Kinesis/common/Include.h:461:5: error: 'UINT32' does not name a type
/home/Projet/Kinesis/common/Include.h:470:14: error: 'ATOMIC_BOOL' does not name a type
/home/Projet/Kinesis/common/Include.h:472:5: error: 'PCHAR' does not name a type
/home/Projet/Kinesis/common/Include.h:475:5: error: 'UINT32' does not name a type
/home/Projet/Kinesis/common/Include.h:476:5: error: 'CHAR' does not name a type
/home/Projet/Kinesis/common/Include.h:477:5: error: 'CHAR' does not name a type
/home/Projet/Kinesis/common/Include.h:478:5: error: 'CHAR' does not name a type
/home/Projet/Kinesis/common/Include.h:479:5: error: 'CHAR' does not name a type
/home/Projet/Kinesis/common/Include.h:481:5: error: 'CHAR' does not name a type
/home/Projet/Kinesis/common/Include.h:482:5: error: 'UINT64' does not name a type
/home/Projet/Kinesis/common/Include.h:483:5: error: 'UINT64' does not name a type
/home/Projet/Kinesis/common/Include.h:484:5: error: 'UINT64' does not name a type
/home/Projet/Kinesis/common/Include.h:485:5: error: 'UINT64' does not name a type
/home/Projet/Kinesis/common/Include.h:486:5: error: 'UINT64' does not name a type
/home/Projet/Kinesis/common/Include.h:487:5: error: 'UINT64' does not name a type
/home/Projet/Kinesis/common/Include.h:489:5: error: 'PSingleList' does not name a type
/home/Projet/Kinesis/common/Include.h:499:5: error: 'UINT32' does not name a type
/home/Projet/Kinesis/common/Include.h:500:5: error: 'SERVICE_CALL_RESULT' does not name a type
/home/Projet/Kinesis/common/Include.h:501:5: error: 'CHAR' does not name a type
/home/Projet/Kinesis/common/Include.h:502:5: error: 'PStackQueue' does not name a type
/home/Projet/Kinesis/common/Include.h:504:5: error: 'PCHAR' does not name a type
/home/Projet/Kinesis/common/Include.h:505:5: error: 'UINT32' does not name a type
/home/Projet/Kinesis/common/Include.h:522:16: warning: ISO C++ forbids declaration of 'STATUS' with no type [-fpermissive]
/home/Projet/Kinesis/common/Include.h:522:16: error: typedef 'STATUS' is initialized (use decltype instead)
/home/Projet/Kinesis/common/Include.h:522:18: error: 'GetCredentialsFunc' was not declared in this scope
/home/Projet/Kinesis/common/Include.h:527:5: error: 'GetCredentialsFunc' does not name a type
/home/Projet/Kinesis/common/Include.h:553:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:562:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:572:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:588:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:597:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:612:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:627:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:644:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:662:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:672:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:682:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:694:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:703:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:725:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:735:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:744:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:753:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:764:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:776:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/common/Include.h:787:1: error: 'PUBLIC_API' does not name a type
[... cut ...]
/home/Projet/Kinesis/cproducer/Include.h:691:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/cproducer/Include.h:700:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/cproducer/Include.h:713:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/cproducer/Include.h:722:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/cproducer/Include.h:739:1: error: 'PUBLIC_API' does not name a type
/home/Projet/Kinesis/cproducer/Include.h:753:1: error: 'PUBLIC_API' does not name a type
make: *** [Makefile:33: build/main.o] Error 1

hassanctech, 2021-03-12:
Are you able to build libproducerc.so ?  What I see is you have a problem linking against it but you are able to build it now, is that correct?

Also can you please put the logs in a text file and link to it, putting that large amount of text directly in the comment it's hard to see context because you have to scroll so much up and down.

waikup83, 2021-03-12:
yes ithe producer is build but the header foudn in src directory have problem to find type name
[Nouveau document texte.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/6132865/Nouveau.document.texte.txt)

closed 2021-04-07 by waikup83

hassanctech, 2021-04-07:
I see you closed the issue, so were you able to get unblocked to use the producer SDK?

waikup83, 2021-04-07:
.. so now i try to figure out how can i stream to kinesis directly with a char buffer and not by a file