awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #672: [QUESTION]Configuring Build directory fails

supriya-s-jadhav, 2021-03-03:
Could not complete the configuration of Build folder. Got error while I run cmake .. command. Below is the output

CMake Deprecation Warning at dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the system variable OPENSSL_ROOT_DIR (missing: OPENSSL_INCLUDE_DIR) 
CMake Error at dependency/libkvscproducer/kvscproducer-src/CMakeLists.txt:118 (message):
  OpenSSL is not found.  Make sure to export PKG_CONFIG_PATH to where
  OpenSSL's pc file is

-- Configuring incomplete, errors occurred!

hassanctech, 2021-03-05:
Did you try to pull down the repo, then create a `build` folder.  Go into `build` folder and run `cmake ..` is that what you did?  If that's not working then can you try with a newer version of CMake (it seems like you're using a really old version from over 7 years ago ?).  If I'm reading that error wrong can you share what version of CMake you're using?

supriya-s-jadhav, 2021-03-05:
@hassanctech Thank you for replying to my question. I just checked the cmake version installed on my mac and it is:
cmake version 3.19.4
I got the build folder working after multiple attempts.
Next, I am trying to get the public ip camera live feed into the KVS. I am using below link as rtsp source url: https://webcams.nyctmc.org/google_popup.php?cid=166
(Is it a valid url? If not how does a valid url look like?)
And I used below command as mentioned in AWS documentation:
**Stream Video from a network camera on macOS:**
gst-launch-1.0 rtspsrc location="https://webcams.nyctmc.org/google_popup.php?cid=166" short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! kvssink stream-name="MyStreamName" storage-size=512  access-key="MyAccessKey" secret-key="MySecretKey" aws-region="us-east-1"

And keep getting below email:

ERROR: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Resource not found.
Additional debug info:
../gst/rtsp/gstrtspsrc.c(8017): gst_rtspsrc_retrieve_sdp (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
No valid RTSP URL was provided
ERROR: pipeline doesn't want to preroll.
Setting pipeline to NULL ...
Freeing pipeline ...
INFO - Freeing Kinesis Video Stream EdgeDemoLiveFeed

I am wondering if I am headed in the right direction? If yes, what am I doing wrong? Thank you in advance for your time and assistance.

supriya-s-jadhav, 2021-03-06:
Update:
I am able to run run the test bigbunny rtsp url using the command as follows:
gst-launch-1.0 rtspsrc location="rtsp://34.227.104.115/vod/mp4:BigBuckBunny_115k.mov" ! rtph264depay ! h264parse ! kvssink stream-name="xxx" storage-size=512 access-key="xxx" secret-key="xxx" aws-region="us-west-2"

My requirement is to be able to run a public ip camera url, for example this http://nyctmc.org/google_popup.php?cid=166
I am wondering whether we have to change such urls to an rtsp url? Or how do I find a public rtsp url of a traffic facing camera?
AWS documentation does not have clear information on rtsp url and its use with GStreamer library. Thank you in advance for your time and help