awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #673: SDK version[3.1.0] Dropped Frame/Fragment[QUESTION]

derek1875, 2021-03-04:
Hi there,

The client shows lots of videos were lost. I 'm not sure whether it is the network issue, but the device could upload info/files to other cloud normally. Do you know what is wrong with this log？Does the log item `Number of latency pressure events` indicate the pressure of the network traffic？Thanks a lot.

Attached is the partial log.
[KVS_FrameFragment_Dropped issue.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/6081288/KVS_FrameFragment_Dropped.issue.txt)

hassanctech, 2021-03-04:
Can you turn on the most verbose logging, if you're using log4cplus config (as in the same) set it to `TRACE` for the SDK.

Also can you please try with release 3.1.1.  Specifically if you have multiple tracks in your stream there maybe relevant fixes in 3.1.1.

derek1875, 2021-03-08:
I will try with release 3.1.1. Also I will submit a new issue if I get more verbose logs :)

closed 2021-03-08 by derek1875