awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #674: [QUESTION] How to stream to kvs with both video and audio from an RTSP source.  Only getting video right now.

pineapplejoe, 2021-03-05:
We have a stream coming from an RTSP source. The stream has both audio and video. Confirmed by streaming to VLC which presents both. 

But when streaming through a raspberry pi - we are only getting video. How do we include audio?

This is the command we are using. 
gst-launch-1.0 rtspsrc location="rtsp://ipaddress/stream" ! rtph264depay ! h264parse ! kvssink stream-name="stream name"

hassanctech, 2021-03-05:
I'm not sure I understand so you have a RTSP stream which plays audio/video in VLC, but when streaming through a Raspberry Pi you only get video.   The one that worked was that coming from a different camera?  Are you running the `gst-launch-1.0` command ON the Raspberry Pi and the IP address is for a different camera so the Raspberry Pi is the thing pulling the RTSP stream and trying to send to `kvssink`?  If that's the case have you tried sending it to `filesink` and playback the file to make sure audio is being produced?

If audio works fine for `filesink` then can you set TRACE level logging in the log4cplus config that you set in the `kvssink` plugin and share that information?

pineapplejoe, 2021-03-05:
Yeah the setup with VLC is as follows: [IP camera ->PC on same network] .  Audio works. 

The Pi config is as follows: [IP camera->raspberry pi->internet cloud front end]. Audio doesn't work. 

Will try your suggestions. Thanks.  

So you don't see anything obviously wrong with the gstreamer command line?

supriya-s-jadhav, 2021-03-05:
@pineapplejoe Are you able to feed live stream video from rtsp url to kinesis video stream? I could stream the video from my mac to KVS.Now, I am trying to get the video stream from ip camera source to kvs and have no luck so far.
If you dont mind, could you share how you accomplish it and what hardware/system you used?

pineapplejoe, 2021-03-06:
I used a raspberry pi (which is basically your mac in this case).  There are a variety of ways to do it though. If you don't want to have the IP camera connected locally - you can connect to the camera from the cloud (like EC2) first then connect to kvs from there.

supriya-s-jadhav, 2021-03-06:
@pineapplejoe Thank you for your response. I have a GStreamer plugin installed on my mac and yes, I am trying to access the public ip camera in a different location. Well, I could try the way you suggested. 
I made some progress after my last comment here and I am able to run the below command using the test bigbunny rtsp url:

gst-launch-1.0 rtspsrc location="rtsp://34.227.104.115/vod/mp4:BigBuckBunny_115k.mov" ! rtph264depay ! h264parse ! kvssink stream-name="xxx" storage-size=512 access-key="xxx" secret-key="xxx" aws-region="us-west-2"

Is there a way to get an rtsp url for public ip camera live streaming traffic, for example I am using this link now http://nyctmc.org/google_popup.php?cid=166

pineapplejoe, 2021-03-08:
Thats not an rtsp source. that's just an updating image.

pineapplejoe, 2021-03-08:
@hassanctech I tried to use filesink. I had to modify the command line as shown below. There was no audio in the mp4 file. But I tried to connect to the camera directly with VLC and audio worked as expected. 

gst-launch-1.0  -e rtspsrc location="rtsp://Test:Test@192.168.1.18/live"  ! rtph264depay ! h264parse ! mp4mux ! filesink location=./file.mp4

I set the the following env var to get trace logging -  GST_DEBUG="*:7". Attached is the log zip file. 

[log.txt.zip](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/6098460/log.txt.zip)

hassanctech, 2021-03-08:
Ah oh I see the problem sorry I didn't notice earlier, so the `rtspsrc` element will output a pad for audio and a pad for video, you have to link to both of them.  The `rtph264depay` will extract h264 video, but you need something in parallel that will extract the audio.  See here:  https://gstreamer.freedesktop.org/documentation/rtp/index.html?gi-language=c

Determine which audio you have and find the correct `rtp*depay` to use, for example if you have PCMA audio you can use `rtppcmadepay` or if you have PCMU audio you can use `rtppcmudepay`, etc.

Let me see if I can find some examples...

disa6302, 2021-03-16:
Assuming question is answered, closing the issue.

closed 2021-03-16 by disa6302

Raj123456788, 2021-03-19:
Can someone post an example for getting the audio without writing it to file with gst command? I do get video
Stuck with gst-launch-1.0 rtspsrc location="rtsp://192.168.1.111:554/" short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! h264parse ! kvssink stream-name="stream" aws-region="east-1" retention-period=1