awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #675: [QUESTION]Sent uLaw audio data to KVS, mobile app playback has a lot of noise.

yhuwyze, 2021-03-09:
We used to send H.264+aLaw data to KVS, there is no problem when playback the audio track on the mobile app. Now we changed the audio format from aLaw to uLaw, when we playback the stream on mobile app, we heard a lot of noise there. On the mobile app side, we get the AAC format audio so KVS service must do some transcoding from aLaw/uLaw to AAC format. It looks like for me the server side still treat the audio track format as aLaw instead of uLaw.

We are using KVS SDK 3.1.0

Here is the code we used to setup audio content type to H.264 + uLaw, 
        STRNCPY(pstreamInfo->streamCaps.contentType, (PCHAR) MKV_H264_MULAW_MULTI_CONTENT_TYPE, MAX_CONTENT_TYPE_LEN);
Old code for aLaw is:
        STRNCPY(pstreamInfo->streamCaps.contentType, (PCHAR) "video/h264,audio/alaw", MAX_CONTENT_TYPE_LEN);

Do you have any suggestion how to setup the KVS audio stream format to uLaw? Did we missed anything there?

hassanctech, 2021-03-09:
We do not transcode in KVS.  Can you please tell me what leads you to believe the server side still treats the audio track as aLaw instead of uLaw?

yhuwyze, 2021-03-10:
I double checked with our mobile developer. when we get the stream like
streaming://b-647daf39.kinesisvideo.us-west-2.amazonaws.com/dash/v1/getMP4MediaFragment.mp4?SessionToken=***************************&SequenceNumber=1&TrackNumber=1
The content type returned was

“Content-Type” =  (
    “audio/aac”
  );
  
the audio data was actually not transcoded. just the original alaw or mulaw.

on the KVS device side, we sent either"video/h264,audio/alaw" or"video/h264,audio/mulaw". it looks like the KVS service did not pass the audio information to the client (mobile app) for some reason.

Do you have any idea why the mobile app get audio/aac?

hassanctech, 2021-03-10:
Which SDK are you using on the Mobile side?  Which API are you calling, is it GetDASHStreamingSessionURL?

hassanctech, 2021-03-10:
Also to be clear this happens for both alaw and mulaw that the content type returned is AAC?

The issue specific to alaw is the added noise, correct?

yhuwyze, 2021-03-10:
Answer from our mobile developer:

Question: "Which SDK are you using on the Mobile side? Which API are you calling, is it GetDASHStreamingSessionURL?"
It is our custom implementation to parse dash stream

Question: "Also to be clear this happens for both alaw and mulaw that the content type returned is AAC?"
Yes.

Question: "The issue specific to alaw is the added noise, correct?"
The problem is not specific to the format, content type is always audio/aac

hassanctech, 2021-03-10:
I've confirmed with our back-end team and this is currently the expected behavior because we have actually hard-coded “audio/aac” in the response no matter what the audio format is.  Is this causing you problems?

Next, as for the noise since we do not do any transcoding our understanding is the noise is introduced at your source and not by KVS, correct?

yhuwyze, 2021-03-10:
Thanks for the checking, it did cause some issue for us. original we were using aLaw for KVS streaming and MP4 file saved locally and uLaw for WebRTC, we have some compatibility issue to playback MP4 with aLaw format. also we don't want to keep both aLaw and uLaw data. so we switched to uLaw for all our audio data buffer. everything works fine except KVS streaming data. We don't have old device that are still using aLaw, since our mobile app has no way to know what exactly audio format it is. if it play aLaw data by uLaw format or uLaw data by aLaw format. you will hear a lot of noise. The noise was caused by the audio data and mismatched format.

hassanctech, 2021-03-10:
Ah ok I see!  Do you have a way now to determine if its alaw or mulaw so you can properly play it back?

yhuwyze, 2021-03-10:
i have no ways to do that. it looks like the KVS server always return audio/aac and codecs is 'alaw'

<Representation id="audio" bandwidth="64321" audioSamplingRate="8000" codecs="alaw">
        <AudioChannelConfiguration schemeIdUri="urn:mpeg:dash:23003:3:audio_channel_configuration:2011"/>
      </Representation>

yhuwyze, 2021-03-10:
Here the returned value:

`
<Representation id="audio" bandwidth="64321" audioSamplingRate="8000" codecs="alaw">
        <AudioChannelConfiguration schemeIdUri="urn:mpeg:dash:23003:3:audio_channel_configuration:2011"/>
      </Representation>`

hassanctech, 2021-03-11:
Understood yes this is a current limitation with our back-end, I've discussed with them and they look into it and determine prioritization.  I'm assigning this ticket to one of our back-end folks for now, we will update you when we have news to share.

MushMal, 2021-03-11:
In the meanwhile, your clients could be able to communicate this offline somehow or set stream tags?

yhuwyze, 2021-03-11:
Yes, we can work around this. Currently all our other modules use uLaw now, only KVS need additional transcode for alaw. We can use other communication channel to set the tag but i need extra overhead on both firmware and cloud/mobile side. eventually this is a problem we'd like to resolve. maybe KVS service can just pass the content type sent from the producer to the client.

niyatim23, 2022-03-10:
Closing since the discussion is stale, feel free to reopen / ask further questions if needed

closed 2022-03-10 by niyatim23