awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #676: [QUESTION]Stream/Upload mp4 format video from local to Kinesis Video Stream

supriya-s-jadhav, 2021-03-09:
I am trying to stream the video to the kinesis video stream and I feel like I have hit the dead end. I am looking for the solutions to below issues:

1) I am able to stream the big buck bunny video but I can't find the video clip in Media Library in KVS after it has completed running. Is there anything I am missing here? I could find the live video recorded using mac camera but not the test video, which was streaming successfully.

2) I am trying to stream the video on the local machine to kinesis video? Is this possible? I opened the video in vlc player and used the rtsp url generated by it (file->open network->URL) which looks something like this: "rtsp://46.16.226.181:88/?action=stream"  and run it using the same command that worked with BigBuckBunny test url:
gst-launch-1.0 rtspsrc location="rtsp://46.16.226.181:88/?action=stream" ! rtph264depay ! h264parse ! kvssink stream-name="DemoStream" storage-size=512 access-key="MyAccessKey" secret-key="MySecretKey " aws-region="us-east-1"

Looking for guidance so that I can make some progress to accomplish the above tasks. Thank you in advance for the time and help.

hassanctech, 2021-03-10:
Ok so if I understood correctly you have two videos that you are streaming using the `gst-launch-1.0` command, one using the big buck bunny test URL and one using your mac's webcam.  For the big buck bunny URL you cannot find the video in the KVS Console is that right?  If so, do you see the KVS stream getting created?  If streaming isn't working can you please elaborate if you see any error messages on the command line while running the `gst-launch-1.0` command?

It's possible it is a timecode issue, one thing you can do start the gst-launch-1.0 command, then go to your KVS Stream in the KVS console and look at the "Monitoring" tab and look at the metrics, specifically the "PutMedia Incoming Data" and "PutMedia Success" metrics, that will tell you if data is actually being received by KVS.  

For your second question, is the video coming from an IP camera or are you using the webcam?  For a webcam you might be able to find a different GStreamer source which can access it it would be unlikely you would need an RTSP url that would be for a remote camera.

If you're using the local to stream RTSP from a remote camera then yeah using VLC to get the RTSP URL should work.  But to test the pipeline I would always try replacing the `sink` element with something simpler like `filesink` just to see if the pipeline is otherwise OK.  If general if it works with `filesink` it should also with `kvssink`

supriya-s-jadhav, 2021-03-12:
Thank you for your response. Yes, there are two different scenarios involved. 

1) I am able to stream the big buck bunny video. After the complete streaming, I could see PutMedia Success value at 100%, and some values in PutMedia Incoming data ensuring that it did run completely and successfully. Now, the video should be there for 24h as that's the retentions period set. But, I seem to not find the video there. Is there anything I missed related to permissions or policies or anything else?

2)And, the second scenario is: I have a video clip on my laptop and I am trying to stream it in the kinesis video stream. To start with, I played that video in vlc player and got its RTSP URL ((file->open network->URL)) and the url looks like this : "rtsp://46.16.226.181:88/?action=stream" 
But it is giving me an error related to an unknown source/rtsp url. 
How should I make this scenario work: Stream a video clip sitting on the desktop to Kinesis Video Stream.

Let me know if I can provide any other information that will help in this matter. Thank you in advance.

disa6302, 2021-03-15:
@supriya-s-jadhav , 

The fragments can be accessed using the ListFragment and GetFragment APIs. The fragments would be available for the retention period for consumption/playback. Check [this](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/API_Operations_Amazon_Kinesis_Video_Streams_Archived_Media.html) section out for more information. 

For second one, I am a little confused, if you have a video clip, why do you need a RTSP URL? You can directly play the video clip file using `filesrc`.

disa6302, 2021-03-18:
Closing assuming question is answered.

closed 2021-03-18 by disa6302