awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #677: Unable to set the source to ready state.

vssharma79, 2021-03-10:
debug messages 
------------------
log4cplus:ERROR Unable to open file: ./log/kvs.log
[INFO ][2021-03-10 10:46:04] Using region: us-west-2
[INFO ][2021-03-10 10:46:04] Using aws credentials for Kinesis Video Streams
[INFO ][2021-03-10 10:46:04] No session token was detected.
[INFO ][2021-03-10 10:46:04] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2021-03-10 10:46:04] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ][2021-03-10 10:46:04] heapInitialize(): Creating AIV heap.
[INFO ][2021-03-10 10:46:04] heapInitialize(): Heap is initialized OK
[DEBUG][2021-03-10 10:46:04] stepStateMachine(): PIC Client State Machine - Current state: 0x1, Next state: 0x2
[DEBUG][2021-03-10 10:46:04] getSecurityTokenHandler invoked
[DEBUG][2021-03-10 10:46:04] Refreshing credentials. Force refreshing: 0 Now time is: 1615373164691667986 Expiration: 0
[INFO ][2021-03-10 10:46:04] New credentials expiration is 1615375564
[DEBUG][2021-03-10 10:46:04] stepStateMachine(): PIC Client State Machine - Current state: 0x2, Next state: 0x10
[INFO ][2021-03-10 10:46:04] createDeviceResultEvent(): Create device result event.
[DEBUG][2021-03-10 10:46:04] stepStateMachine(): PIC Client State Machine - Current state: 0x10, Next state: 0x40
[DEBUG][2021-03-10 10:46:04] clientReadyHandler invoked
[DEBUG][2021-03-10 10:46:04] Client is ready
[INFO ][2021-03-10 10:46:04] Creating Kinesis Video Stream webcam-video-stream
[INFO ][2021-03-10 10:46:04] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG][2021-03-10 10:46:04] stepStateMachine(): PIC Stream State Machine - Current state: 0x1, Next state: 0x2
[DEBUG][2021-03-10 10:46:04] stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
[INFO ][2021-03-10 10:46:05] writeHeaderCallback(): RequestId: 7fc17fb8-a212-423e-b566-7c26d0ae809a
[DEBUG][2021-03-10 10:46:05] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.615298195922E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:us-west-2:791299146659:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:791299146659:stream/webcam-video-stream/1615298195922","StreamName":"webcam-video-stream","Version":"iNaVH35InLO0ckDEN5Rb"}}
[INFO ][2021-03-10 10:46:05] describeStreamResultEvent(): Describe stream result event.
[DEBUG][2021-03-10 10:46:05] stepStateMachine(): PIC Stream State Machine - Current state: 0x2, Next state: 0x20
[DEBUG][2021-03-10 10:46:05] stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
[INFO ][2021-03-10 10:46:05] writeHeaderCallback(): RequestId: 270b0dda-aa87-4615-8826-f1c0e5a17bde
[DEBUG][2021-03-10 10:46:05] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com"}
[INFO ][2021-03-10 10:46:05] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG][2021-03-10 10:46:05] stepStateMachine(): PIC Stream State Machine - Current state: 0x20, Next state: 0x10
[DEBUG][2021-03-10 10:46:05] stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
[DEBUG][2021-03-10 10:46:05] getStreamingTokenHandler invoked
[DEBUG][2021-03-10 10:46:05] Refreshing credentials. Force refreshing: 1 Now time is: 1615373165544997645 Expiration: 1615375564
[INFO ][2021-03-10 10:46:05] New credentials expiration is 1615375565
[INFO ][2021-03-10 10:46:05] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG][2021-03-10 10:46:05] stepStateMachine(): PIC Stream State Machine - Current state: 0x10, Next state: 0x40
[DEBUG][2021-03-10 10:46:05] defaultStreamReadyCallback(): Reported streamReady callback for stream handle 94893459481545
[DEBUG][2021-03-10 10:46:05] streamReadyHandler invoked
[DEBUG][2021-03-10 10:46:05] Stream is ready
[INFO ][2021-03-10 10:46:05] Streaming from live source
[DEBUG][2021-03-10 10:46:05] Streaming with live source and width: 0, height: 0, fps: 25, bitrateInKBPS512
Unable to set the source to ready state.
[INFO ][2021-03-10 10:46:05] stopKinesisVideoStreamSync(): Synchronously stopping Kinesis Video Stream 0000564e1af1a3c9.
[DEBUG][2021-03-10 10:46:05] defaultStreamClosedCallback(): Reported streamClosed callback for stream handle 94893459481545. Upload handle 18446744073709551615
[DEBUG][2021-03-10 10:46:05] streamClosedHandler invoked for upload handle: 18446744073709551615
[INFO ][2021-03-10 10:46:05] Freeing Kinesis Video Stream webcam-video-stream
[INFO ][2021-03-10 10:46:05] freeKinesisVideoStream(): Freeing Kinesis Video stream.
[DEBUG][2021-03-10 10:46:06] defaultStreamShutdownCallback(): Reported streamShutdown callback for stream handle 94893459481545
[INFO ][2021-03-10 10:46:06] Freeing Kinesis Video Stream webcam-video-stream
[INFO ][2021-03-10 10:46:06] freeKinesisVideoClient(): Freeing Kinesis Video Client
log4cplus:ERROR Failed to open file ./log/kvs.log

vssharma79, 2021-03-10:
export GST_DEBUG=4 execution :
-------------------------------------
log4cplus:ERROR Unable to open file: ./log/kvs.log
[INFO ][2021-03-10 11:27:57] Using region: us-west-2
[INFO ][2021-03-10 11:27:57] Using aws credentials for Kinesis Video Streams
[INFO ][2021-03-10 11:27:57] No session token was detected.
[INFO ][2021-03-10 11:27:57] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2021-03-10 11:27:57] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ][2021-03-10 11:27:57] heapInitialize(): Creating AIV heap.
[INFO ][2021-03-10 11:27:57] heapInitialize(): Heap is initialized OK
[DEBUG][2021-03-10 11:27:57] stepStateMachine(): PIC Client State Machine - Current state: 0x1, Next state: 0x2
[DEBUG][2021-03-10 11:27:57] getSecurityTokenHandler invoked
[DEBUG][2021-03-10 11:27:57] Refreshing credentials. Force refreshing: 0 Now time is: 1615375677175535013 Expiration: 0
[INFO ][2021-03-10 11:27:57] New credentials expiration is 1615378077
[DEBUG][2021-03-10 11:27:57] stepStateMachine(): PIC Client State Machine - Current state: 0x2, Next state: 0x10
[INFO ][2021-03-10 11:27:57] createDeviceResultEvent(): Create device result event.
[DEBUG][2021-03-10 11:27:57] stepStateMachine(): PIC Client State Machine - Current state: 0x10, Next state: 0x40
[DEBUG][2021-03-10 11:27:57] clientReadyHandler invoked
[DEBUG][2021-03-10 11:27:57] Client is ready
[INFO ][2021-03-10 11:27:57] Creating Kinesis Video Stream webcam-video-stream
[INFO ][2021-03-10 11:27:57] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG][2021-03-10 11:27:57] stepStateMachine(): PIC Stream State Machine - Current state: 0x1, Next state: 0x2
[DEBUG][2021-03-10 11:27:57] stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
[INFO ][2021-03-10 11:27:57] writeHeaderCallback(): RequestId: 9a9788aa-bb70-4d10-9a45-994380d4a7b0
[DEBUG][2021-03-10 11:27:57] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.615298195922E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:us-west-2:791299146659:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:791299146659:stream/webcam-video-stream/1615298195922","StreamName":"webcam-video-stream","Version":"iNaVH35InLO0ckDEN5Rb"}}
[INFO ][2021-03-10 11:27:57] describeStreamResultEvent(): Describe stream result event.
[DEBUG][2021-03-10 11:27:57] stepStateMachine(): PIC Stream State Machine - Current state: 0x2, Next state: 0x20
[DEBUG][2021-03-10 11:27:57] stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
[INFO ][2021-03-10 11:27:57] writeHeaderCallback(): RequestId: 4429c225-6312-4f48-8af0-fb19aa316a09
[DEBUG][2021-03-10 11:27:57] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com"}
[INFO ][2021-03-10 11:27:58] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG][2021-03-10 11:27:58] stepStateMachine(): PIC Stream State Machine - Current state: 0x20, Next state: 0x10
[DEBUG][2021-03-10 11:27:58] stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
[DEBUG][2021-03-10 11:27:58] getStreamingTokenHandler invoked
[DEBUG][2021-03-10 11:27:58] Refreshing credentials. Force refreshing: 1 Now time is: 1615375678021119523 Expiration: 1615378077
[INFO ][2021-03-10 11:27:58] New credentials expiration is 1615378078
[INFO ][2021-03-10 11:27:58] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG][2021-03-10 11:27:58] stepStateMachine(): PIC Stream State Machine - Current state: 0x10, Next state: 0x40
[DEBUG][2021-03-10 11:27:58] defaultStreamReadyCallback(): Reported streamReady callback for stream handle 94814261121993
[DEBUG][2021-03-10 11:27:58] streamReadyHandler invoked
[DEBUG][2021-03-10 11:27:58] Stream is ready
0:00:00.000067092 136769 0x563baa59ce00 INFO                GST_INIT gst.c:586:init_pre: Initializing GStreamer Core Library version 1.16.2
0:00:00.000109257 136769 0x563baa59ce00 INFO                GST_INIT gst.c:587:init_pre: Using library installed in /usr/lib/x86_64-linux-gnu
0:00:00.000125681 136769 0x563baa59ce00 INFO                GST_INIT gst.c:605:init_pre: Linux ip-172-31-29-203 5.4.0-1038-aws #40-Ubuntu SMP Fri Feb 5 23:50:40 UTC 2021 x86_64
0:00:00.000333251 136769 0x563baa59ce00 INFO                GST_INIT gstmessage.c:128:_priv_gst_message_initialize: init messages
0:00:00.000860087 136769 0x563baa59ce00 INFO                GST_INIT gstcontext.c:84:_priv_gst_context_initialize: init contexts
0:00:00.001086454 136769 0x563baa59ce00 INFO      GST_PLUGIN_LOADING gstplugin.c:318:_priv_gst_plugin_initialize: registering 0 static plugins
0:00:00.001236022 136769 0x563baa59ce00 INFO      GST_PLUGIN_LOADING gstplugin.c:226:gst_plugin_register_static: registered static plugin "staticelements"
0:00:00.001293453 136769 0x563baa59ce00 INFO      GST_PLUGIN_LOADING gstplugin.c:228:gst_plugin_register_static: added static plugin "staticelements", result: 1
0:00:00.001349729 136769 0x563baa59ce00 INFO            GST_REGISTRY gstregistry.c:1720:ensure_current_registry: reading registry cache: /home/ubuntu/.cache/gstreamer-1.0/registry.x86_64.bin
[... cut ...]
[INFO ][2021-03-10 11:27:58] Freeing Kinesis Video Stream webcam-video-stream
[INFO ][2021-03-10 11:27:58] freeKinesisVideoStream(): Freeing Kinesis Video stream.
[DEBUG][2021-03-10 11:27:58] defaultStreamShutdownCallback(): Reported streamShutdown callback for stream handle 94814261121993
[INFO ][2021-03-10 11:27:58] Freeing Kinesis Video Stream webcam-video-stream
[INFO ][2021-03-10 11:27:58] freeKinesisVideoClient(): Freeing Kinesis Video Client
log4cplus:ERROR Failed to open file ./log/kvs.log

vssharma79, 2021-03-10:
Error : log4cplus:ERROR Unable to open file: ./log/kvs.log has been resolved as I've created log folder. But other errors are still there

hassanctech, 2021-03-10:
I gather you are using the `kvssink` element.  Can you please some more details like what your pipeline exactly looks like, what system are you running this on, what commit / release version you're using, etc.

vssharma79, 2021-03-11:
I'm using kinesis video stream to capture the live video from laptop  webcam and push to kinesis video stream for further analysis.

vssharma79, 2021-03-12:
There is some progress and issue solved but I'm getting below error about camara device not configured:

root@ip-172-31-29-203:/home/ubuntu# gst-inspect-1.0 kvssink
Factory Details:
  Rank                     primary + 10 (266)
  Long-name                KVS Sink
  Klass                    Sink/Video/Network
  Description              GStreamer AWS KVS plugin
  Author                   AWS KVS <kinesis-video-support@amazon.com>

Plugin Details:
  Name                     kvssink
  Description              GStreamer AWS KVS plugin
  Filename                 /home/ubuntu/amazon-kinesis-video-streams-producer-sdk-cpp/build/libgstkv                                                  ssink.so
  Version                  1.0
  License                  Proprietary
  Source module            kvssinkpackage
  Binary package           GStreamer
  Origin URL               http://gstreamer.net/

GObject
 +----GInitiallyUnowned
       +----GstObject
             +----GstElement
                   +----GstKvsSink

root@ip-172-31-29-203:/home/ubuntu#
root@ip-172-31-29-203:/home/ubuntu# gst-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! x264enc  bframes=0 key-int-max=45 bitrate=500 ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink stream-name="webcam-video-stream" storage-size=512 access-key="AKIA3QPI2YOR3YADW3UA" secret-key="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" aws-region=us-east-1
Setting pipeline to PAUSED ...
log4cplus:ERROR could not open file ./kvs_log_configuration
INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2021-03-12 07:07:01 [140646992290880] INFO - heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
2021-03-12 07:07:01 [140646992290880] INFO - heapInitialize(): Creating AIV heap.
2021-03-12 07:07:01 [140646992290880] INFO - heapInitialize(): Heap is initialized OK
2021-03-12 07:07:01 [140646992290880] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2021-03-12 07:07:01 [140646992290880] DEBUG - getSecurityTokenHandler invoked
2021-03-12 07:07:01 [140646992290880] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1615532821376939155 Expiration: 0
2021-03-12 07:07:01 [140646992290880] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
2021-03-12 07:07:01 [140646992290880] INFO - createDeviceResultEvent(): Create device result event.
2021-03-12 07:07:01 [140646992290880] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
2021-03-12 07:07:01 [140646992290880] DEBUG - clientReadyHandler invoked
2021-03-12 07:07:01 [140646992290880] INFO - try creating stream
2021-03-12 07:07:01 [140646992290880] INFO - Creating Kinesis Video Stream webcam-video-stream
2021-03-12 07:07:01 [140646992290880] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2021-03-12 07:07:01 [140646992290880] DEBUG - logStreamInfo(): Kinesis Video Stream Info
2021-03-12 07:07:01 [140646992290880] DEBUG - logStreamInfo():  Stream name: webcam-video-stream
2021-03-12 07:07:01 [140646992290880] DEBUG - logStreamInfo():  Streaming type: STREAMING_TYPE_REALTIME
2021-03-12 07:07:01 [140646992290880] DEBUG - logStreamInfo():  Content type: video/h264
2021-03-12 07:07:01 [140646992290880] DEBUG - logStreamInfo():  Max latency (100ns): 600000000
2021-03-12 07:07:01 [140646992290880] DEBUG - logStreamInfo():  Fragment duration (100ns): 20000000
2021-03-12 07:07:01 [140646992290880] DEBUG - logStreamInfo():  Key frame fragmentation: Yes
2021-03-12 07:07:01 [140646992290880] DEBUG - logStreamInfo():  Use frame timecode: Yes
2021-03-12 07:07:01 [140646992290880] DEBUG - logStreamInfo():  Absolute frame timecode: Yes
2021-03-12 07:07:01 [140646992290880] DEBUG - logStreamInfo():  Nal adaptation flags: 0
2021-03-12 07:07:01 [140646992290880] DEBUG - logStreamInfo():  Average bandwith (bps): 4194304
2021-03-12 07:07:01 [140646992290880] DEBUG - logStreamInfo():  Framerate: 25
2021-03-12 07:07:01 [140646992290880] DEBUG - logStreamInfo():  Buffer duration (100ns): 1200000000
2021-03-12 07:07:01 [140646992290880] DEBUG - logStreamInfo():  Replay duration (100ns): 400000000
2021-03-12 07:07:01 [140646992290880] DEBUG - logStreamInfo():  Connection Staleness duration (100ns): 600000000
2021-03-12 07:07:01 [140646992290880] DEBUG - logStreamInfo():  Store Pressure Policy: 1
2021-03-12 07:07:01 [140646992290880] DEBUG - logStreamInfo():  View Overflow Policy: 1
2021-03-12 07:07:01 [140646992290880] DEBUG - logStreamInfo():  Segment UUID: NULL
2021-03-12 07:07:01 [140646992290880] DEBUG - logStreamInfo():  Frame ordering mode: 0
2021-03-12 07:07:01 [140646992290880] DEBUG - logStreamInfo(): Track list
2021-03-12 07:07:01 [140646992290880] DEBUG - logStreamInfo():  Track id: 1
2021-03-12 07:07:01 [140646992290880] DEBUG - logStreamInfo():  Track name: kinesis_video
2021-03-12 07:07:01 [140646992290880] DEBUG - logStreamInfo():  Codec id: V_MPEG4/ISO/AVC
2021-03-12 07:07:01 [140646992290880] DEBUG - logStreamInfo():  Track type: TRACK_INFO_TYPE_VIDEO
2021-03-12 07:07:01 [140646992290880] DEBUG - logStreamInfo():  Track cpd: NULL
2021-03-12 07:07:01 [140646992290880] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[... cut ...]
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Could not open device '/dev/video0' for reading and writing.
Additional debug info:
v4l2_calls.c(621): gst_v4l2_open (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
system error: No such device or address
Setting pipeline to NULL ...
Freeing pipeline ...
INFO - Freeing Kinesis Video Stream webcam-video-stream
2021-03-12 07:07:01 [140646992290880] INFO - freeKinesisVideoStream(): Freeing Kinesis Video stream.
2021-03-12 07:07:01 [140646992290880] DEBUG - curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty

MushMal, 2021-03-12:
None of this is KVS related. Could you remove KVS sink from the pipeline and stabilize it with fakesink first. Please resolve this if you have no further questions related to KVS itself. GStreamer forums might be a better source for help regarding the pipeline

vssharma79, 2021-03-12:
I didn't get you, what correction need to make to work video stream?

MushMal, 2021-03-12:
I am not sure what corrections your pipeline needs in order to work but you can debug it a lot easier without KVS being the sink. Set the sink as fakesink, debug your pipeline. Once working, use KVS

hassanctech, 2021-03-12:
As @MushMal stated this is the general guidance.  Always try your pipeline with something simple, my recommendation is to use `filesink`. Once it is working properly with `filesink` then replace with `kvssink`. If there are issues then we know they are specific to `kvssink` and we can debug further.  

As of now the error is:

ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Could not open device '/dev/video0' for reading and writing.
Additional debug info:
v4l2_calls.c(621): gst_v4l2_open (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
system error: No such device or address

It is clear that you have an issue with your pipeline not related to `kvssink`

Perhaps you can start with figuring out with why you can't open /dev/video0, is that the right thing, maybe you need to properly configure your source element so it can recognize your media source.

I'm going to close this issue for now because there is no KVS issue.  Once you've tested it with `filesink` and you know it works there and if it doesn't work with `kvssink` please open a new issue with that information.

Finally for more guidance I suggest you visit gstreamer specific forums to help you get your media source started.

closed 2021-03-12 by hassanctech