awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #678: [QUESTION] Could not resolve host: kinesisvideo.us-east1.amazonaws.com

sonln5, 2021-03-11:
Dear all,
I'm running a Docker container which launch a gstreamer pipeline to stream a rtsp camera to kinesis video stream. The pipeline was:
gst-launch-1.0 rtspsrc location="rtsp://admin:123456aA@192.168.40.6:554" short-header=TRUE ! rtph264depay ! video/x-h264,stream-format=avc,alignment=au,width=1280,height=720 ! kvssink stream-name="XXX" storage-size=512 access-key=XXXX secret-key=XXX aws-region=us-east1
(I removed the information about AWS access credentials).

This pipeline is OK on my real PC but I got the bellow error when running on a Docker container:
"**2021-03-11 09:15:45 [140642044863296] WARN - curlCompleteSync(): curl perform failed for url https://kinesisvideo.us-east1.amazonaws.com/describeStream with result Couldn't resolve host name: Could not resolve host: kinesisvideo.us-east1.amazonaws.com
2021-03-11 09:15:45 [140641310746368] WARN - curlCompleteSync(): HTTP Error 0 : Response: (null)**"

** Logging **
Setting pipeline to PAUSED ...
log4cplus:ERROR could not open file ./kvs_log_configuration
INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2021-03-11 09:15:45 [140642044863296] INFO - heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
2021-03-11 09:15:45 [140642044863296] INFO - heapInitialize(): Creating AIV heap.
2021-03-11 09:15:45 [140642044863296] INFO - heapInitialize(): Heap is initialized OK
2021-03-11 09:15:45 [140642044863296] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2021-03-11 09:15:45 [140642044863296] DEBUG - getSecurityTokenHandler invoked
2021-03-11 09:15:45 [140642044863296] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1615454145204041542 Expiration: 0
2021-03-11 09:15:45 [140642044863296] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
2021-03-11 09:15:45 [140642044863296] INFO - createDeviceResultEvent(): Create device result event.
2021-03-11 09:15:45 [140642044863296] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
2021-03-11 09:15:45 [140642044863296] DEBUG - clientReadyHandler invoked
2021-03-11 09:15:45 [140642044863296] INFO - try creating stream
2021-03-11 09:15:45 [140642044863296] INFO - Creating Kinesis Video Stream frida
2021-03-11 09:15:45 [140642044863296] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2021-03-11 09:15:45 [140642044863296] DEBUG - logStreamInfo(): Kinesis Video Stream Info
2021-03-11 09:15:45 [140642044863296] DEBUG - logStreamInfo(): 	Stream name: frida 
2021-03-11 09:15:45 [140642044863296] DEBUG - logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME 
2021-03-11 09:15:45 [140642044863296] DEBUG - logStreamInfo(): 	Content type: video/h264 
2021-03-11 09:15:45 [140642044863296] DEBUG - logStreamInfo(): 	Max latency (100ns): 600000000
2021-03-11 09:15:45 [140642044863296] DEBUG - logStreamInfo(): 	Fragment duration (100ns): 20000000
2021-03-11 09:15:45 [140642044863296] DEBUG - logStreamInfo(): 	Key frame fragmentation: Yes
2021-03-11 09:15:45 [140642044863296] DEBUG - logStreamInfo(): 	Use frame timecode: Yes
2021-03-11 09:15:45 [140642044863296] DEBUG - logStreamInfo(): 	Absolute frame timecode: Yes
2021-03-11 09:15:45 [140642044863296] DEBUG - logStreamInfo(): 	Nal adaptation flags: 0
2021-03-11 09:15:45 [140642044863296] DEBUG - logStreamInfo(): 	Average bandwith (bps): 4194304
2021-03-11 09:15:45 [140642044863296] DEBUG - logStreamInfo(): 	Framerate: 25
2021-03-11 09:15:45 [140642044863296] DEBUG - logStreamInfo(): 	Buffer duration (100ns): 1200000000
2021-03-11 09:15:45 [140642044863296] DEBUG - logStreamInfo(): 	Replay duration (100ns): 400000000
2021-03-11 09:15:45 [140642044863296] DEBUG - logStreamInfo(): 	Connection Staleness duration (100ns): 600000000
2021-03-11 09:15:45 [140642044863296] DEBUG - logStreamInfo(): 	Store Pressure Policy: 1
2021-03-11 09:15:45 [140642044863296] DEBUG - logStreamInfo(): 	View Overflow Policy: 1
2021-03-11 09:15:45 [140642044863296] DEBUG - logStreamInfo(): 	Segment UUID: NULL
2021-03-11 09:15:45 [140642044863296] DEBUG - logStreamInfo(): 	Frame ordering mode: 0
2021-03-11 09:15:45 [140642044863296] DEBUG - logStreamInfo(): Track list
2021-03-11 09:15:45 [140642044863296] DEBUG - logStreamInfo(): 	Track id: 1
2021-03-11 09:15:45 [140642044863296] DEBUG - logStreamInfo(): 	Track name: kinesis_video
2021-03-11 09:15:45 [140642044863296] DEBUG - logStreamInfo(): 	Codec id: V_MPEG4/ISO/AVC
2021-03-11 09:15:45 [140642044863296] DEBUG - logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
2021-03-11 09:15:45 [140642044863296] DEBUG - logStreamInfo(): 	Track cpd: NULL
2021-03-11 09:15:45 [140642044863296] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2021-03-11 09:15:45 [140642044863296] WARN - curlCompleteSync(): curl perform failed for url https://kinesisvideo.us-east1.amazonaws.com/describeStream with result Couldn't resolve host name: Could not resolve host: kinesisvideo.us-east1.amazonaws.com
2021-03-11 09:15:45 [140641310746368] WARN - curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://kinesisvideo.us-east1.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=XXXXX/20210311/us-east1/kinesisvideo/aws4_request, SignedHeaders=host;user-agent;x-amz-date, Signature=XXXXX
    content-length: 26
    content-type: application/json
    host: kinesisvideo.us-east1.amazonaws.com
    user-agent: AWS-SDK-KVS/3.0.0 G
2021-03-11 09:15:45 [140641310746368] DEBUG - describeStreamCurlHandler(): DescribeStream API response: 
2021-03-11 09:15:45 [140641310746368] INFO - describeStreamResultEvent(): Describe stream result event.
2021-03-11 09:15:45 [140641310746368] WARN - curlCompleteSync(): curl perform failed for url https://kinesisvideo.us-east1.amazonaws.com/describeStream with result Couldn't resolve host name: Could not resolve host: kinesisvideo.us-east1.amazonaws.com
2021-03-11 09:15:45 [140641302353664] WARN - curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://kinesisvideo.us-east1.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=XXXXXX/20210311/us-east1/kinesisvideo/aws4_request, SignedHeaders=host;user-agent;x-amz-date, Signature=XXXXX
    content-length: 26
    content-type: application/json
    host: kinesisvideo.us-east1.amazonaws.com
    user-agent: AWS-SDK-KVS/3.0.0 G

Thanks and Best Regards!

disa6302, 2021-03-11:
@sonln5 ,

Pleasev check the network config for DNS resolution. It is not a KVS specific issue.

sonln5, 2021-03-11:
On docker contain, I tried pinging to some other hosts and it was OK. So I think it's not DNS resolution issue.
Thanks!

MushMal, 2021-03-11:
@sonln5 please debug your rocker/network configuration. This has got nothing to do with KVS and you should be able to reduce the problem down. Try running the samples with no modifications on stock now without rocket first. 

Resolving as non KVS related

closed 2021-03-11 by MushMal