awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #684: [BUG] Single stream works but multistream doesn't. "Source link FAILED" Error.

chestonchow, 2021-03-16:
Thanks for having a look!

**To Reproduce**
(No local source here)
Steps to reproduce the behavior:
1. Clone the repo, build, export environment variable according to the readme.
2. `cd` to the `build` folder
3. Add a file named `rtsp-url-file` with a testing RTSP link (or repeating links). I'm repeating twice here.
```
rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov
rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov
```
I have tested the link with VLC player. The link is fine.
4. Run `./kvs_gstreamer_multistream_sample camera rtsp-url-file`
5. What I got was the following and it never recovers. 
**Tl:dr:** It fails after two lines of `Source link FAILED` and gives the `Failed to submit frame to Kinesis Video client. status: 0x3200000b` error and keeps printing the same.
I repeated the same everyday for the last whole week or two since it started to fail.
```
~/amazon-kinesis-video-streams-producer-sdk-cpp/build$ ./kvs_gstreamer_multistream_sample camera rtsp-url-file 
[INFO ] [15-03-2021 00:20:31:401.396 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [15-03-2021 00:20:31:422.105 GMT] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [15-03-2021 00:20:31:422.168 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [15-03-2021 00:20:31:422.196 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [15-03-2021 00:20:31:422.339 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[DEBUG] [15-03-2021 00:20:31:422.399 GMT] getSecurityTokenHandler invoked
[DEBUG] [15-03-2021 00:20:31:422.448 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1615767631422425026 Expiration: 0
[INFO ] [15-03-2021 00:20:31:422.464 GMT] New credentials expiration is 1615771231
[DEBUG] [15-03-2021 00:20:31:422.493 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[INFO ] [15-03-2021 00:20:31:422.526 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [15-03-2021 00:20:31:422.547 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [15-03-2021 00:20:31:422.565 GMT] clientReadyHandler invoked
[DEBUG] [15-03-2021 00:20:31:422.611 GMT] Client is ready
[INFO ] [15-03-2021 00:20:31:422.684 GMT] Creating Kinesis Video Stream camera_0
[INFO ] [15-03-2021 00:20:31:422.726 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG] [15-03-2021 00:20:31:422.748 GMT] logStreamInfo(): Kinesis Video Stream Info
[DEBUG] [15-03-2021 00:20:31:422.770 GMT] logStreamInfo():      Stream name: camera_0 
[DEBUG] [15-03-2021 00:20:31:422.789 GMT] logStreamInfo():      Streaming type: STREAMING_TYPE_REALTIME 
[DEBUG] [15-03-2021 00:20:31:422.807 GMT] logStreamInfo():      Content type: video/h264 
[DEBUG] [15-03-2021 00:20:31:422.819 GMT] logStreamInfo():      Max latency (100ns): 600000000
[DEBUG] [15-03-2021 00:20:31:422.831 GMT] logStreamInfo():      Fragment duration (100ns): 20000000
[DEBUG] [15-03-2021 00:20:31:422.842 GMT] logStreamInfo():      Key frame fragmentation: Yes
[DEBUG] [15-03-2021 00:20:31:422.853 GMT] logStreamInfo():      Use frame timecode: Yes
[DEBUG] [15-03-2021 00:20:31:422.864 GMT] logStreamInfo():      Absolute frame timecode: Yes
[DEBUG] [15-03-2021 00:20:31:422.875 GMT] logStreamInfo():      Nal adaptation flags: 0
[DEBUG] [15-03-2021 00:20:31:422.905 GMT] logStreamInfo():      Average bandwith (bps): 4194304
[DEBUG] [15-03-2021 00:20:31:422.917 GMT] logStreamInfo():      Framerate: 25
[DEBUG] [15-03-2021 00:20:31:422.933 GMT] logStreamInfo():      Buffer duration (100ns): 1200000000
[DEBUG] [15-03-2021 00:20:31:422.949 GMT] logStreamInfo():      Replay duration (100ns): 400000000
[DEBUG] [15-03-2021 00:20:31:422.963 GMT] logStreamInfo():      Connection Staleness duration (100ns): 600000000
[DEBUG] [15-03-2021 00:20:31:422.982 GMT] logStreamInfo():      Store Pressure Policy: 1
[DEBUG] [15-03-2021 00:20:31:423.001 GMT] logStreamInfo():      View Overflow Policy: 1
[DEBUG] [15-03-2021 00:20:31:423.013 GMT] logStreamInfo():      Segment UUID: NULL
[DEBUG] [15-03-2021 00:20:31:423.024 GMT] logStreamInfo():      Frame ordering mode: 0
[DEBUG] [15-03-2021 00:20:31:423.034 GMT] logStreamInfo(): Track list
[DEBUG] [15-03-2021 00:20:31:423.045 GMT] logStreamInfo():      Track id: 1
[DEBUG] [15-03-2021 00:20:31:423.056 GMT] logStreamInfo():      Track name: kinesis_video
[DEBUG] [15-03-2021 00:20:31:423.068 GMT] logStreamInfo():      Codec id: V_MPEG4/ISO/AVC
[DEBUG] [15-03-2021 00:20:31:423.078 GMT] logStreamInfo():      Track type: TRACK_INFO_TYPE_VIDEO
[DEBUG] [15-03-2021 00:20:31:423.089 GMT] logStreamInfo():      Track cpd: NULL
[DEBUG] [15-03-2021 00:20:31:423.138 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[INFO ] [15-03-2021 00:20:31:621.457 GMT] writeHeaderCallback(): RequestId: 3217a533-f408-4a02-815d-8245814949d7
[DEBUG] [15-03-2021 00:20:31:621.706 GMT] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.615443323305E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:ap-southeast-2:196616580645:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:ap-southeast-2:196616580645:stream/camera_0/1615443323305","StreamName":"camera_0","Version":"dFOHy4JjEXBRLF8d0ZXj"}}
[INFO ] [15-03-2021 00:20:31:623.498 GMT] describeStreamResultEvent(): Describe stream result event.
[DEBUG] [15-03-2021 00:20:31:623.593 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
[INFO ] [15-03-2021 00:20:31:823.057 GMT] writeHeaderCallback(): RequestId: 0c081ab1-89d5-4e64-b581-0bee0d70a2eb
[DEBUG] [15-03-2021 00:20:31:823.151 GMT] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-ca658586.kinesisvideo.ap-southeast-2.amazonaws.com"}
[INFO ] [15-03-2021 00:20:31:823.845 GMT] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG] [15-03-2021 00:20:31:823.878 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
[DEBUG] [15-03-2021 00:20:31:823.895 GMT] getStreamingTokenHandler invoked
[DEBUG] [15-03-2021 00:20:31:823.912 GMT] Refreshing credentials. Force refreshing: 1 Now time is: 1615767631823909200 Expiration: 1615771231
[... 598 lines cut ...]
        >> Current view byte size: 16362
        >> Overall view byte size: 34605
        >> Current frame rate (fps): 3433.18
        >> Current transfer rate (bps): 17259080 (16854 Kbps)
[DEBUG] [14-03-2021 10:18:23:878.148 GMT] postReadCallback(): Wrote 18716 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [14-03-2021 10:18:23:878.517 GMT] postReadCallback(): Wrote 501 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [14-03-2021 10:18:23:878.625 GMT] postReadCallback(): Wrote 655 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [14-03-2021 10:18:23:878.741 GMT] postReadCallback(): Wrote 580 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [14-03-2021 10:18:23:878.826 GMT] postReadCallback(): Pausing CURL read for upload handle: 0
[DEBUG] [14-03-2021 10:18:24:046.456 GMT] postWriteCallback(): Curl post body write function for stream with handle: camera_0 and upload handle: 0 returned: {"EventType":"RECEIVED","FragmentTimecode":1615717087431,"FragmentNumber":"91343852333181704789008306990700655604946111197"}

[DEBUG] [14-03-2021 10:18:24:046.638 GMT] fragmentAckReceivedHandler invoked
[DEBUG] [14-03-2021 10:18:24:046.694 GMT] Reporting fragment ack received. Ack timecode 1615717087431
[DEBUG] [14-03-2021 10:18:24:046.771 GMT] postReadCallback(): Wrote 2095 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [14-03-2021 10:18:24:046.981 GMT] postReadCallback(): Pausing CURL read for upload handle: 0
[DEBUG] [14-03-2021 10:18:24:056.232 GMT] postWriteCallback(): Curl post body write function for stream with handle: camera_0 and upload handle: 0 returned: {"EventType":"BUFFERING","FragmentTimecode":1615717089420,"FragmentNumber":"91343852333181704793960067147842177312379710818"}

[DEBUG] [14-03-2021 10:18:24:056.364 GMT] fragmentAckReceivedHandler invoked
[... cut ...]
[DEBUG] [14-03-2021 10:18:24:126.474 GMT] postWriteCallback(): Curl post body write function for stream with handle: camera_0 and upload handle: 0 returned: {"EventType":"PERSISTED","FragmentTimecode":1615717087431,"FragmentNumber":"91343852333181704789008306990700655604946111197"}

[DEBUG] [14-03-2021 10:18:24:136.656 GMT] fragmentAckReceivedHandler invoked
[DEBUG] [14-03-2021 10:18:24:136.735 GMT] Reporting fragment ack received. Ack timecode 1615717087431
[DEBUG] [14-03-2021 10:18:24:136.778 GMT] postReadCallback(): Pausing CURL read for upload handle: 0
```

MushMal, 2021-03-16:
0x3200000b is STATUS_MKV_MAX_FRAME_TIMECODE. This happens when the SDK can't represent a frame timestamp within the 32bits (relative to the start of the cluster). Most likely the cause is missing key-frames.

I am not sure why the linkage of the pad fails - seems like the video pad is linked just fine.

At any rate, playable doesn't yet mean streamable media. KVS has a strict requirement for streamability and packaging.

Please ensure your media source has proper key-frame intervals.

P.S. 0x32000002 is an indication of an invalid timestamp - it's not immediately clear why the pipeline outputs bad timecodes initially.

As this is not a KVS bug, I am removing the 'bug' tag. Please work on debugging your source.

chestonchow, 2021-03-17:
Hi Mushegh,

Good to know that it’s not a KVS issue! I will have to check out other rtsp links since I don’t have control over this one. Since you are here, can I ask how accurate does the time stamp have to be for it to stream normally? Or is it a parameter tunable somewhere in the code?

Thanks,
Cheston

MushMal, 2021-03-17:
They don't need to be "accurate" per-say. The main issue is that

* The fragmentation happens on an I-frame boundary. Some encoded streams have very large GoPs. KVS has a limit of 10 second fragment durations. Moreover, the packaging technology we use (Matroska/MKV) uses 15bits to represent the frame timestamp relative to the beginning of the cluster (fragment) in default time units (1ms). So, the first issue is to ensure the source produces up-to 10 second GoPs. Many encoders are configured via durations while others are configured by specifying the GoP size in the number of frames.
* TImestamps should have monotonically increasing PTS (presentation timestamps) - this is important for the stramability and indexing.

More info can be found here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/timestamps.md

chestonchow, 2021-03-17:
Many Thanks Mushegh!

closed 2021-03-17 by chestonchow