awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #685: [QUESTION] How do I squash the stream output messages?

M-Davies, 2021-03-16:
**I would like to be able to squash the Kinesis stream debug messages. I am unsure if this is the correct place to raise this.**

** Logging **
As an example, after running `gst-launch` I receive a ton of noise that essentially tells me the stream is running. This noise continues throughout the stream. I would like to be able to pick which type of messages (e.g. `INFO`, `WARN` etc) are dumped where all others are squashed (preferable option).
```
log4cplus:ERROR could not open file ./kvs_log_configuration
INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2021-03-16 17:48:41 [0x113249e00] INFO - heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
2021-03-16 17:48:41 [0x113249e00] INFO - heapInitialize(): Creating AIV heap.
2021-03-16 17:48:41 [0x113249e00] INFO - heapInitialize(): Heap is initialized OK
2021-03-16 17:48:41 [0x113249e00] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2021-03-16 17:48:41 [0x113249e00] DEBUG - getSecurityTokenHandler invoked
2021-03-16 17:48:41 [0x113249e00] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1615916921776467 Expiration: 0
2021-03-16 17:48:41 [0x113249e00] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
2021-03-16 17:48:41 [0x113249e00] INFO - createDeviceResultEvent(): Create device result event.
2021-03-16 17:48:41 [0x113249e00] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
2021-03-16 17:48:41 [0x113249e00] DEBUG - clientReadyHandler invoked
2021-03-16 17:48:41 [0x113249e00] INFO - try creating stream
2021-03-16 17:48:41 [0x113249e00] INFO - Creating Kinesis Video Stream CameraVideoStream
2021-03-16 17:48:41 [0x113249e00] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2021-03-16 17:48:41 [0x113249e00] DEBUG - logStreamInfo(): Kinesis Video Stream Info
2021-03-16 17:48:41 [0x113249e00] DEBUG - logStreamInfo(): 	Stream name: CameraVideoStream 
2021-03-16 17:48:41 [0x113249e00] DEBUG - logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME 
2021-03-16 17:48:41 [0x113249e00] DEBUG - logStreamInfo(): 	Content type: video/h264 
2021-03-16 17:48:41 [0x113249e00] DEBUG - logStreamInfo(): 	Max latency (100ns): 600000000
2021-03-16 17:48:41 [0x113249e00] DEBUG - logStreamInfo(): 	Fragment duration (100ns): 20000000
2021-03-16 17:48:41 [0x113249e00] DEBUG - logStreamInfo(): 	Key frame fragmentation: Yes
2021-03-16 17:48:41 [0x113249e00] DEBUG - logStreamInfo(): 	Use frame timecode: Yes
2021-03-16 17:48:41 [0x113249e00] DEBUG - logStreamInfo(): 	Absolute frame timecode: Yes
2021-03-16 17:48:41 [0x113249e00] DEBUG - logStreamInfo(): 	Nal adaptation flags: 0
2021-03-16 17:48:41 [0x113249e00] DEBUG - logStreamInfo(): 	Average bandwith (bps): 4194304
2021-03-16 17:48:41 [0x113249e00] DEBUG - logStreamInfo(): 	Framerate: 25
2021-03-16 17:48:41 [0x113249e00] DEBUG - logStreamInfo(): 	Buffer duration (100ns): 1200000000
2021-03-16 17:48:41 [0x113249e00] DEBUG - logStreamInfo(): 	Replay duration (100ns): 400000000
2021-03-16 17:48:41 [0x113249e00] DEBUG - logStreamInfo(): 	Connection Staleness duration (100ns): 600000000
2021-03-16 17:48:41 [0x113249e00] DEBUG - logStreamInfo(): 	Store Pressure Policy: 1
2021-03-16 17:48:41 [0x113249e00] DEBUG - logStreamInfo(): 	View Overflow Policy: 1
2021-03-16 17:48:41 [0x113249e00] DEBUG - logStreamInfo(): 	Segment UUID: NULL
2021-03-16 17:48:41 [0x113249e00] DEBUG - logStreamInfo(): 	Frame ordering mode: 0
2021-03-16 17:48:41 [0x113249e00] DEBUG - logStreamInfo(): Track list
2021-03-16 17:48:41 [0x113249e00] DEBUG - logStreamInfo(): 	Track id: 1
2021-03-16 17:48:41 [0x113249e00] DEBUG - logStreamInfo(): 	Track name: kinesis_video
2021-03-16 17:48:41 [0x113249e00] DEBUG - logStreamInfo(): 	Codec id: V_MPEG4/ISO/AVC
2021-03-16 17:48:41 [0x113249e00] DEBUG - logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
2021-03-16 17:48:41 [0x113249e00] DEBUG - logStreamInfo(): 	Track cpd: NULL
2021-03-16 17:48:41 [0x113249e00] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2021-03-16 17:48:41 [0x70000ad1f000] INFO - describeStreamResultEvent(): Describe stream result event.
2021-03-16 17:48:41 [0x70000ad1f000] WARN - describeStreamResult(): Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
2021-03-16 17:48:41 [0x70000ad1f000] WARN - describeStreamResult(): Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
2021-03-16 17:48:41 [0x70000ad1f000] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
2021-03-16 17:48:42 [0x70000ada2000] INFO - getStreamingEndpointResultEvent(): Get streaming endpoint result event.
2021-03-16 17:48:42 [0x70000ada2000] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
2021-03-16 17:48:42 [0x70000ada2000] DEBUG - getStreamingTokenHandler invoked
2021-03-16 17:48:42 [0x70000ada2000] DEBUG - Refreshing credentials. Force refreshing: 1 Now time is: 1615916922094371 Expiration: 18446744073709551615
2021-03-16 17:48:42 [0x70000ada2000] INFO - getStreamingTokenResultEvent(): Get streaming token result event.
2021-03-16 17:48:42 [0x70000ada2000] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
2021-03-16 17:48:42 [0x70000ada2000] DEBUG - streamReadyHandler invoked
2021-03-16 17:48:42 [0x70000ada2000] Stream is ready
```

As said above, I am unsure if I should raise this here or against g-streamer (although I have also tried some options for squashing output on that side too). If anyone has an existing solution for me other than redirecting output to `dev/null`, I would appreciate it :)

MushMal, 2021-03-16:
Ideally, you supply the log4cplus configuration file with the correct logging level. However, if you want to modify the default, you can do so here:  https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/gstreamer/gstkvssink.cpp#L640

M-Davies, 2021-03-16:
> Ideally, you supply the log4cplus configuration file with the correct logging level

What would I have to define in that file and where?

MushMal, 2021-03-16:
You could supply the full path for the log configuration as a GStreamer param

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/gstreamer/gstkvssink.cpp#L542

We have a sample config file: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/samples/kvs_log_configuration

For more information about the config file you should check out the documentation for log4cplus.

Please go ahead and resolve this issue as it's been addressed. Cut a separate issue if it's related to KVS itself

closed 2021-03-16 by M-Davies

M-Davies, 2021-03-17:
Thanks for the help and quick responses @MushMal 🙂

M-Davies, 2021-03-17:
For the curious who want a quick way of doing this, I am including all the changes I make to squish output after the Kinesis stream has begun on my [squash_debug branch](https://github.com/M-Davies/amazon-kinesis-video-streams-producer-sdk-cpp/tree/squash_debug)