awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #687: [QUESTION]How to stream to kvs with both video and audio from an RTSP source. Only getting video right now

pineapplejoe, 2021-03-20:
Related to #674. Asking here since that one is closed. 

I am trying to stream both audio and video from an IP camera, through a raspberry pi, and up into aws kinesis. 

I know the IP camera is producing both audio and video because I can use VLC to connect to the camera and both are present. 

I now have a pipeline that should produce both audio and video but im getting an error. 

The pipeline: 
```
gst-launch-1.0 rtspsrc location="rtsp://Pass:Word@192.168.1.15/stream" latency=0 name=d d. ! queue ! \
  rtppcmudepay ! mulawdec ! audioresample ! audioconvert ! queue ! matroskamux name=mux d. ! queue ! \
  rtph264depay ! h264parse ! queue ! mux. mux. ! \
  kvssink  <...aws creds ... > 
```

The error im getting is 

`WARNING: erroneous pipeline: could not link mux to kvssink0
`
If I remove the ! mux. mix. ! - the stream will work but there is no audio. just video. 

What am I missing?

Raj123456788, 2021-03-23:
@disa6302 @MushMal : Can someone please answer this? Which mux to use with kvssink along with an example if possible?

pineapplejoe, 2021-03-25:
After a lot of help from the get community this is what I have. its still not working but its progress.

```
gst-launch-1.0 rtspsrc name=src location="rtsp://Test:Test@192.168.1.22/live" short-header=TRUE do-timestamp=true ! rtph264depay ! h264parse ! kvssink name=sink <aws creds>  src.  ! rtppcmadepay ! alawdec ! audioresample ! audioconvert  ! voaacenc ! aacparse ! queue ! sink. 
```

It actually sorta works as I can see video though it seems to be basically frozen. I cant hear audio. 

In the debug output I see a ton of these 

2021-03-25 14:15:10 [1665090656] WARN - putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 47014082 presentation timestamp: 16167069102260600

pineapplejoe, 2021-03-26:
After a very long slog it is finally working. 

`gst-launch-1.0 rtspsrc name=src location="rtsp://Pass:Word@192.168.1.22/live" short-header=true do-timestamp=true ! rtph264depay ! h264parse !   kvssink name=sink stream-name="name"  access-key="key" secret-key="secret aws-region="region" frame-timecodes=false src.  !  rtppcmadepay ! alawdec ! audioresample ! audioconvert  ! voaacenc ! aacparse ! queue ! sink. 
`
You'll want to swap out (rtppcmadepay ! alawdec) with the specifics of your audio format (i.e rtppcmudepay ! mulawdec if your audio is mulaw).   

I havent stress tested it to see how long it can run just yet but so far its sustaining over a minute on stable network conditions. 

FYI @Raj123456788

cwardcode, 2021-03-26:
@pineapplejoe, et. al. Do you have an idea of how this would work with either AAC audio or mp3?

pineapplejoe, 2021-03-26:
So I am in no way an expert. But kvssink is expecting AAC so if your rtsp source is generating AAC audio directly then in theory you can just directly connect them.  

Im not totally sure how to get the AAC out of the rtsp source ... but for mp3 maybe this will work. 

rtpmpadepay ! mpg123audiodec  ! audioresample ! audioconvert ! voaacenc ! aacparse ! queue ! sink.

If there was a way to get AAC out of rtp packets then that would easiest because it would be something like. 

rtpaacdepay ! queue ! sink. 

But I havent found something like rtpaacdepay.

Again not an expert.  but hope it helps you along.

Raj123456788, 2021-03-26:
@pineapplejoe : Can you please tell us are you using HLS or Dash Player on AWS? The frames are repeating again and again couple of times and so is the audio.

pineapplejoe, 2021-03-26:
Cant say I've really seen that. Tested against a video with sound and clock and it never jumps around. What I have seen is that playback will sometimes stop though im not sure where that issue is since on the device the stream just continues.  In this case refreshing resumes to its possible that something on our playback side is faulty. 

What I have noticed is that we can remove the last ! queue ! and performance appears to improve and the likelihood of the stream stopping goes down. 

I dont know what your exact config is but one big thing that helped was adding `frame-timecodes=false ` which stopped a problem of very jittery video and a bunch of messages in the output that said something like 

`Failed to submit frame to Kinesis Video client`.

Maybe removing it in your case will help with frames appearing in order? And maybe just maybe you wont experience the "failed to submit frame..." problem?  Im skeptical but maybe try it and see what happens.

Raj123456788, 2021-03-29:
I tried removing frame-timecodes=false. I do receive Failed to submit frame to Kinesis Video client

hassanctech, 2021-04-08:
@pineapplejoe it looks like you've figured it out!  Can this issue be closed?

@Raj123456788 Perhaps you need to open a new issue and please include exactly what error code you're receiving with the message you stated.

closed 2021-04-09 by pineapplejoe

Raj123456788, 2021-04-09:
@pineapplejoe :Thanks you for the help. Sorry for the late reply.  Below is the gst command:
gst-launch-1.0 rtspsrc  debug=true name=src location=rtsp://IP/stream short-header=true do-timestamp=false ! identity sync=true ! rtph264depay ! h264parse ! kvssink name=sink stream-name="" storage-size=1024 aws-regio
n=us-east-1 retention-period=1  streaming-type=2 src. ! rtpopusdepay ! opusdec ! audioresample ! audioconvert ! voaacenc ! queue ! sink.

luisberga, 2021-11-22:
Hello! 

Below is my gts command: 

`gst-launch-1.0 rtspsrc name=src location="rtsp://192.168.14.5:5554/playlist.m3u" short-header=true do-timestamp=true ! rtph264depay ! h264parse ! kvssink name=sink stream-name="name" access-key="key" secret-key="secret" aws-region="sa-east-1" src. ! rtpmpadepay ! mpg123audiodec ! audioresample ! audioconvert ! voaacenc ! aacparse ! queue ! sink.`

however, i'm getting this error: 

`...Setting pipeline to PLAYING ...
New clock: GstSystemClock
Progress: (request) Sending PLAY request
Progress: (request) Sending PLAY request
Progress: (request) Sent PLAY request
WARNING: from element /GstPipeline:pipeline0/GstRTSPSrc:src/GstUDPSink:udpsink3: Error sending UDP packets
Additional debug info:
gstmultiudpsink.c(726): gst_multiudpsink_send_messages (): /GstPipeline:pipeline0/GstRTSPSrc:src/GstUDPSink:udpsink3:
client 192.168.15.3:0, reason: Error sending message: Invalid argument
INFO - kinesisVideoStreamFormatChanged(): Stream format changed.
2021-11-22 18:26:31 [140587632580352] WARNING: from element /GstPipeline:pipeline0/GstRTSPSrc:src: Delayed linking failed.
Additional debug info:
./grammar.y(506): gst_parse_no_more_pads (): /GstPipeline:pipeline0/GstRTSPSrc:src:
failed delayed linking some pad of GstRTSPSrc named src to some pad of GstRtpPcmaDepay named rtppcmadepay0
WARNING: from element /GstPipeline:pipeline0/GstRTSPSrc:src/GstUDPSink:udpsink1: Error sending UDP packets
Additional debug info:
gstmultiudpsink.c(726): gst_multiudpsink_send_messages (): /GstPipeline:pipeline0/GstRTSPSrc:src/GstUDPSink:udpsink1:
client 192.168.14.5:0, reason: Error sending message: Invalid argument
`

And I've got stuck in this part... Anyone could help me? 

Thanks!!

hassanctech, 2021-11-22:
@luisberga I see you've already cut a new issue for this, let's continue addressing it there.

aman-aguptaaa, 2023-06-08:
I am using kvssink plugin to write audio and video to same kvs.  And consume audio and video files from kvs as well.
But is there a way to combine the audio and video fragments ? I thought that kvs might maintain audio and video streams as tracks and we can easily read the combined media from kvs. But this is not the case.
Before sending audio and video fragments, muxing does not work (because of difference in pad capabilities of muxer and kvssink plugin).  Is there a way to do so before sending audio and video fragments ? The example mentioned above also does not use any mux before sending media.

so, my question is -
- Is there a way to combine audio and video before sending them via kvs plugin?
- Is there a way to combine audio and video fragments after receiving them ? (i can think of timestamp based combination of fragments, but not sure if this is recommended approach)
- What is the recommended way to proceed here ?