awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #689: [QUESTION] gst-launch-1.0 and kvssink inside kubernetes

jroy-reeldata, 2021-04-05:
I encounter issues when I try to deploy my docker container in a kubernetes cluster.

**Command**
`gst-launch-1.0 rtspsrc protocols="tcp" location=<rtsp-url> short-header=TRUE ! rtph264depay ! h264parse ! kvssink stream-name=<aws-stream=name> storage-size=512 access-key=<aws-access-key> secret-key=<aws-secret-key> aws-region=<aws-region>`

**Logs**
[logs.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/6258152/logs.txt)

**Any design considerations/constraints**
When I run the docker container directly on the cluster node, everything works as expected. When I deploy this same container in the cluster (on the same node), it doesn't work.

**Questions**
The question is; Are there any limitations or special configuration to use GStreamer with kvssink in a kubernetes cluster ?

hassanctech, 2021-04-05:
Which version of the SDK you're using?

From your logs I see the error you're getting is 0x32000002 which is [STATUS_MKV_INVALID_FRAME_TIMESTAMP](https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/5efcd01b75069716e1ace2a9fc00b729a2d1573f/src/mkvgen/include/com/amazonaws/kinesis/video/mkvgen/Include.h#L27). 

Can you please try with `filesink` and see if you get something playable?

hassanctech, 2021-04-05:
There is a big difference between the decoding and presentation timestamps in the log for example, this is your issue, it looks like the decoding timestamp is an offset and the presentation timestamp is an actual time:

decoding timestamp: 320000 presentation timestamp: 16176255232272042
decoding timestamp: 340000 presentation timestamp: 16176255232272042
decoding timestamp: 360000 presentation timestamp: 16176255232272042
decoding timestamp: 380000 presentation timestamp: 16176255232272042
decoding timestamp: 400000 presentation timestamp: 16176255232272042
decoding timestamp: 420000 presentation timestamp: 16176255232272042
decoding timestamp: 440000 presentation timestamp: 16176255232272042
decoding timestamp: 460000 presentation timestamp: 16176255232272042
decoding timestamp: 480000 presentation timestamp: 16176255232272042
decoding timestamp: 500000 presentation timestamp: 16176255232272042
decoding timestamp: 520000 presentation timestamp: 16176255232272042
decoding timestamp: 540000 presentation timestamp: 16176255232272042
decoding timestamp: 560000 presentation timestamp: 16176255232272042
decoding timestamp: 580000 presentation timestamp: 16176255232272042
decoding timestamp: 600000 presentation timestamp: 16176255232272042
decoding timestamp: 620000 presentation timestamp: 16176255232272042
decoding timestamp: 640000 presentation timestamp: 16176255232272042
decoding timestamp: 660000 presentation timestamp: 16176255232272042
decoding timestamp: 680000 presentation timestamp: 16176255232272042
decoding timestamp: 700000 presentation timestamp: 16176255232272042
decoding timestamp: 720000 presentation timestamp: 16176255232272042
decoding timestamp: 740000 presentation timestamp: 16176255232272042
decoding timestamp: 760000 presentation timestamp: 16176255232272042
decoding timestamp: 780000 presentation timestamp: 16176255232272042
decoding timestamp: 800000 presentation timestamp: 16176255232272042

hassanctech, 2021-04-05:
Even in the example that works I see the pts doesn't change for the last 5 frames. You say this is working but it is suspect, the encoder should not be producing multiple frames with the same time stamps.  In the example that didn't work it looks like all the frames that failed had the exact same pts.  Can you please investigate the cause of this?  Frames should have strictly monotonically increasing time stamps (for both dts and pts).

jroy-reeldata, 2021-04-06:
We changed our codec to use h.265 and that solved our problem. Looks like we have a problem with our h.264 encoder.

Since this is not a problem related to this SDK, I close this issue.

closed 2021-04-06 by jroy-reeldata