awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #690: [BUG]

kishorekumarkadari, 2021-04-08:
**Logging**
Used the access key id and secret access key in the command for gstreamer

**Describe the bug**
not able to create the required executables( may be kvs_gstreamer_sample) to stream.
Errors: "bash: ./kvs_gstreamer_sample: No such file or directory" or No such element or plugin, kvssink

**One liner description about the use case:** Edgility labs is trying to use the kinesis video streams to stream video using our video producer i.e raspberry pi camera interfaced to the raspberry pi 4B.  
One liner description about the use case: Edgility labs is trying to use the kinesis video streams to stream video using our video producer i.e raspberry pi camera interfaced to the raspberry pi 4B.  

We have followed the steps from the AWS documentation, in specific it is in Amazon Kinesis Video Streams’s developer guide( available till March 31st and week prior to that).
As suggested in step 1,we did setup the account and then followed step 2 and created the kinesis video stream. we have faced an issue in step 3.
Issue: In step 3 we have followed the steps as mentioned in the following link 

**To Reproduce**
1. We have followed the steps from the AWS documentation, in specific it is in Amazon Kinesis Video Streams’s developer guide( available till March 31st and week prior to that).
2. As suggested in step 1,we did setup the account and then followed step 2 and created the kinesis video stream. we have faced an issue in step 3.
3. Issue: In step 3 we have followed the steps as mentioned in the following link 
 https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/raspberry-pi.md#running-the-gstreamer-sample-application-to-upload-a-video-file

4. We tried to run few commands to stream mentioned below but we were not able to get executables kvs_gstreamer_sample or kvssink. “gst-launch-1.0 -v rtspsrc location=rtsp://YourCameraRtspUrl short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! h264parse ! kvssink stream-name=YourStreamName storage-size=128”“AWS_ACCESS_KEY_ID=YourAccessKeyId AWS_SECRET_ACCESS_KEY=YourSecretAccessKey ./kvs_gstreamer_sample <my-stream> </path/to/file>”
We keep getting errors below because of not able to create the required executables
( may be kvs_gstreamer_sample) to stream.
Errors: "bash: ./kvs_gstreamer_sample: No such file or directory"
ASK: We would like to seek assistance from kinesis video support center as soon as possible. If you have any follow up questions, please feel free to reach out to me.

**Expected behavior**
We need to stream the videos captured in a folder using the SDK

**SDK version number**
SDK version:f68a506. 
commit :disa6302 Pull in frame rate changes from PIC and correct documentation f68a506 13 days ago

**Open source building**
pi camera is used to capture the video.
**Platform (please complete the following information):**
 - OS:Raspbian 
 - Processor:ARMv8
 - Version  (raspberry pi 4B)

disa6302, 2021-04-08:
Please follow github instructions to build the SDK properly and generate the executables.

closed 2021-04-08 by disa6302

kishorekumarkadari, 2021-04-08:
Hi Divya, I tried github documentation but was never able to get the executable. could you please double check the documentation.

disa6302, 2021-04-08:
@kishorekumarkadari ,
The instructions are correct. Clone->create build directory->run cmake with -DBUILD_GSTREAMER_PLUGIN=ON from build directory -> run make. 

Ensure your build step passes

kishorekumarkadari, 2021-04-08:
@disa6302 I was able to get the executable after I exported path to plugin from root of repo rather than from the build, but when running gst-launch command i.e "AWS_ACCESS_KEY_ID=YourAccessKeyId AWS_SECRET_ACCESS_KEY=YourSecretAccessKey ./kvs_gstreamer_sample <my-stream> </path/to/file>", I have been getting the following error, DescribeStream API response: {"message":"The security token included in the request is invalid."}. so I created other user with admin privileges, I got an error below describeStreamCurlHandler(): DescribeStream API response: {"message":"The request signature we calculated does not match the signature you provided. Check your AWS Secret Access Key and signing method. Consult the service documentation for details."}. While authorizing it is checking in us west 2 region, but im from east-1 region,Is this because of the region ? could you please help me resolve this issue?

disa6302, 2021-04-08:
Please set the appropriate region. The documentation contains instruction to set the region. Ensure your AWS credentials are valid as well. 
`export AWS_DEFAULT_REGION=<region-name>`

kishorekumarkadari, 2021-04-08:
I changed the region and checked the secret access key gst launch is working "gst-launch-1.0 -v v4l2src do-timestamp=TRUE device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! omxh264enc periodicty-idr=45 inline-header=FALSE ! h264parse ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name=YourStreamName access-key="YourAccessKey" secret-key="YourSecretKey""

but I'm still getting this error "message ":"The request signature we calculated does not match the signature you provided. Check your AWS Secret Access Key and signing method. Consult the service documentation for details." for "AWS_ACCESS_KEY_ID=YourAccessKeyId AWS_SECRET_ACCESS_KEY=YourSecretAccessKey ./kvs_gstreamer_sample </path/to/file>" 
how to check on signing method?