awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #691: [BUG] video uploaded to kinesis is missing the first ~3 seconds

wlinInspire, 2021-04-11:
Hi KVS team,

I used to use `kinesis_video_gstreamer_audio_video_sample_app` and `kinesis_video_gstreamer_sample_app` pinned to commit 824de90 for uploading video clips without any problem. However, after updating them to the latest commit f68a506, I found the videos can still be uploaded but the first a few seconds (~3 seconds) will be deleted. The source code for `kinesis_video_gstreamer_audio_video_sample_app` and `kinesis_video_gstreamer_sample_app` does not have much change. So I was wondering if there could be major updates for the libraries such as PIC. Let me know if you can reproduce the issue

disa6302, 2021-04-12:
@wlinInspire , 

What do you mean by deleted? Are you losing the stream on the playback?

wlinInspire, 2021-04-12:
For  a video of 10 seconds. If I upload it to kinesis, I can only playback from second ~3 to 10. The first  3 seconds are missing

disa6302, 2021-04-22:
Hmm. That is strange. I am unable to reproduce this. Can you attach VERBOSE logs please?

Also any changes that went into your sample?

wlinInspire, 2021-04-22:
I found all the videos uploaded actually start with non-key frame. So maybe first a few frames are discarded by the uploader before it encounters the first key frame. Could this be the root cause?

wlinInspire, 2021-04-26:
@disa6302 Here is a sample mp4 file. Would you be able to reproduce the error using it?
https://user-images.githubusercontent.com/16128658/116028736-b0258780-a60c-11eb-9b31-60a5fa27c3bc.mp4

wlinInspire, 2021-05-01:
@disa6302 Any update on this?
@MushMal could you take a look at this issue as well?
Is there any specs on MP4 files that KVS can ingest?

MushMal, 2021-05-01:
The commit you are using is very old. If I had to guess, it is likely caused by the change that was introduced a while ago that would skip the frames until the first I-frame. Check your mpeg file and see whether the first frame is marked as an I-frame and if not, it will be skipped by default until the first I-frame (which becomes a key-frame).

You could change the default behavior and try to not skip the initial non-key-frames but it would be semantically wrong for H264 anyway so I would recommend fixing up your source to start with an I-frame

wlinInspire, 2021-05-01:
@MushMal I actually fixed my source to have the first frame to be the key frame always, but the issue still exists with latest commit in this repo. I found something quite interesting shown below, which could explain the missing first few seconds. 

I dump out the video track's `pkt_pts_time` using ffmpeg for the first 4 frames:
```
pkt_pts_time=6486.312000 (<=== key frame verified)
pkt_pts_time=6486.380000 (0.06800 seconds apart from previous frame)
pkt_pts_time=6486.445333 (0.06533 seconds apart from previous frame)
pkt_pts_time=6486.512667 (0.06733 seconds apart from previous frame)
```
Then, the TRACE log of latest kvs_gstreamer_audio_video_sample for the first 3 frames:
```
putKinesisVideoFrame(): debug frame info pts: 16198183296607557, dts: 0, duration: 0, size: 1229, trackId: 1, isKey 0
putKinesisVideoFrame(): debug frame info pts: 16198183297260890, dts: 0, duration: 0, size: 1782, trackId: 1, isKey 0 (653333 apart from previous)
putKinesisVideoFrame(): debug frame info pts: 16198183297934224, dts: 0, duration: 0, size: 1850, trackId: 1, isKey 0
(673334 apart from previous)
```

It is clearly that the first frame of my video (key frame) was not ingested by KVS. It is until ~ 2 seconds later, the first key frame got ingested by KVS. Happy to dig more into the issue. Thanks.

wlinInspire, 2021-05-01:
OK. The root cause is this line.
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/f68a50697436e8d1285ef297d5817b112841cb73/samples/kvs_gstreamer_audio_video_sample.cpp#L462

Do you know this it is still necessary to have that line?

MushMal, 2021-05-01:
This makes little sense to me. Perhaps a better logic at the kvs sink layer would be to just let any frames in - the PIC layer will ensure that the frames are skipped until the key-frame.

@disa6302 @hassanctech what do you think? Similar case might be in the KVS plugin code in demo repo.

wlinInspire, 2021-05-01:
also, first frames could be dropped if pipeline is not in playing state according to the following line, which I don't think is necessary. Removing it seems working fine.

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/f68a50697436e8d1285ef297d5817b112841cb73/samples/kvs_gstreamer_audio_video_sample.cpp#L396

MushMal, 2021-05-01:
This particular line needs a lot of scrutiny as there are multitude of scenarios and pipelines that need to be tested. I would recommend focusing on the first issue and not touching this.

hassanctech, 2021-05-03:
> OK. The root cause is this line.
> https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/f68a50697436e8d1285ef297d5817b112841cb73/samples/kvs_gstreamer_audio_video_sample.cpp#L462
> 
> Do you know this it is still necessary to have that line?

The reason this is there is if you have multiple tracks (audio + video) then it's possible that we started receiving audio frames before video.  If this is the case when the first fragment will get cut immediately before before resulting in an audio frames only fragment.  Our back-end will reject this because all tracks are required to have at last some frames in any given fragment.  Let me think about the best way forward here.

MushMal, 2021-05-03:
> The reason this is there is if you have multiple tracks (audio + video) then it's possible that we started receiving audio frames before video. If this is the case when the first fragment will get [...]

Curious about that case. In multi-track, the audio frames will not be stamped as key frames so they will be skipped anyway by the PIC. The processing would start from the first video I-frame. Is there a case where in between of two video I-frames there are no audio frames?

hassanctech, 2021-05-03:
I spoke with @chehefen and he indicated the answer is yes it could happen.  I will dig deeper into it maybe since the last time he looked at the the behavior has changed.

disa6302, 2021-10-29:
Change merged. Resolving.

PR #779 (2021-10-29): Fix bug that leads to losing initial frames
*Issue #, if available:*
#778 
#691 

*Description of changes:*
The bug leads to skipping the first frame even if it is a key frame since it is put inside the else clause. The `if` condition imply sets the flag to false to indicate the first video frame. 

Resolves #778 
Resolves #691 

By submitting this pull request, I confirm that you can use, modify, copy, and redistribute this contribution, under the terms of your choice.
closed 2021-10-29 by disa6302