awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #693: kvssink in gst_parse_launch error

waikup83, 2021-04-20:
Pipeline state changed from NULL to READY:
Pipeline state changed from READY to PAUSED:
Pipeline state changed from PAUSED to PLAYING:
Error received from element h264parse3: Internal data stream error.
Debugging information: gstbaseparse.c(3611): gst_base_parse_loop (): /GstPipeline:pipeline3/GstH264Parse:h264parse3:
streaming stopped, reason not-linked (-1)

i have a GStreamer kvssink command that work in command line but not in c++

// Command line working
gst-launch-1.0 -q filesrc location=frame.h264 ! h264parse ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name="XXXXXXX" access-key="XXXXXXXXX" secret-key="XXXXXXXX" aws-region="us-west-2"

// C++ not working
GstElement* pipeline = gst_parse_launch("filesrc location=frame.h264 ! h264parse ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name=\"XXXXXXX\" access-key=\"XXXXXXX\" secret-key=\"XXXXXXX\" aws-region=\"us-west-2\"", NULL);

gst_element_set_state(pipeline, GST_STATE_PLAYING);

**SDK version number**
latest

**Platform (please complete the following information):**
 - OS: Linux
 - Processor: intel

**Additional context**
im building my c++ with visual studios 2019 and it but and run remotely on a ubuntu 20 with gstreamer and kinesis video builded in it.

disa6302, 2021-04-20:
@waikup83  ,

please debug your pipeline. The issue is not related to KVS from the logs. Test your pipeline with fakesink and ensure it works. It looks like there is a data stream error. You might have to modify the sample application to adapt to your camera module

waikup83, 2021-04-20:
The same command work in command line and the output of h264parse with the filter is the sink of kvssink.
I continue to check anyway but because it work directly with gst-launch i doesn't understand why the same but parsed directly cant work.

waikup83, 2021-04-20:
by the way .. no error with fakesink

Pipeline state changed from NULL to READY:
Pipeline state changed from READY to PAUSED:
Pipeline state changed from PAUSED to PLAYING:
End-Of-Stream reached.

disa6302, 2021-04-22:
You might have to modify the sample to accommodate your camera specs.

disa6302, 2021-04-26:
Closing assuming question is answered. Feel free to re-open if you have any other questions.

closed 2021-04-26 by disa6302