awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #697: [BUG] IoTAuthCallback conflates stream and thing names, limiting to one stream per thing when using IoT cert auth provider

armyofevilrobots, 2021-04-29:
When creating a new stream via the gstreamer gst-launch method as documented, everything works perfectly when you follow the example code. I am using the Certificate based IoT credential provider, passing crt, key, rolealias, etc. Part of this includes setting the name of the stream to match the name of the IoT device. Unfortunately, my IoT device will be streaming three separate video streams (looking forward, backward, and internally). When setting the stream name to anything that doesn't match the IoT thing name, it fails.

Of course, I checked my RoleAlias, policy, Role, etc. My Role's inline policy is set to allow all Kinesis actions on:
```"Resource": "arn:aws:kinesisvideo:*:*:stream/*/*"``` , which DOES allow the $AWS_IOT_NAME stream, but fails with a 403 on all the others. So I dug a little deeper:

The Cert provider uses:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/blob/master/src/source/IotAuthCallback.c
which expects the streamName as a parameter, which is pretty straightforward. Unfortunately, that then ends up calling:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/blob/master/src/source/Common/IotCredentialProvider.c
and doing so with the streamName in the place of the thingName parameter. At this point, it's obvious what's happening; the credential provider s trying to get temporary credentials for a thing that does not exist, and fails with a 403.

I think the correct fix here would be to add a new parameter to the kvssink which expects the thingName, and which is then passed instead of the streamName (with fallback to streamName for backwards compatibility), this would decouple the thingName and streamName.

I realize that also a bug in the C provider (specifically the IoTAuthCallback.c code), but that's just an incorrectly named parameter, whereas the bug in this repo looks like a mismatch the use of the AWS auth tooling. I chose this repo because it makes more sense 

The logs are included below, which match what other people have posted in a few other places, generally without resolutions other than "the thing name and stream name need to be the same"

**Logging**
```
➜  amazon-kinesis-video-streams-producer-sdk-cpp git:(master) ✗ AWS_DEFAULT_REGION=us-east-2 gst-launch-1.0 videotestsrc  is-live=TRUE ! \
    clockoverlay halignment=left valignment=top text="FOO" time-format=%FT%H:%M:%S%Z  shaded-background=true font-desc="Sans, 16" \!  \
    textoverlay text="Noise" color=16711680 valignment=bottom halignment=right font-desc="Sans, 10" \! \
    videoconvert \!    video/x-raw,format=I420,width=640,height=480,framerate=30/1 \!    videoconvert \! \
    omxh264enc control-rate=1 target-bitrate=1200000 periodicity-idr=45 inline-header=FALSE  \!    \
    h264parse \!    video/x-h264,stream-format=avc,alignment=au,profile=baseline \!
    kvssink name=sink stream-name="${AWS_IOT_THING_NAME}-front" \
    iot-certificate="iot-certificate,endpoint=REDACTED.credentials.iot.us-east-2.amazonaws.com, \
    cert-path=REDACTED.crt,key-path=REDACTED.key,ca-path=REDACTED.pem, \
    role-aliases=KvsCameraIoTRoleAlias" retention-period=4320
Setting pipeline to PAUSED ...
DEBUG - Creating IoT auth callbacks.
2021-04-29 17:05:13 ERROR   blockingCurlCall(): Curl call response failed with http status 403
ERROR - Unable to create Iot Credential provider. Error status: 0x15000011
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstKvsSink:sink: Could not initialise supporting library.
Additional debug info:
/opt/beep-oes-py/amazon-kinesis-video-streams-producer-sdk-cpp/src/gstreamer/gstkvssink.cpp(1393): gst_kvs_sink_change_state (): /GstPipeline:pipeline0/GstKvsSink:sink:
Failed to init kvs producer. Error: Unable to create Iot Credential provider. Error status: 0x15000011
Setting pipeline to NULL ...
Freeing pipeline ...
```

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new video capable thing following the instructions here: https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/how-iot.html
2. Modify the inline role policy to be pointing at the resource `arn:aws:kinesisvideo:*:*:stream/*/*"`
3. Using gstreamer, as per the instructions, start a stream with the same name as the iot thing name (it should work). This eliminates the possibility that the inline role policy is blocking all streams.
4. Using gstreamer, start another stream, where the only difference is that the streamName is different. I would suggest `${AWS_IOT_THING_NAME}-no-such-thing`, or similar. This will fail because the credential provider attempts to connect as a thing that does not exist, even though all we changed was the streamName.

**Expected behavior**
The stream name should only be gated by the role and policy connected to the certificates via the roleAlias, and not expect to be identical to the IoT thing name.

**SDK version number**
master@e664fb1616c5d5c35966bafd3dcc6506b7b1d82a

**Platform (please complete the following information):**
 - OS: Linux/ARM Raspberry Pi
 - Processor: ARMv7 Processor rev 3 (v7l)

MushMal, 2021-04-29:
@armyofevilrobots  thanks for the detailed outline. The main issue is that the IoT auth is not really designed for a multi-stream scenario. The backend auth also presumes a single Thing -> Stream mapping currently. In order to handle multi-stream scenario, you might need to create separate IoT ROTs (certs + keys) for each of the streams and enhance the IoT auth to take the stream name into consideration when calling the credential provider with the appropriate cert/PK for that particular stream.

armyofevilrobots, 2021-04-29:
> The backend auth also presumes a single Thing -> Stream mapping currently. In order to handle multi-stream scenario, you might need to create separate IoT ROTs (certs + keys) for each of the streams [...]

OK, I was afraid of that. Just to be clear; you are saying that the backend IoT cert based auth system at AWS is currently requiring that the streamName and thingName are identical, even if I have an IAM policy that allows multiple streams for those credentials?

I was hoping that I could hack my way around the current auth to accept an additional thingName parameter on the gstreamer KVS plugin, but if the certificate auth endpoint is forcing that correlation, then I'll need some other method.
In that case, the workaround is pretty obvious, but it's a bit ugly to have to either send multiple sets of credentials (and have multiple camera "things" per actual thing), and/or to have to use STS to generate temporary creds that I manually rotate.

MushMal, 2021-04-29:
> OK, I was afraid of that. Just to be clear; you are saying that the backend IoT cert based auth system at AWS is currently requiring that the streamName and thingName are identical, even if I have [...]

Indeed, it requires a single thing name <-> stream name mapping. If your device has a secure way of accessing STS for the credentials then you can certainly use that to fetch the credentials. NOTE: You don't want to store your accounts long-term access key/secret key on the device.

armyofevilrobots, 2021-04-29:
>NOTE: You don't want to store your accounts long-term access key/secret key on the device.

Loud and clear, no chance I am keeping any credentials on the device outside of the greengrass certs.

disa6302, 2021-05-06:
Closing assuming question is answered.

closed 2021-05-06 by disa6302

clogwog, 2021-11-04:
@MushMal is there any work on making a single IoT device capable to handle more video streams ? It seems like an unusual decision to be assuming that an edge device will only have 1 camera.

MushMal, 2021-11-04:
> @MushMal is there any work on making a single IoT device capable to handle more video streams ? It seems like an unusual decision to be assuming that an edge device will only have 1 camera.

I agree that this is awkward. I would let AWS KVS folks to comment.