awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #699: how to remove debug kvssink message ?

waikup83, 2021-05-03:
how to remove debug kvssink message ?

i've allready add => putenv("GST_DEBUG=0,kvssink:0"); before initialisation of GStreamer

but a lot of kvssink debug and warning message are out anyway

waikup83, 2021-05-06:
[image]

disa6302, 2021-05-06:
If you want to get rid of KVS messages, you can set log level to SILENT. Check [this](https://github.com/awslabs/amazon-kinesis-video-streams-webrtc-sdk-c#setup-desired-log-level).

Not sure what you mean by kvssink messages and I do not understand how you have set the env, but kvssink in itself shouldnt have any logs. It is the producer SDK that would generate logs.

waikup83, 2021-05-06:
kvssink came from libgstkvssink.so called from the GStreamer =>
gst_parse_launch("appsrc name=frame ! queue max-size-buffers=1 leaky=0 ! h264parse ! avdec_h264 ! queue max-size-buffers=1 leaky=0  ! x264enc key-int-max=10 ! queue max-size-buffers=1 leaky=0 ! kvssink stream-name=camera access-key=xxxxx secret-key=xxxx aws-region=xxx fragment-acks=true absolute-fragment-times=false frame-timecodes=false max-latency=1 replay-duration=0 frame-timestamp=1", NULL);

I set the environment variable from a class constructor in c++ with the putenv command and even if i set => putenv("AWS_KVS_LOG_LEVEL=7"); i get plenty of kinesis and curl debug log

disa6302, 2021-05-06:
Please run `export AWS_KVS_LOG_LEVEL=7` to see if it works.

disa6302, 2021-05-06:
Closing assuming question is answered.

closed 2021-05-06 by disa6302

waikup83, 2021-05-06:
i've done it myself on the vm and it still havgin message

waikup83, 2021-05-06:
... maybe it because i'Ts cant open log file ..
[image]

waikup83, 2021-05-06:
nope it's still

MushMal, 2021-05-06:
@waikup83 the kvssink source is based on CPP repository which is substituting the default logger callback with the log4cplus.  By default it uses the configuration specified here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/gstreamer/gstkvssink.cpp#L106

You can specify your own path by applying the following gst param: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/gstreamer/gstkvssink.cpp#L542

There is a sample log4cplus configuration available at: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/samples/kvs_log_configuration

You can write your own to control the verbosity levels.

NOTE: This will be applicable to KVS logs ONLY. You will still need to control GST logs. You can use `export GST_DEBUG=0` to disable the logs

waikup83, 2021-05-06:
cool it work, thanks you very much !