awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #700: [QUESTION] frames not sending to kvs in last fragment

3xlogicDanielXu, 2021-05-03:
we are doing motion recording to kvs, and found an issue with the last few seconds frames.
set the keyFrameFragmentation to false and fragmentDuration to 5 seconds.
if the frames in last fragment doesn't enough for 5 seconds, it can't be pushed to kvs.
Seems there is a way to force pushing the last fragment to kvs even it's not 5 seconds long by calling putKinesisVideoFrame(streamHandle, &eofr) and set the FRAME_FLAG_END_OF_FRAGMENT flag in eofr.

but we noticed that in stream.c line 773-775, the producer sdk will need to check the end of fragment flag if keyFrameFragmentation is false, and return error status.   Anyone know why we need to check the flag when keyFrameFragmentation is false? or is there other way to fix the issue?

 if (!pKinesisVideoStream->streamInfo.streamCaps.keyFrameFragmentation) {
        CHK(!CHECK_FRAME_FLAG_END_OF_FRAGMENT(pFrame->flags), STATUS_END_OF_FRAGMENT_FRAME_INVALID_STATE);
    }

MushMal, 2021-05-03:
EoFR is reserved for a very few scenarios - mostly for non-video streams. While I am not sure about the actual lines you've commented (I am guessing they are guarding against some logic for auto-intermittent scenario) I am wondering why you chose the non-key-frame fragmentation.

3xlogicDanielXu, 2021-05-03:
when playback footage from kvs we prefer to search a timeline of 6 hours, as there is a limitation of 5000 fragments can be searched, we have to set the fragment duration to 5 seconds, hence need to set the keyFrameFragmentation to false.

MushMal, 2021-05-03:
Non-key-frame fragmentation would still only cut fragments on a key-frame boundary. This means that if the SDK cuts 5 second fragments, you do supply key-frames at least every 5 seconds. If, however, your pipeline produces more frequent I-frames, your application could chose to not mark the I-frame as a key-frame until your desired time elapses and still use key-frame-fragmentation. It really is up to your application to take the control of the fragmentation by specifying which frame is a key-frame. For example, if you have an I-frame every second, your application could mark a key-frame then not mark the consecutive 4 I-frames as key-frames.

3xlogicDanielXu, 2021-05-04:
@MushMal Thank you very much for your information.

3xlogicDanielXu, 2021-05-09:
@MushMal 
I tried above method to still use key-frame-fragmentation (true), and mark the key-frame flags every 5 secs, now I can see the fragment duration being cut off by each 5 secs, that works great.  

However, I noticed another issue, 
In order to push the last fragment to cloud in a motion recording process, we need to put a key frame to producer, the last key frame will be included in the next fragment, not in the last fragment pushed to cloud. The last key frame actually no need to be pushed to the cloud.

If we don't sent the key frame for cutting off the last fragment, seems producer still can push it to the cloud, but there will be a few secs delay, for example, when motion triggered, we observed each 5 secs fragment can be sent to the cloud continuously until motion stop, however, the last fragment is not sent out immediately following the second to last, there is a few secs lag.

Wondering if there is a way to let producer know after filling all the frames in the last fragment without marking and sending the extra key frame to avoid the lag? or the few secs delay is normal for the producer? 
Will a call of stopKinesisVideoStreamSync  and kinesisVideoStreamResetStream be the proper way for the purpose?

Thanks,

MushMal, 2021-05-10:
What you are describing is an intermittent producer scenario. You can have the following 3 solutions:

1) Use defaults and do absolutely nothing. By default, there will be an intermittent producer listener that will "close" the fragment after 20 second of inactivity. The only drawback of this is that it will take over 20 seconds for the fragment to be available to the consumer.
2) Use a special EoFR frame to indicate the end of the fragment.  You can use the initializer like `Frame eofr = EOFR_FRAME_INITIALIZER; `and then put this frame when you are done with the fragment `stream->putFrame(eofr));`
3) Explicitly use close stream sync and reset as you've mentioned. The drawback of this is that the extra call to DescribeStream API .

disa6302, 2021-05-19:
Closing due to inactivity.

closed 2021-05-19 by disa6302

deicool, 2021-05-19:
I would request you to keep this issue open and to wait for a while. Will close the issue myself. Thank You