awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #701: [BUG] Raspberry Pi 3b: Pipeline works with rtmpsink but fails with kvssink after a minute

JayDouglass, 2021-05-11:
**Logging**
```
DEBUG - postReadCallback(): Pausing CURL read for upload handle: 0
WARN - curlCompleteSync(): curl perform failed for url https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds
2021-05-11 15:45:03 [3827418208] WARN - curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com/putMedia
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=***/20210511/us-west-2/kinesisvideo/aws4_request, SignedHeaders=connection;host;transfer-encoding;user-agent;x-amz-date;x-amzn-fragment-acknowledgment-required;x-amzn-fragment-timecode-type;x-amzn-producer-start-timestamp;x-amzn-stream-name, Signature=e329b3195b5e5f1e53616a923cac26c92baae0381ea79ac
2021-05-11 15:45:03 [3827418208] DEBUG - putStreamCurlHandler(): Network thread for Kinesis Video stream: videotestsrc with upload handle: 0 exited. http status: 0
2021-05-11 15:45:03 [3827418208] WARN - putStreamCurlHandler(): Stream with streamHandle 31759992 uploadHandle 0 has exited without triggering end-of-stream. Service call result: 599
2021-05-11 15:45:03 [3827418208] INFO - kinesisVideoStreamTerminated(): Stream 0x1e49e78 terminated upload handle 0 with service call result 599.
2021-05-11 15:45:03 [3827418208] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000100, Next state: 0x0000000000000200
0:01:40.6 / 99:99:99.3827418208] 0:01:40.6 / 99:99:99.
```

**Describe the bug**
This pipeline works with rtmpsink, producing around 30% CPU utilization on the rpi 3:
```
gst-launch-1.0 rpicamsrc preview=false sensor-mode=5 bitrate=1700000 \
! video/x-raw,format=I420,width=1280,height=720 \
! tee name=t \
t. ! queue ! omxh264enc ! h264parse ! flvmux streamable=true ! rtmpsink location=\"rtmp://**** live=true\" \
t. ! queue ! videoscale ! video/x-raw,width=320,height=180 ! videoconvert ! glimagesink render-rectangle="<0,0,640,360>" 
```
If I switch the pipeline to kvssink:
```
gst-launch-1.0 rpicamsrc preview=false sensor-mode=5 bitrate=1700000 \
! video/x-raw,format=I420,width=1280,height=720 \
! tee name=t \
t. ! queue ! omxh264enc control-rate=1 periodicity-idr=45 inline-header=FALSE target-bitrate=1700000 \
    ! h264parse \
    ! kvssink buffer-duration="10" max-latency="10" storage-size="128" \
        stream-name="videotestsrc" \
        access-key="***" secret-key="***" aws-region="us-west-2" \
t. ! queue ! videoscale ! video/x-raw,width=320,height=180 ! videoconvert ! glimagesink render-rectangle="<0,0,640,360>" 
```
The pipeline fails consistently at the 1:30 mark, with a curl error as noted above. It also performs worse than rtmpsink, with lower FPS on the device, and the debug log indicates that it's choking from the beginning. Performance progressively gets worse and then it fails.

**To Reproduce**
1. Compile kvssink & Gstreamer with appropriate settings for the rpi 3b
2. Deploy to the device
2. Run the gst-launch-1.0 command quoted above

**Expected behavior**
The stream does not fail and performs as well as rtmpsink (~30% CPU usage).

**SDK version number**
We are pulling from github and building from source.

**Platform (please complete the following information):**
 - OS: balenalib/rpi-raspbian:bullseye Docker image running inside BalenaOS
 - Processor: Quad Core 1.2GHz Broadcom BCM2837 64bit CPU
 - Version: Raspberry Pi 3b

**Additional context**
kvssink works great with simpler pipelines like this one:
```
gst-launch-1.0 rpicamsrc bitrate=1700000 sensor-mode=5 \
    ! h264parse ! video/x-h264,stream-format=avc,alignment=au,width=1296,height=730,framerate=30/1,profile=baseline \
    ! kvssink buffer-duration="10" max-latency="10" \
      stream-name="videotestsrc" \
      access-key="*" secret-key="**" aws-region="us-west-2" 
```

MushMal, 2021-05-11:
A network timeout could be triggered not only when there is a network bandwidth issue (dropped packets, etc...) but also when the media pipeline is not producing frames. The default timeout is 30 seconds so the connection will get dropped after no data exchange. I would recommend debugging into your media pipeline and understanding why the frames are not being produced into the KVS sink. KVS sink has a "filter" property to drop frames outside of the range: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/gstreamer/gstkvssink.cpp#L253

Perhaps, disabling the clipping can help.

JayDouglass, 2021-05-11:
> A network timeout could be triggered not only when there is a network bandwidth issue (dropped packets, etc...) but also when the media pipeline is not producing frames. The default timeout is 30 [...]

Thanks. I agree that's likely due to something upstream of kvssink in the pipeline, but it's very hard to debug & reason about that when the pipeline works with rtmpsink, and I don't see the other pipeline elements erroring in the log. At the very least there is a problematic interaction in the pipeline that only happens when kvssink is present.

I'll try your suggestion about disabling clipping.

JayDouglass, 2021-05-11:
> Thanks. I agree that's likely due to something upstream of kvssink in the pipeline, but it's very hard to debug & reason about that when the pipeline works with rtmpsink, and I don't see the other [...]

Same result. I'm going to attribute this problem to the pi's resource constrained environment.  Using glimagesink and kvssink together using tee appears to seize the pipeline after a short period, even when using the queue element. I've also tried leaky=downstream on the glimagesink part of the pipeline and it has no effect.

disa6302, 2021-05-24:
Closing assuming KVS is not the issue here. Feel free to reopen if it is otherwise.

closed 2021-05-24 by disa6302