awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #702: No such element or plugin 'kvssink'

deicool, 2021-05-15:
Hello

I am trying to work this out: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp

However I am getting the following error on (Ubuntu 20):
No such element or plugin 'kvssink'

How do I solve this?

Thanks
Deepak

disa6302, 2021-05-15:
Please follow the Readme for instructions. You need to run cmake with -DBUILD_GSTREAMER_PLUGIN=ON and export GST_PLUGIN_PATH

deicool, 2021-05-15:
Getting the below error when I run (cmake with -DBUILD_GSTREAMER_PLUGIN=ON)

CMake Warning:
  No source or binary directory provided.  Both will be assumed to be the
  same as the current working directory, but note that this warning will
  become a fatal error in future CMake releases.

CMake Error at CMakeLists.txt:7 (include):
  include could not find load file:

    Utilities

-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Kinesis Video Cpp Producer path is /home/deicool/Ganu/TechXp-Forum/Amazon-Kinesis-Video
-- dependencies install path is /home/deicool/Ganu/TechXp-Forum/Amazon-Kinesis-Video/open-source
-- Begin building dependencies.
CMake Error at CMakeLists.txt:73 (build_dependency):
  Unknown CMake command "build_dependency".

-- Configuring incomplete, errors occurred!
See also "/home/deicool/Ganu/TechXp-Forum/Amazon-Kinesis-Video/CMakeFiles/CMakeOutput.log".

disa6302, 2021-05-15:
@deicool ,

Ensure your workspace is clean and then follow the instructions on the Readme. Also ensure you have the appropriate CMake version.

deicool, 2021-05-15:
I did that exactly

deicool, 2021-05-16:
I got a fresh installation of Ubuntu-Server and followed the instructions in the link:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp

The command below is giving the following error:
cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_JNI=TRUE

CMake Error at /snap/cmake/870/share/cmake-3.20/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
Call Stack (most recent call first):
  /snap/cmake/870/share/cmake-3.20/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /snap/cmake/870/share/cmake-3.20/Modules/FindPkgConfig.cmake:70 (find_package_handle_standard_args)
  dependency/libkvscproducer/kvscproducer-src/CMakeLists.txt:111 (find_package)

disa6302, 2021-05-16:
Please install the dependencies. Please go through the steps here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp#configure

It looks like you are missing pkgconfig.

deicool, 2021-05-17:
I have installed all dependencies and pkgconfig is also there.

P.S: Why this issue has been closed before a working-solution is given

disa6302, 2021-05-17:
Apologies. Missed reopening the issue. Can you try reinstalling the dependencies? I am unable to reproduce your issue on Ubuntu.

deicool, 2021-05-17:
Steps I have done:
=======================
1. git clone https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git

2.  mkdir -p amazon-kinesis-video-streams-producer-sdk-cpp/build
cd amazon-kinesis-video-streams-producer-sdk-cpp/build

3. sudo apt-get install libssl-dev libcurl4-openssl-dev liblog4cplus-dev libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev gstreamer1.0-plugins-base-apps gstreamer1.0-plugins-bad gstreamer1.0-plugins-good gstreamer1.0-plugins-ugly gstreamer1.0-tools

4. cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_JNI=TRUE

I am getting the following error:
=======================
[ 33%] Linking C executable lib556
collect2: fatal error: ld terminated with signal 11 [Segmentation fault], core dumped
compilation terminated.
make[5]: *** [tests/libtest/CMakeFiles/lib556.dir/build.make:118: tests/libtest/lib556] Error 1
make[5]: *** Deleting file 'tests/libtest/lib556'
make[4]: *** [CMakeFiles/Makefile2:2488: tests/libtest/CMakeFiles/lib556.dir/all] Error 2
make[3]: *** [Makefile:141: all] Error 2
make[2]: *** [CMakeFiles/project_libcurl.dir/build.make:113: build/src/project_libcurl-stamp/project_libcurl-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/project_libcurl.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
CMake Error at dependency/libkvscproducer/kvscproducer-src/CMake/Utilities.cmake:93 (message):
  CMake step for libcurl failed: 2
Call Stack (most recent call first):
  dependency/libkvscproducer/kvscproducer-src/CMakeLists.txt:86 (build_dependency)

-- Configuring incomplete, errors occurred!
See also "/home/deicool/Ganu/TechXp-Forum/Amazon-Kinesis-Video/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeOutput.log"

deicool, 2021-05-17:
The toolchains seems to be ok. Can you please tell me in more detail what do you mean by clean-install-of-the-toolchain?

MushMal, 2021-05-17:
In the logs I see your toolchains are crashing. You might want to debug this further to ensure the toolchain is OK. Try to re-install CMake. Try to build the core first by specifying cmake .. only. Nothing much else comes to mind.

deicool, 2021-05-17:
You need to be more descriptive about the-toolchain-to-be-ok.

I tried with reinstalling cmake as well as a fresh-installation of cmake (https://vitux.com/how-to-install-cmake-on-ubuntu/). 

When I run (cmake ..) only, I get the following errors:

CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find JNI (missing: JAVA_AWT_LIBRARY JAVA_INCLUDE_PATH
  JAVA_INCLUDE_PATH2 JAVA_AWT_INCLUDE_PATH)
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.16/Modules/FindJNI.cmake:372 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:180 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/deicool/Ganu/TechXp-Forum/Amazon-Kinesis-Video/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeOutput.log".
See also "/home/deicool/Ganu/TechXp-Forum/Amazon-Kinesis-Video/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeError.log".

MushMal, 2021-05-17:
The issues you are raising are not KVS specific.

In your case, after updating, are you able to run git clean -ddfx ? Seems that it's still attempting to build JNI.

deicool, 2021-05-17:
git clean -ddfx
warning: failed to remove ./: Invalid argument
Removing ./CMakeFiles
Removing ./dependency
Removing ./CMakeCache.txt

disa6302, 2021-05-24:
@deicool , 

This issue is not reproducible on Ubuntu 20. Following the instructions on README worked as is. You might have to look into why each dependency is failing for you. Looking into issues in the repositories corresponding to the dependencies would help.

olsnacky, 2021-07-22:
> Please follow the Readme for instructions. You need to run cmake with -DBUILD_GSTREAMER_PLUGIN=ON and export GST_PLUGIN_PATH

This instruction is misleading because the readme claims, in at lest two locations, that the correct flag value is:
> -DBUILD_GSTREAMER_PLUGIN=TRUE

Can we get confirmation on which one it is? Thank you :)

disa6302, 2021-07-28:
Closing since it has gone stale.

closed 2021-07-28 by disa6302

abhishekPurandare-obs, 2022-01-03:
I am still getting the same error after executing the following commands.
```
>> mkdir -p build
>> cd build
>> sudo apt-get install libssl-dev libcurl4-openssl-dev liblog4cplus-dev libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev
gstreamer1.0-plugins-base-apps gstreamer1.0-plugins-bad gstreamer1.0-plugins-good
gstreamer1.0-plugins-ugly gstreamer1.0-tools
>> cmake -DBUILD_GSTREAMER_PLUGIN=TRUE ..
>> make
In repo's root dir
>> export GST_PLUGIN_PATH=`pwd`/build
>> export LD_LIBRARY_PATH=`pwd`/open-source/local/lib
 
```

Build directory contents

```
CMakeCache.txt       dependency                          kvs_gstreamer_multistream_sample  libKinesisVideoProducer.so
CMakeFiles           kvs_gstreamer_audio_video_sample    kvs_gstreamer_sample              Makefile
cmake_install.cmake  kvs_gstreamer_file_uploader_sample  libgstkvssink.so
```

It seems like the shared objects were created. But I am still getting 

No such element or plugin 'kvssink' after executing `gst-inspect-1.0 kvssink`