awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #704: [QUESTION] How can stream a Kinesis video over a nginx or Apache ?

LoboTormenta, 2021-05-28:
Hi appreciated  Amazon Kinesis  developers and mantainers

We have a project for smart city improvement in security measures in México City, so we need stream a scalable flush of Amazon kinesis video form cameras into a web server like nginx or Apache for many users with a few administrators, like a web page with many video rectangles with the streams, we hope to reduce the impact of mobs around here.

How can we make that achievement ?

P.D. The info in the net is few and ambiguos

Thanks for your very importan attention

MushMal, 2021-05-28:
There are lots of details that need to be hashed out. If your solution is to use KVS then the ingestion is one part and the egress (consumption of video) is the separate part. KVS persistence path (using fragmented egress like the console viewer) is a near-realtime solution adding a few seconds to the latency whereas the GetMedia path is close to realtime with little latency. 
Ingestion can be achieved by using one of the producer SDKs on the device or as a gateway solution that ingests video from rtsp cameras and streams to KVS if the cameras themselves are not possible to install KVS producer SDK.
The viewing experience, as I said, is a separate application that can be modeled based on a few samples available. The streaming format, MKV, is a standard format that can be parsed by multiple technologies.

disa6302, 2021-06-17:
Assuming answered. Closing

closed 2021-06-17 by disa6302