awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #707: [QUESTION]/[Bugs?] Facing crash issue in SDK

crcsylvia, 2021-06-16:
We are facing crash issue in some random situations.
Environment:
SDK version: 3.0.0
Curl version: 7.57.0
Tracks: video(H264) and audio(AAC)
Platform: embedded system (VATICS vienna SoC)
Description:
We found crash when we did long term tests to our application.

**Case 1:** 
We found crash when the stale connection callback appeared for a long time (never come back for all tests). 
Stale happened when we test for more than one day(sometime more days), we suspected that is caused by our environment. However, we thought there is no reason for the crash after stale happened.
Question: 
1. Does the stream come back to normal after stale connection happened
2. How to collect more details for the sdk?
We collect core dump and found it crashed in curl library, but there are no more detail information for our debug build.
In debug mode, stale connection happened more frequently. We are afraid of debug mode might be another variation.
```
Jun 04 09:09:49.984| WARN|11612|ProducerCallback.cpp:28|streamReady : stream handle = 719888
Jun 04 09:09:49.996| INFO|11612|DefaultCallbackProvider.cpp:318|putStreamResultEvent(): Put stream result event. New upload handle 8
Jun 04 09:09:51.139| INFO|22418|DefaultCallbackProvider.cpp:318|getStreamData(): Handle 7 waiting for last persisted ack with ts 16227977895640000
Jun 04 09:09:51.379| INFO|22418|DefaultCallbackProvider.cpp:318|getStreamData(): Indicating an EOS after last persisted ACK is received for stream upload handle 7
Jun 04 09:09:51.392| INFO|22418|DefaultCallbackProvider.cpp:318|postReadCallback(): Reported end-of-stream for stream IT9360-H-release. Upload handle: 7
Jun 04 09:09:51.440| INFO|22476|DefaultCallbackProvider.cpp:318|writeHeaderCallback(): RequestId: e5f6e79b-e353-0ac6-bcc5-1784ffeb2bad
Jun 04 09:17:59.212| WARN|22476|DefaultCallbackProvider.cpp:318|curlCompleteSync(): curl perform failed for url https://s-96e4f3e9.kinesisvideo.ap-northeast-1.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds
Jun 04 09:17:59.219| WARN|22476|DefaultCallbackProvider.cpp:318|curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://s-96e4f3e9.kinesisvideo.ap-northeast-1.amazonaws.com/putMedia
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=ASIAW7MROKMMRXZIKYWW/20210604/ap-northeast-1/kinesisvideo/aws4_request, SignedHeaders=connection;host;transfer-encoding;user-agent;x-amz-date;x-amzn-fragment-acknowledgment-required;x-amzn-fragment-timecode-type;x-amzn-producer-start-timestamp;x-amzn-stream-name, Signature=5793cef9da2f6fdae9a1315ea1fa866b24fef
Jun 04 09:17:59.227| WARN|22476|DefaultCallbackProvider.cpp:318|putStreamCurlHandler(): Stream with streamHandle 719888 has exited without triggering end-of-stream. Service call result: 599
Jun 04 09:17:59.257| INFO|22476|DefaultCallbackProvider.cpp:318|kinesisVideoStreamTerminated(): Stream 0xafc10 terminated upload handle 8 with service call result 599.
Jun 04 09:17:59.280| WARN|22476|ProducerCallback.cpp:28|streamReady : stream handle = 719888
Jun 04 09:17:59.291| INFO|22476|DefaultCallbackProvider.cpp:318|putStreamResultEvent(): Put stream result event. New upload handle 9
Jun 04 09:18:00.919| INFO|22486|DefaultCallbackProvider.cpp:318|writeHeaderCallback(): RequestId: dab5cfb3-76fb-1849-8386-3dbb56b2fa5b
Jun 04 09:18:22.065| WARN|22486|ProducerCallback.cpp:99|Reporting stream stale. Last ACK received 600020000
Jun 04 09:18:22.159| WARN|22486|DefaultCallbackProvider.cpp:318|postReadCallback(): Reported abort-connection for Upload handle: 9
Jun 04 09:18:22.184| WARN|22486|DefaultCallbackProvider.cpp:318|curlCompleteSync(): HTTP Error 408 : Response: (null)
Request URL: https://s-96e4f3e9.kinesisvideo.ap-northeast-1.amazonaws.com/putMedia
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=ASIAW7MROKMMRXZIKYWW/20210604/ap-northeast-1/kinesisvideo/aws4_request, SignedHeaders=connection;host;transfer-encoding;user-agent;x-amz-date;x-amzn-fragment-acknowledgment-required;x-amzn-fragment-timecode-type;x-amzn-producer-start-timestamp;x-amzn-stream-name, Signature=1b2d1b40ab8e545255e025c389c2e968ea7
Jun 04 09:18:23.553| INFO|22489|DefaultCallbackProvider.cpp:318|writeHeaderCallback(): RequestId: 250b58eb-5b38-4ed0-99eb-80281774079a
Jun 04 09:18:23.563| INFO|22489|DefaultCallbackProvider.cpp:318|getStreamingEndpointResultEvent(): Get streaming endpoint result event.
Jun 04 09:18:26.016| INFO|22491|CredentialHandler.cpp:135|server response expire time: 2021-06-04T10:18:26Z
Jun 04 09:18:26.020| INFO|22491|CredentialHandler.cpp:138|current time: 1622798306, rotation time: 3300
Jun 04 09:18:26.029| INFO|22489|CredentialHandler.cpp:24|updateCredentials() old expire time: 1622801089
Jun 04 09:18:26.030| INFO|22489|CredentialHandler.cpp:25|updateCredentials() new expire time: 1622801606
Jun 04 09:18:26.035| INFO|22489|DefaultCallbackProvider.cpp:318|getStreamingTokenResultEvent(): Get streaming token result event.
Jun 04 09:18:26.039| WARN|22489|ProducerCallback.cpp:28|streamReady : stream handle = 719888
Jun 04 09:18:26.050| INFO|22489|DefaultCallbackProvider.cpp:318|putStreamResultEvent(): Put stream result event. New upload handle 10
Jun 04 09:18:27.195| INFO|22510|DefaultCallbackProvider.cpp:318|writeHeaderCallback(): RequestId: c9fe4edc-be3c-3381-90cd-bcff9e75d193
Jun 04 09:18:27.216| WARN|22510|ProducerCallback.cpp:99|Reporting stream stale. Last ACK received 600120000
Jun 04 09:18:27.278| WARN|22510|ProducerCallback.cpp:99|Reporting stream stale. Last ACK received 600120000
Jun 04 09:18:27.311| WARN|22510|ProducerCallback.cpp:99|Reporting stream stale. Last ACK received 600130000
Jun 04 09:18:27.371| WARN|22510|ProducerCallback.cpp:99|Reporting stream stale. Last ACK received 600130000
... many (Reporting stream stale.) logs from StreamConnectionStaleCallback
Jun 04 09:26:16.450| WARN|22510|ProducerCallback.cpp:99|Reporting stream stale. Last ACK received 600130000
Jun 04 09:26:16.465| WARN|22510|ProducerCallback.cpp:99|Reporting stream stale. Last ACK received 600130000
Jun 04 09:26:16.490| WARN|22510|ProducerCallback.cpp:99|Reporting stream stale. Last ACK received 600120000
Jun 04 09:26:16.535| WARN|22510|ProducerCallback.cpp:99|Reporting stream stale. Last ACK received 600130000
Jun 04 09:26:16.568| WARN|22510|ProducerCallback.cpp:99|Reporting stream stale. Last ACK received 600130000
Jun 04 09:26:16.893| INFO|11612|main.cpp:52|parent pid: 11612, stopped child: -1
Jun 04 09:26:16.937| INFO|11612|main.cpp:38|Start child process, pid: 22520
Jun 04 09:26:16.967| INFO|11612|Notifier.cpp:55|Init aenc notifier
Jun 04 09:26:16.970| INFO|11612|Notifier.cpp:55|Init venc notifier
Jun 04 09:26:17.037| INFO|11612|DefaultCallbackProvider.cpp:318|createKinesisVideoClient(): Creating Kinesis Video Client
Jun 04 09:26:17.038| INFO|11612|DefaultCallbackProvider.cpp:318|heapInitialize(): Initializing native heap with limit size 20971520, spill ratio 0% and flags 0x00000001
```
**Case 2:** 
We assumed stream might not back to normal when stale happened and we did some experiments to add a mechanism to observe if there are other crash issues.
We stop the process when the stale occurs more than 20 times in 5 seconds and called pProducer->freeStreams() when we release the producer. We met the crash again.
[... 87 lines cut ...]
[130718.469729] Hardware name: VATICS vienna SoC (Flattened Device Tree)
[130718.476287] task: c3222ec0 task.stack: c2d64000
[130718.480933] PC is at 0xb6c00bb4
[130718.484189] LR is at 0xb6c02c80
[130718.487496] pc : [<b6c00bb4>]    lr : [<b6c02c80>]    psr: 60000010
[130718.487496] sp : b4cc0100  ip : 000748f2  fp : 00000000
[130718.499241] r10: 00147958  r9 : 0014f5f8  r8 : 0014c1b0
[130718.504603] r7 : 00000000  r6 : 0014c268  r5 : 001478e8  r4 : 0014c18c
[130718.511236] r3 : 0014c268  r2 : 00000000  r1 : 00000000  r0 : 0014c1b0
[130718.517932] Flags: nZCv  IRQs on  FIQs on  Mode USER_32  ISA ARM  Segment user
[130718.525319] Control: 10c53c7d  Table: 02194059  DAC: 00000055
[130718.531213] CPU: 0 PID: 9479 Comm: uploadtokinesis Tainted: G           O    4.9.0-1 #2
[130718.539406] Hardware name: VATICS vienna SoC (Flattened Device Tree)
[130718.545976] [<c0015220>] (unwind_backtrace) from [<c0012f80>] (show_stack+0x10/0x14)
[130718.553861] [<c0012f80>] (show_stack) from [<c00199c8>] (__do_user_fault+0xc4/0xc8)
[130718.561721] [<c00199c8>] (__do_user_fault) from [<c0019b88>] (do_page_fault+0x1bc/0x328)
[130718.570008] [<c0019b88>] (do_page_fault) from [<c00092cc>] (do_DataAbort+0x38/0xb4)
[130718.577856] [<c00092cc>] (do_DataAbort) from [<c0013d1c>] (__dabt_usr+0x3c/0x40)
[130718.585391] Exception stack(0xc2d65fb0 to 0xc2d65ff8)
[130718.591019] 5fa0:                                     0014c1b0 00000000 00000000 0014c268
[130718.599442] 5fc0: 0014c18c 001478e8 0014c268 00000000 0014c1b0 0014f5f8 00147958 00000000
[130718.607810] 5fe0: 000748f2 b4cc0100 b6c02c80 b6c00bb4 60000010 ffffffff
```

Thanks for all of your kindly help.

MushMal, 2021-06-16:
@crcsylvia there is very little information that's useful. Which sample asset are you using? Could you get more debugging into? I see the second crash is a NULL deref while the first one is a little weird. You might want to capture the symbols or run a debug build to get the stack traces. The logs don't indicate anything useful and I am not sure whether the logs preceding the actual crash. For example, in the first log, the last entry is the heap initialization. It attempted to initialize a heap with size 20971520 bytes. Is it possible that you get OOM killer taking down the process?

Recommend:

* Update to the latest SDK version. KVS SDK v 3.0.0 is quite old and a lot of changes went in since then.
* Run debug build or enable symbol capture
* The samples presume a continuous streaming even in case of errors - they should recover.
* The samples and the SDK should not crash. They can produce errors but not crash.
* Run the stock sample with minimal modification on your platform
* Run the stock sample with NO modification on a commodity hardware and OS and ensure it works first, add your modifications to a stock platform and test before porting to your proprietary platform.

Hope to get information.

crcsylvia, 2021-06-18:
Thank for your reply. We will try to update sdk to latest version first and observe if the issue exists or not.

cijo7, 2021-06-24:
hi,

How to restart uploading after connection become stale. Can you point me to some documentation?

cijo7, 2021-06-24:
Can i handle stream staleness without loosing the video feed,

crcsylvia, 2021-09-06:
Update:
We have updated the SDK to v3.2.0 and test about 3 days on our platform.
Case 1: looks like the stream will be terminated after stale connection, and then continue streaming
```Sep 05 20:06:07.288| WARN|10101|ProducerCallback.cpp:97|Reporting stream stale. Last ACK received 600130000
Sep 05 20:06:07.317| WARN|10101|ProducerCallback.cpp:97|Reporting stream stale. Last ACK received 600130000
Sep 05 20:06:07.360| WARN|10101|ProducerCallback.cpp:97|Reporting stream stale. Last ACK received 600130000
Sep 05 20:06:42.044| WARN|10060|DefaultCallbackProvider.cpp:318|curlCompleteSync(): curl perform failed for url https://s-808fdb44.kinesisvideo.us-east-1.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds
Sep 05 20:06:42.047| WARN|10060|DefaultCallbackProvider.cpp:318|curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://s-808fdb44.kinesisvideo.us-east-1.amazonaws.com/putMedia
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=ASIA4KF5KTT6BOWZQS6F/20210905/us-east-1/kinesisvideo/aws4_request, SignedHeaders=connection;host;transfer-encoding;user-agent;x-amz-date;x-amzn-fragment-acknowledgment-required;x-amzn-fragment-timecode-type;x-amzn-producer-start-timestamp;x-amzn-stream-name, Signature=9ba747e99c0ba8932ac34965ba2b916a29a0e3d5a7494fe
Sep 05 20:06:42.061| WARN|10060|DefaultCallbackProvider.cpp:318|putStreamCurlHandler(): Stream with streamHandle 529504 uploadHandle 152 has exited without triggering end-of-stream. Service call result: 599
Sep 05 20:06:42.067| INFO|10060|DefaultCallbackProvider.cpp:318|kinesisVideoStreamTerminated(): Stream 0x81460 terminated upload handle 152 with service call result 599.
Sep 05 20:06:42.069| WARN|10060|DefaultCallbackProvider.cpp:318|streamTerminatedEvent(): Last fragment with timestamp 16308723652790000 for upload handle 152 might not be fully persisted
Sep 05 20:08:08.463| INFO|10106|DefaultCallbackProvider.cpp:318|writeHeaderCallback(): RequestId: 08b25c19-3de2-4e56-b25b-b1471a8ce770
Sep 05 20:08:08.488| INFO|10106|DefaultCallbackProvider.cpp:318|getStreamingEndpointResultEvent(): Get streaming endpoint result event.
Sep 05 20:08:11.053| INFO|10106|DefaultCallbackProvider.cpp:318|getStreamingTokenResultEvent(): Get streaming token result event.
Sep 05 20:08:11.056| WARN|10106|ProducerCallback.cpp:26|streamReady : stream handle = 529504
Sep 05 20:08:11.059| INFO|10106|DefaultCallbackProvider.cpp:318|putStreamResultEvent(): Put stream result event. New upload handle 154
Sep 05 20:08:11.098| INFO|10101|DefaultCallbackProvider.cpp:318|getStreamData(): Handle 153 waiting for last persisted ack with ts 16308724883050000
Sep 05 20:08:11.390| INFO|10101|DefaultCallbackProvider.cpp:318|getStreamData(): Indicating an EOS after last persisted ACK is received for stream upload handle 153
Sep 05 20:08:11.391| INFO|10101|DefaultCallbackProvider.cpp:318|postReadCallback(): Reported end-of-stream for stream ib9360-no-reconnect. Upload handle: 153
Sep 05 20:08:12.500| INFO|10126|DefaultCallbackProvider.cpp:318|writeHeaderCallback(): RequestId: c0213ba6-cd4f-6562-998e-1e4a5875730e
Sep 05 20:45:35.632| INFO|10165|DefaultCallbackProvider.cpp:318|writeHeaderCallback(): RequestId: 1283f6d3-0731-4c38-9d30-7466b545c364
```
Case 2: we didn't test to call free stream, due to case 1 can continue streaming after stale connection.
Case 3: we didn't see the crash issue so far.

Question for SDK v3.2.0: 
For Case1: 
We observe that the streaming will continue after 30 mins when stale connection occurred.
Is there any method to adjust this time range? like 10 mins?
New: 
We met HTTP 408 frequently in SDK v3.2.0 and continuously received HTTP 408 for a long time. (restart the program will recover sooner)
How long the server request timeout is?
Is there any method to adjust the timeout to be more tolerated?

Thank you!

closed 2021-09-09 by crcsylvia