awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #714: Installation on Windows 10 fails with LINK : fatal error LNK1181: cannot open input file 'apps\app_rand.obj'

hakonhagland, 2021-07-02:
Platform: Windows 10, x64 Native Tools Command Prompt VS 2019:

```
$ git clone git@github.com:awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git
$ mv amazon-kinesis-video-streams-producer-sdk-cpp akvspsc
$ cd akvspsc
$ mkdir build
$ cd build
$ cmake .. -G "NMake Makefiles" -DBUILD_GSTREAMER_PLUGIN=ON
C:\test\akvspsc\build> cmake .. -G "NMake Makefiles" -DBUILD_GSTREAMER_PLUGIN=ON
-- The C compiler identification is MSVC 19.26.28806.0
-- The CXX compiler identification is MSVC 19.26.28806.0
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.26.28801/bin/Hostx64/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.26.28801/bin/Hostx64/x64/cl.exe - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.26.28801/bin/Hostx64/x64/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.26.28801/bin/Hostx64/x64/cl.exe - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Kinesis Video Cpp Producer path is C:/test/akvspsc
-- dependencies install path is C:/test/akvspsc/open-source
-- Begin building dependencies.
-- Configuring done
-- Generating done
-- Build files have been written to: C:/test/akvspsc/open-source/liblog4cplus
Scanning dependencies of target project_log4cplus
[ 11%] Creating directories for 'project_log4cplus'
[ 22%] Performing download step (git clone) for 'project_log4cplus'
Cloning into 'project_log4cplus'...
Note: switching to 'REL_1_2_2'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 6ae6110a Update Automake to 1.16.1.
[ 33%] Performing update step for 'project_log4cplus'
[ 44%] No patch step for 'project_log4cplus'
[ 55%] Performing configure step for 'project_log4cplus'
-- The C compiler identification is MSVC 19.26.28806.0
-- The CXX compiler identification is MSVC 19.26.28806.0
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.26.28801/bin/Hostx64/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.26.28801/bin/Hostx64/x64/cl.exe - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.26.28801/bin/Hostx64/x64/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.26.28801/bin/Hostx64/x64/cl.exe - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Generating build for Log4cplus version 1.2.2
[... 516 lines cut ...]
        C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.26.28801/bin/Hostx64/x64/cl.exe  /Zi /Fdossl_static.pdb /Gs0 /GF /Gy /MD /W3 /wd4090 /nologo /O2 /I "." /I "include" -D"L_ENDIAN" -D"OPENSSL_PIC" -D"OPENSSLDIR=\"C:\\test\\akvspsc\\open-source\\local\"" -D"ENGINESDIR=\"C:\\test\\akvspsc\\open-source\\local\\lib\\engines-1_1\"" -D"OPENSSL_SYS_WIN32" -D"WIN32_LEAN_AND_MEAN" -D"UNICODE" -D"_UNICODE" -D"_CRT_SECURE_NO_DEPRECATE" -D"_WINSOCK_DEPRECATED_NO_WARNINGS" -D"OPENSSL_USE_APPLINK" -D"NDEBUG"  -c /Foapps\s_socket.obj "apps\s_socket.c"
        C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.26.28801/bin/Hostx64/x64/cl.exe  /Zi /Fdossl_static.pdb /Gs0 /GF /Gy /MD /W3 /wd4090 /nologo /O2 /I "." /I "include" -D"L_ENDIAN" -D"OPENSSL_PIC" -D"OPENSSLDIR=\"C:\\test\\akvspsc\\open-source\\local\"" -D"ENGINESDIR=\"C:\\test\\akvspsc\\open-source\\local\\lib\\engines-1_1\"" -D"OPENSSL_SYS_WIN32" -D"WIN32_LEAN_AND_MEAN" -D"UNICODE" -D"_UNICODE" -D"_CRT_SECURE_NO_DEPRECATE" -D"_WINSOCK_DEPRECATED_NO_WARNINGS" -D"OPENSSL_USE_APPLINK" -D"NDEBUG"  /Zs /showIncludes "apps\s_socket.c" 2>&1 > apps\s_socket.d
        C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.26.28801/bin/Hostx64/x64/cl.exe  /Zi /Fdossl_static.pdb /Gs0 /GF /Gy /MD /W3 /wd4090 /nologo /O2 /I "." /I "include" -D"L_ENDIAN" -D"OPENSSL_PIC" -D"OPENSSLDIR=\"C:\\test\\akvspsc\\open-source\\local\"" -D"ENGINESDIR=\"C:\\test\\akvspsc\\open-source\\local\\lib\\engines-1_1\"" -D"OPENSSL_SYS_WIN32" -D"WIN32_LEAN_AND_MEAN" -D"UNICODE" -D"_UNICODE" -D"_CRT_SECURE_NO_DEPRECATE" -D"_WINSOCK_DEPRECATED_NO_WARNINGS" -D"OPENSSL_USE_APPLINK" -D"NDEBUG"  -c /Foapps\win32_init.obj "apps\win32_init.c"
        C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.26.28801/bin/Hostx64/x64/cl.exe  /Zi /Fdossl_static.pdb /Gs0 /GF /Gy /MD /W3 /wd4090 /nologo /O2 /I "." /I "include" -D"L_ENDIAN" -D"OPENSSL_PIC" -D"OPENSNMAKE : SLDIR=\"C:\\test\\akvspsc\\open-source\\local\"" -D"ENGINESDIR=\"C:\\test\\akvspsc\\open-source\\local\\lib\\engines-1_1\"" -D"OPENSSL_SYS_WIN32" -D"WIN32_LEAN_AND_MEAN" -D"UNICODE" -D"_UNICODE" -D"_CRT_SECURE_NO_DEPRECATE" -D"_WINSOCK_DEPRECATED_NO_WARNINGS" -D"OPENSSL_USE_APPLINK" -D"NDEBUG"  /Zs /showIncludes "apps\win32_init.c" 2>&1 > apps\win32_init.d
        lib /nologo /out:apps\libapps.lib @C:\Users\hakon\AppData\Local\Temp\nm4683.tmp
LINK : fatal error LNK1181: cannot open input file 'apps\app_rand.obj'
fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\bin\HostX64\x64\lib.exe"' : return code '0x49d'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\bin\Hostx64\x64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: 'echo' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
CMake Error at dependency/libkvscproducer/kvscproducer-src/CMake/Utilities.cmake:93 (message):
  CMake step for libopenssl failed: 2
Call Stack (most recent call first):
  dependency/libkvscproducer/kvscproducer-src/CMakeLists.txt:66 (build_dependency)

-- Configuring incomplete, errors occurred!
See also "C:/test/akvspsc/build/CMakeFiles/CMakeOutput.log".

```

shirogaku, 2021-08-18:
Try build twice.
First time I got openssl build error like you but passed on second time.
My Windows 10 SDK version is 10.0.19041.0

disa6302, 2021-09-08:
Openssl is a little flaky on Windows and rebuilding usually works. I am closing this ticket now. Feel free to reach out if you have any other questions.

closed 2021-09-08 by disa6302