awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #715: Installation on Windows 10 fails with : Checking for module 'gstreamer-app-1.0' ... A required package was not found

hakonhagland, 2021-07-02:
Platform: Windows 10, x64 Native Tools Command Prompt VS 2019. If I rerun the `cmake` command in issue #714, the error `cannot open input file 'apps\app_rand.obj'` disappears and the build runs further before another error occurs:

```
$ git clone git@github.com:awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git
$ mv amazon-kinesis-video-streams-producer-sdk-cpp akvspsc
$ cd akvspsc
$ mkdir build
$ cd build
$ cmake .. -G "NMake Makefiles" -DBUILD_GSTREAMER_PLUGIN=ON
[...]
LINK : fatal error LNK1181: cannot open input file 'apps\app_rand.obj'
[...]
$ cmake .. -G "NMake Makefiles" -DBUILD_GSTREAMER_PLUGIN=ON
[...]
-- Installing: C:/test/akvspsc/open-source/local/lib/cmake/CURL/CURLConfig.cmake
-- Installing: C:/test/akvspsc/open-source/local/lib/libcurl-d_imp.lib
-- Installing: C:/test/akvspsc/open-source/local/bin/libcurl-d.dll
[ 88%] No test step for 'project_libcurl'
[100%] Completed 'project_libcurl'
[100%] Built target project_libcurl
-- Finished building dependencies.
-- Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.19043.
-- Configuring done
-- Generating done
-- Build files have been written to: C:/test/akvspsc/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic
Microsoft (R) Build Engine version 16.6.0+5ff7b0c9e for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  Checking Build System
  Performing update step for 'libkvspic-download'
  Building Custom Rule C:/test/akvspsc/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/CMakeLists.txt
  No configure step for 'libkvspic-download'
  No build step for 'libkvspic-download'
  No install step for 'libkvspic-download'
  No test step for 'libkvspic-download'
  Completed 'libkvspic-download'
  Building Custom Rule C:/test/akvspsc/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/CMakeLists.txt
-- Bitness 64 bits
-- Checking for module 'gstreamer-app-1.0'
--   
CMake Error at C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.17/Modules/FindPkgConfig.cmake:467 (message):
  A required package was not found
Call Stack (most recent call first):
  C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.17/Modules/FindPkgConfig.cmake:647 (_pkg_check_modules_internal)
  CMakeLists.txt:189 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "C:/test/akvspsc/build/CMakeFiles/CMakeOutput.log".
See also "C:/test/akvspsc/build/CMakeFiles/CMakeError.log".
```
The complete build log for the second `cmake` run is [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/6756724/log.txt).

Note that I use Strawberry Perl 5.32.1, whereas in issue #713, ActivePerl was used. That might explain why I did not get the `PERL_EXECUTABLE-NOTFOUND` error as was reported in #713.

prelich, 2021-07-23:
I have the exact same issue on the same platform right at checking for module 'gstreamer-app-1.0'.  I installed gstreamer-1.0-msvc-x86_64-1.18.4.exe and gstreamer-1.0-devel-msvc-x86_64-1.18.4.exe and put the C:\gstreamer\1.0\msvc_x86_64\ bin(include)(lib) paths in my system environment.

Maybe there is another path that needs to be added?

prelich, 2021-07-26:
https://github.com/alexa/avs-device-sdk/issues/462  

You can check to see if pkg-config works:
With pkg-config --cflags --libs gstreamer-1.0 to check your default gstreamer path.

finally figured it out.  You need to set the PKG_CONFIG_PATH environmental variables:

[image]

I'm compiling now.  Just figured I'd post this while I wait to see what happens.

prelich, 2021-07-27:
I've wasted too much time on this :(  I feel like I was close, but not close enough.

The compiler is calling these POSIX threads.  CMAKE was a pain, so I copied the headers and .dll files into my compiler path, but then I got thrown the pthread_atfork definition which is getting called somewhere:

`Performing C SOURCE FILE Test CMAKE_HAVE_LIBC_PTHREAD failed with the following output:
Change Dir: C:/Users/prelich/code/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeTmp

Run Build Command(s):C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/MSBuild/Current/Bin/MSBuild.exe cmTC_73556.vcxproj /p:Configuration=Debug /p:Platform=x64 /p:VisualStudioVersion=16.0 /v:m && Microsoft (R) Build Engine version 16.7.0+b89cb5fde for .NET Framework

Copyright (C) Microsoft Corporation. All rights reserved.

  Microsoft (R) C/C++ Optimizing Compiler Version 19.27.29111 for x64

  src.c

  Copyright (C) Microsoft Corporation.  All rights reserved.

  cl /c /I"C:\Users\prelich\code\vcpkg\scripts\buildsystems\msbuild\..\..\..\installed\x64-windows\include" /Zi /W3 /WX- /diagnostics:column /Od /Ob0 /D _MBCS /D WIN32 /D _WINDOWS /D CMAKE_HAVE_LIBC_PTHREAD /D "CMAKE_INTDIR=\"Debug\"" /Gm- /RTC1 /MDd /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /Fo"cmTC_73556.dir\Debug\\" /Fd"cmTC_73556.dir\Debug\vc142.pdb" /Gd /TC /errorReport:queue "C:\Users\prelich\code\amazon-kinesis-video-streams-producer-sdk-cpp\build\CMakeFiles\CMakeTmp\src.c"

C:\Users\prelich\code\amazon-kinesis-video-streams-producer-sdk-cpp\build\CMakeFiles\CMakeTmp\src.c(15,17): warning C4013: 'pthread_atfork' undefined; assuming extern returning int [C:\Users\prelich\code\amazon-kinesis-video-streams-producer-sdk-cpp\build\CMakeFiles\CMakeTmp\cmTC_73556.vcxproj]

src.obj : error LNK2019: unresolved external symbol pthread_atfork referenced in function main [C:\Users\prelich\code\amazon-kinesis-video-streams-producer-sdk-cpp\build\CMakeFiles\CMakeTmp\cmTC_73556.vcxproj]

C:\Users\prelich\code\amazon-kinesis-video-streams-producer-sdk-cpp\build\CMakeFiles\CMakeTmp\Debug\cmTC_73556.exe : fatal error LNK1120: 1 unresolved externals [C:\Users\prelich\code\amazon-kinesis-video-streams-producer-sdk-cpp\build\CMakeFiles\CMakeTmp\cmTC_73556.vcxproj]

Source file was:
#include <pthread.h>

static void* test_func(void* data)
{
  return data;
}

int main(void)
{
  pthread_t thread;
  pthread_create(&thread, NULL, test_func, NULL);
  pthread_detach(thread);
  pthread_cancel(thread);
  pthread_join(thread, NULL);
  pthread_atfork(NULL, NULL, NULL);
  pthread_exit(NULL);

  return 0;
}
`

rsurati, 2021-08-05:
EDIT: OK, I installed 32 bit libraries instead of the 64 bit libraries. vcpkg lets you install both. Make sure to install glib, gettext, along with pthreads and do it in 64 bit. My environment is slow to build, I'll update if I'm successful.

How do u set the path of pthread in windows? As CMake error logs shows that pthread functions are not available.....

prelich, 2021-08-05:
Windows doesn't have pthread natively.

https://sourceforge.net/projects/pthreads4w/

I set the path for pthread.h and then there were a bunch of pthread
derivatives I had to link as well, but didn't.  I gave up on that, but I
think the main issue is with GStreamer.  For whatever reason, the gstreamer
app is not getting installed correctly and it's screwing up the aws build
downstream.  I was able to build everything except for the GStreamer plugin.

I finally gave up and set up a dev environment in Ubuntu and everything
compiled in 20 minutes.  This is MS's motto:   "if it ain't broke, lets
patch it.".

Lepm11, 2021-09-21:
I'm building on windows 10 and I had the same error.

in my CMakeList had the following route for the gstreamer:

C:\\gstreamer\\1.0\\x86_64\\lib

but my gstramer is installed on C:\gstreamer\1.0\**msvc_x86_64**\lib 

so I changed the route from the CMakeList and its worked !!

Madesh1997, 2022-01-18:
@Lepm11 which CMakeList did you change the route?

niyatim23, 2022-03-07:
Hi, apologies for the delayed response. Can you please refer to the batch file [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/develop/.github/build_windows.bat) and the windows build [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/develop/.github/workflows/ci.yml#L268-L298) and let us know if it works for you?

disa6302, 2022-03-21:
Closing due to inactivity. Please follow the CI build files provided for windows

closed 2022-03-21 by disa6302