awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #716: Error uploading - stabbility issues

cijo7, 2021-07-04:
I was looking at code base.

How is STATUS_ACK_ERR_FRAGMENT_DURATION_REACHED a recoverable error?

https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/master/src/client/include/com/amazonaws/kinesis/video/client/Include.h#L198

If this is due to missing i frame how this error will recover? Isn't this a bug?

MushMal, 2021-11-11:
This is generated from KVS backend to indicate that the fragment is longer than 10 sec (I believe this is the default - should check the limits docs). This is normally caused by the encoder not producing regular interval Idr frames so the fragments go for longer.

niyatim23, 2022-03-10:
Closing assuming answered

closed 2022-03-10 by niyatim23