awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #717: [BUG] : Pipeline stuck at "Setting pipeline to PLAYING ... New clock: GstSystemClock"

Winneee, 2021-07-05:
Hi,
I have built kvsproducer using amazon-kinesis-video-streams-producer-sdk-cpp and using kvssink for streaming.
Below is the pipeline Used for streaming

**gst-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video0 ! video/x-raw, width=640, height=480 ! videoconvert! vpuenc_h264 ! h264parse ! video/x-h264, stream-format=avc, alignment=au, width=640, height=480 ! kvssink stream-name=415ae8e5-f115-4125-8ee7-082e13f2578c access-key=AKIAIB3DDJAN2HFQZHGQ secret-key=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 aws-region=ap-south-1 restart_on_error=false connection_staleness=40**

Pipeline is set without any error but it got stuck after setting pipeline to playing.

INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2021-07-05 12:35:33 [281472894467232] INFO - heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
2021-07-05 12:35:33 [281472894467232] INFO - heapInitialize(): Creating AIV heap.
2021-07-05 12:35:33 [281472894467232] INFO - heapInitialize(): Heap is initialized OK
2021-07-05 12:35:33 [281472894467232] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2021-07-05 12:35:33 [281472894467232] DEBUG - getSecurityTokenHandler invoked
2021-07-05 12:35:33 [281472894467232] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1625488533853010809 Expiration: 0
2021-07-05 12:35:33 [281472894467232] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
2021-07-05 12:35:33 [281472894467232] INFO - createDeviceResultEvent(): Create device result event.
2021-07-05 12:35:33 [281472894467232] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
2021-07-05 12:35:33 [281472894467232] DEBUG - clientReadyHandler invoked
2021-07-05 12:35:33 [281472894467232] INFO - try creating stream
2021-07-05 12:35:33 [281472894467232] INFO - Creating Kinesis Video Stream 415ae8e5-f115-4125-8ee7-082e13f2578c
2021-07-05 12:35:33 [281472894467232] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2021-07-05 12:35:33 [281472894467232] DEBUG - logStreamInfo(): Kinesis Video Stream Info
2021-07-05 12:35:33 [281472894467232] DEBUG - logStreamInfo(): Stream name: 415ae8e5-f115-425-8ee7-082e13f2578c
2021-07-05 12:35:33 [281472894467232] DEBUG - logStreamInfo(): Streaming type: STREAMING_TYPE_REALTIME
2021-07-05 12:35:33 [281472894467232] DEBUG - logStreamInfo(): Content type: video/h264
2021-07-05 12:35:33 [281472894467232] DEBUG - logStreamInfo(): Max latency (100ns): 600000000
2021-07-05 12:35:33 [281472894467232] DEBUG - logStreamInfo(): Fragment duration (100ns): 20000000
2021-07-05 12:35:33 [281472894467232] DEBUG - logStreamInfo(): Key frame fragmentation: Yes
2021-07-05 12:35:33 [281472894467232] DEBUG - logStreamInfo(): Use frame timecode: Yes
2021-07-05 12:35:33 [281472894467232] DEBUG - logStreamInfo(): Absolute frame timecode: Yes
2021-07-05 12:35:33 [281472894467232] DEBUG - logStreamInfo(): Nal adaptation flags: 0
2021-07-05 12:35:33 [281472894467232] DEBUG - logStreamInfo(): Average bandwith (bps): 4194304
2021-07-05 12:35:33 [281472894467232] DEBUG - logStreamInfo(): Framerate: 25
2021-07-05 12:35:33 [281472894467232] DEBUG - logStreamInfo(): Buffer duration (100ns): 1200000000
2021-07-05 12:35:33 [281472894467232] DEBUG - logStreamInfo(): Replay duration (100ns): 400000000
2021-07-05 12:35:33 [281472894467232] DEBUG - logStreamInfo(): Connection Staleness duration (100ns): 400000000
2021-07-05 12:35:33 [281472894467232] DEBUG - logStreamInfo(): Store Pressure Policy: 1
2021-07-05 12:35:33 [281472894467232] DEBUG - logStreamInfo(): View Overflow Policy: 1
2021-07-05 12:35:33 [281472894467232] DEBUG - logStreamInfo(): Segment UUID: NULL
2021-07-05 12:35:33 [281472894467232] DEBUG - logStreamInfo(): Frame ordering mode: 0
2021-07-05 12:35:33 [281472894467232] DEBUG - logStreamInfo(): Track list
2021-07-05 12:35:33 [281472894467232] DEBUG - logStreamInfo(): Track id: 1
2021-07-05 12:35:33 [281472894467232] DEBUG - logStreamInfo(): Track name: kinesis_video
2021-07-05 12:35:33 [281472894467232] DEBUG - logStreamInfo(): Codec id: V_MPEG4/ISO/AVC
2021-07-05 12:35:33 [281472894467232] DEBUG - logStreamInfo(): Track type: TRACK_INFO_TYPE_VIDEO
2021-07-05 12:35:33 [281472894467232] DEBUG - logStreamInfo(): Track cpd: NULL
2021-07-05 12:35:33 [281472894467232] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2021-07-05 12:35:33 [281472894467232] INFO - writeHeaderCallback(): RequestId: a0a19700-985a-4beb-959b-d425c4e408f4
2021-07-05 12:35:34 [281472719503840] DEBUG - describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.624626808443E9,"DataRetentionInHours":1,"DeviceName":null,"IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:ap-south-1:308222340895:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:ap-south-1:308222340895:stream/415ae8e5-f115-4125-8ee7-082e13f2578c/1624626808443","StreamName":"415ae8e5-f115-4125-8ee7-082e13f2578c","Version":"AJEhmCaus22KPfZsxI49"}}
2021-07-05 12:35:34 [281472719503840] INFO - describeStreamResultEvent(): Describe stream result event.
2021-07-05 12:35:34 [281472719503840] WARN - describeStreamResult(): Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
2021-07-05 12:35:34 [281472719503840] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
2021-07-05 12:35:34 [281472719503840] INFO - writeHeaderCallback(): RequestId: 2a14e9c2-a0a4-4f94-8588-a483204a0831
2021-07-05 12:35:34 [281472711111136] DEBUG - getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-703775db.kinesisvideo.ap-south-1.amazonaws.com"}
2021-07-05 12:35:34 [281472711111136] INFO - getStreamingEndpointResultEvent(): Get streaming endpoint result event.
2021-07-05 12:35:34 [281472711111136] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
2021-07-05 12:35:34 [281472711111136] DEBUG - getStreamingTokenHandler invoked
2021-07-05 12:35:34 [281472711111136] DEBUG - Refreshing credentials. Force refreshing: 1 Now time is: 1625488534399843068 Expiration: 18446744073709551615
2021-07-05 12:35:34 [281472711111136] INFO - getStreamingTokenResultEvent(): Get streaming token result event.
2021-07-05 12:35:34 [281472711111136] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
2021-07-05 12:35:34 [281472711111136] DEBUG - streamReadyHandler invoked
2021-07-05 12:35:34 [281472711111136] Stream is ready
====== VPUENC: 4.5.5 build on Mar 29 2021 07:40:42. ======
wrapper: 3.0.0 (VPUWRAPPER_ARM64_LINUX Build on Sep 14 2020 22:44:33)
vpulib: 1.1.1
firmware: 1.1.1.43690
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock

**

Winneee, 2021-07-21:
Hi Any comments on this bug ??

MushMal, 2021-07-21:
The failure you see originates from your upstream components and not KVS. Recommend replacing KVS sink with fakesink or another, stabilize your pipeline and only afterwards try to use KVS sink. The best place to get help with GStreamer related issues is their forums.

disa6302, 2021-08-02:
Closing since it does not seem to be a KVS specific issue.

closed 2021-08-02 by disa6302