awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #722: [BUG] kvssink gstreamer plugin says it is "Proprietary"

spotaws, 2021-07-06:
**Describe the bug**
The kvssink plugin has "Proprietary" in its gstreamer metadata License field. This is because
gstreamer checks the license field against these strings (and adds it to a denylist if it does not match)

`LGPL GPL QPL GPL/QPL MPL BSD MIT/X11 0BSD Proprietary`

https://github.com/GStreamer/gstreamer/blob/master/gst/gstplugin.c suggests that you should file a bug
to request additional licenses. This needs to be done for Apache 2.0. In the meantime, the plugin lies about
what license it has.

Basically, someone from this team should file a bug with gstreamer to get "Apache 2.0" added to the string checked for acceptable licensing, and when that happens, consider checking the system gstreamer for that support and if found, replacing the license string with the correct one.

clogwog, 2022-05-28:
created a [PR in gstreamer](https://gitlab.freedesktop.org/gstreamer/gstreamer/-/merge_requests/2514) for this

clogwog, 2022-05-29:
This will be available in GStreamer  1.20.3