awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #725: Is there anyway to detect when the Gstreamer pipeline fails to connect to kvs after it is started [QUESTION]

AmineFrioua, 2021-07-15:
so basically i m asking if the operation of sending data to kvs  receiver takes too much time or when the network of my machine fails, is there a way to detect after the pipeline has already started (state is playing) 
in my current case it keeps trying to sending data and loses the stream handle and when the connection is back it says that timeout issues happened in the logs

hassanctech, 2021-08-16:
Yes so we provide callbacks but stream latency pressure and buffer pressure (among others), which fire in the case when the stream latency you've specified gets exceeded (likely due to bad network, you accumulate frames faster than you can send them out), similar for buffer pressure if the memory buffer you specify is full.  `max-latency` and `storage-size` are the relevant properties in the `kvssink` plugin.

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/gstreamer/gstkvssink.cpp#L457

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/gstreamer/gstkvssink.cpp#L547

We provide ACKs from the backend which get logged for RECEIVED/BUFFERED/PERSISTED fragments.  

As long as you enable for example DEBUG level logging you will see all this information.

If you're seeing timeout issues then likely there are network issues.

hassanctech, 2021-09-07:
Assume the question is answered, closing due to inactivity.

closed 2021-09-07 by hassanctech