awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #728: log4cplus:ERROR could not open file ./kvs log configuration

RohanK1411, 2021-07-24:
log4cplus:ERROR could not open file ./kvs log configuration

After running rtsp connection

disa6302, 2021-07-28:
@RohanK1411 , 

`kvs_log_configuration` file exists in `samples` directory. If you are running GStreamer pipeline, you can set the path using the `log-config` element. Closing. Feel free to reach out if you more questions.

closed 2021-07-28 by disa6302

rach21nag, 2021-12-23:
@disa6302, after being able to successfully stream video once from my Raspberry Pi, I am getting an error when I try to run it again. 

log4cplus: ERROR Failed to open file ./log/kvs.log 

Please let me know what is wrong?

jorgejgnz, 2022-06-08:
Any solution? I'm experiencing the same problem. I'm using the Gstreamer pipeline but I don't know what to change in `kvs_log_configuration`