awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #729: [QUESTION]

ryfalafel, 2021-07-27:
CMake Error at dependency/libkvscproducer/kvscproducer-src/CMakeLists.txt:118 (message):
  OpenSSL is not found.  Make sure to export PKG_CONFIG_PATH to where
  OpenSSL's pc file is

i keelp getting this error, pls help

disa6302, 2021-07-28:
@ryfalafel ,

There is no information in this ticket. Please update with details on what OS is used, version of SDK, build output.

ryfalafel, 2021-07-29:
> @ryfalafel ,
> 
> There is no information in this ticket. Please update with details on what OS is used, version of SDK, build output.

Sorry, I was running MacOS 1.5 bigSur.
 I managed to fix the error but then i got tis error instead
**CMake Error at CMake/Utilities.cmake:99 (message):
  CMake step for liblog4cplus failed: 2
Call Stack (most recent call first):
  CMakeLists.txt:76 (build_dependency)**

disa6302, 2021-08-02:
Please ensure you run brew install to set up log4cplus on your Mac. Follow the build instructions on the main README in this repository. Due to lack of details, I am closing this issue. If you still face issues after looking at some previous github issues for solution or following the README, feel free to open a new issue with more details. Thanks.

closed 2021-08-02 by disa6302